The-OpenROAD-Project/OpenROAD #6517: cts: clock tree not generated when clock port is buffered in netlist with a buffer cell used by CTS

titan73, 2025-01-13:
### Describe the bug

I encountered this issue when making a test case with a clock port buffered by a clock buffer in the input netlist.
The clock buffer type is gf180mcu_fd_sc_mcu9t5v0__buf_4.

If the sale buffer type is used to generate clock tree with:
`clock_tree_synthesis -buf_list gf180mcu_fd_sc_mcu9t5v0__buf_4`
The CTS fails:
WARNING CTS-0104] Clock wire resistance/capacitance values are zero.
Use set_wire_rc to set them.
```
[INFO CTS-0050] Root buffer is gf180mcu_fd_sc_mcu9t5v0__buf_4.
[INFO CTS-0051] Sink buffer is gf180mcu_fd_sc_mcu9t5v0__buf_4.
[INFO CTS-0052] The following clock buffers will be used for CTS:
                    gf180mcu_fd_sc_mcu9t5v0__buf_4
[INFO CTS-0049] Characterization buffer is gf180mcu_fd_sc_mcu9t5v0__buf_4.
[INFO CTS-0007] Net "clk" found for clock "clock_osc_4m".
[WARNING CTS-0105] Net "clk" already has clock buffer mytop/buf. Skipping...
[WARNING CTS-0105] Net "mytop\/myclk" already has clock buffer mytop/buf. Skipping...
[WARNING CTS-0083] No clock nets have been found.
[INFO CTS-0008] TritonCTS found 0 clock nets.
[WARNING CTS-0082] No valid clock nets in the design.
```

Using another buffer type to make clock tree with:
`clock_tree_synthesis -buf_list gf180mcu_fd_sc_mcu9t5v0__buf_8`
The CTS succeeds:
```
[WARNING CTS-0104] Clock wire resistance/capacitance values are zero.
Use set_wire_rc to set them.
[INFO CTS-0050] Root buffer is gf180mcu_fd_sc_mcu9t5v0__buf_8.
[INFO CTS-0051] Sink buffer is gf180mcu_fd_sc_mcu9t5v0__buf_8.
[INFO CTS-0052] The following clock buffers will be used for CTS:
                    gf180mcu_fd_sc_mcu9t5v0__buf_8
[INFO CTS-0049] Characterization buffer is gf180mcu_fd_sc_mcu9t5v0__buf_8.
[INFO CTS-0007] Net "clk" found for clock "clock_osc_4m".
[WARNING CTS-0041] Net "clk" has 1 sinks. Skipping...
[INFO CTS-0010]  Clock net "mytop/myclk" has 100 sinks.
[INFO CTS-0008] TritonCTS found 1 clock nets.
[INFO CTS-0097] Characterization used 1 buffer(s) types.
[INFO CTS-0200] 0 placement blockages have been identified.
[INFO CTS-0201] 0 placed hard macros will be treated like blockages.
[INFO CTS-0027] Generating H-Tree topology for net mytop\/myclk.
[INFO CTS-0028]  Total number of sinks: 100.
[INFO CTS-0030]  Number of static layers: 0.
[INFO CTS-0020]  Wire segment unit: 50400  dbu (25 um).
[INFO CTS-0023]  Original sink region: [(1229735, 4945), (1724775, 156335)].
[INFO CTS-0024]  Normalized sink region: [(24.3995, 0.0981151), (34.2217, 3.10188)].
[INFO CTS-0025]     Width:  9.8222.
[INFO CTS-0026]     Height: 3.0038.
 Level 1
    Direction: Horizontal
    Sinks per sub-region: 50
    Sub-region size: 4.9111 X 3.0038
[INFO CTS-0034]     Segment length (rounded): 2.
 Level 2
    Direction: Vertical
    Sinks per sub-region: 25
    Sub-region size: 4.9111 X 1.5019
[INFO CTS-0034]     Segment length (rounded): 1.
 Level 3
    Direction: Horizontal
    Sinks per sub-region: 13
    Sub-region size: 2.4556 X 1.5019
[INFO CTS-0034]     Segment length (rounded): 1.
[INFO CTS-0032]  Stop criterion found. Max number of sinks is 15.
[INFO CTS-0035]  Number of sinks covered: 100.
[INFO CTS-0018]     Created 9 clock buffers.
[INFO CTS-0012]     Minimum number of buffers in the clock path: 2.
[INFO CTS-0013]     Maximum number of buffers in the clock path: 2.
[INFO CTS-0015]     Created 9 clock nets.
[INFO CTS-0016]     Fanout distribution for the current clock = 9:2, 11:1, 12:2, 15:1, 16:2..
[INFO CTS-0017]     Max level of the clock tree: 3.
[INFO CTS-0098] Clock net "mytop\/myclk"
[INFO CTS-0099]  Sinks 106
[INFO CTS-0100]  Leaf buffers 0
[INFO CTS-0101]  Average sink wire length 141.80 um
[INFO CTS-0102]  Path depth 2 - 2
[INFO CTS-0207]  Leaf load cells 6

```

### Expected Behavior

Clock tree should be generated.

### Environment

```shell
Latest OR
```

### To Reproduce

Test case:
[no_cts.zip](https://github.com/user-attachments/files/18400056/no_cts.zip)

maliberty, 2025-01-13:
I'm curious why you have a buffer in the clock tree pre-cts.  What is the goal?

titan73, 2025-01-13:
As mentioned, it was for a test case for another issue. I just noticed this behavior.

More concretely, in a real design that could happen to add a buffer to have a design invariant with a dont_touch on it so a specific sdc constraint can be applied on it. We usually use such hardcells in our designs for this purpose.

maliberty, 2025-01-13:
If the buffer is connected to an input then you could just apply the constraint to the input pin to equivalent effect, no?

My concern is that having a partially constructed clock tree and expecting OR to fill in the rest is non-trivial.  We do handle clock-gating but I'm hesitant to try to generalize it.

gadfort, 2025-01-13:
@maliberty this is the same issue as https://github.com/The-OpenROAD-Project/OpenROAD/issues/6352

titan73, 2025-01-13:
> If the buffer is connected to an input then you could just apply the constraint to the input pin to equivalent effect, no?
> 
Yes it'll work in that specific case but I was more thinking about a more general case with some stuff such as dft muxes, clock dividers, ... after the clock port. We had some designs like this.

> My concern is that having a partially constructed clock tree and expecting OR to fill in the rest is non-trivial. We do handle clock-gating but I'm hesitant to try to generalize it.
>
I'm not facing this issue right now so it can be low priority.  I'm just reporting since I encountered it.

maliberty, 2025-01-13:
Duplicate of #6352

closed 2025-01-13 by maliberty