The-OpenROAD-Project/OpenROAD #6532: DRT integration with GRT include issue

bnmfw, 2025-01-15:
### Description

For #5867 to be possible, it is essential for the GRT to have access to the DRT interface. During the attempts to make all the includes to allow for this to happen many issues arose. @eder-matheus mentioned that there have been previous attempts to implement this integration which lead to the same problems. The goal is, in any function of GlobalRouter.cpp, to be able to call drt_->anyPublicFunction().

###Main Attempted Changes Diff:
drt/CMakeLists.txt:
```
--- a/src/grt/CMakeLists.txt
+++ b/src/grt/CMakeLists.txt
@@ -88,6 +88,7 @@ target_link_libraries(grt_lib
     dbSta_lib
     stt_lib
     rsz_lib
+    drt
     OpenSTA
     Boost::boost
     OpenMP::OpenMP_CXX
@@ -103,6 +104,7 @@ target_link_libraries(grt
     gui
     stt
     rsz
+    drt
     OpenSTA
     Boost::boost
     OpenMP::OpenMP_CXX
@@ -131,6 +133,7 @@ if (Python3_FOUND AND BUILD_PYTHON)
       dbSta
       gui
       stt
+      drt
       OpenSTA
       Boost::boost
   )
```
GlobalRouter.cpp:

```
--- a/src/grt/src/GlobalRouter.cpp
+++ b/src/grt/src/GlobalRouter.cpp
@@ -69,6 +69,7 @@
 #include "odb/wOrder.h"
 #include "rsz/Resizer.hh"
 #include "rsz/SpefWriter.hh"
+#include "triton_route/TritonRoute.h"
 #include "sta/Clock.hh"
 #include "sta/MinMax.hh"
 #include "sta/Parasitics.hh"
```

bnmfw, 2025-01-15:
@maliberty eder mentioned that you might have some ideas on how to tackle this.

maliberty, 2025-01-15:
What are the issues you are facing?

maliberty, 2025-01-15:
I tried the above and it looks like a simple linking issue.  I've got a PR coming.

maliberty, 2025-01-15:
Please make a small reproducer on a branch or PR

bnmfw, 2025-01-15:
I have the [branch here](https://github.com/bnmfw/OpenROAD_Bernardo_Sandoval/tree/drt_grt_integration), it uses a dummy function to call drt

QuantamHD, 2025-01-16:
Largely the thing you need to do is decouple the openroad singleton from drt

What that boils down to in practice is a bunch of calls to the gui
If you look at the resizer this requires you creating an abstract class that encapsulates the gui behavior and injecting that abstract class into the drt object

You also need to seperate the MakeResizer and swig libraries from drt target into drt_lib which should contain most of the drt code, and the drt which should include drt_lib to the MakeDrt.cc + any other openroad app specific junk

PR #6534 (2025-01-15): Depend on utl_lib not utl
Fixes #6532
commit 1abb8087 (2025-01-15): Depend on utl_lib not utl

Fixes #6532
closed 2025-03-11 by maliberty