The-OpenROAD-Project/OpenROAD #6542: Placer divergence on bp_be/nangate45

povik, 2025-01-16:
### Describe the bug

In an experiment to start inferring clock gates from Yosys, the global placer diverges on the design. The last lines are:
```
4:57:07  [NesterovSolve] Iter: 1640 overflow: 0.176 HPWL: 3556483454
14:57:08  [NesterovSolve] Iter: 1650 overflow: 0.155 HPWL: 3551918924
14:57:08  [NesterovSolve] Iter: 1660 overflow: 0.138 HPWL: 3549444690
14:57:08  [INFO GPL-0100] Timing-driven iteration 6/6, virtual: false.
14:57:08  [INFO GPL-0101] Iter: 1660, overflow: 0.144, keep rsz at: 0.3
14:57:34  [INFO GPL-0106] Timing-driven: worst slack -2.61e-10
14:57:34  [INFO GPL-0103] Timing-driven: weighted 5369 nets.
14:57:34  [INFO GPL-0107] Timing-driven: repair_design delta area: -126.244 um^2 (-0.11%)
14:57:34  [INFO GPL-0108] Timing-driven: new target density: 0.43047988
14:57:34  [NesterovSolve] Iter: 1670 overflow: 0.143 HPWL: 4368350234
14:57:34  [NesterovSolve] Iter: 1680 overflow: 0.113 HPWL: 4746991274
14:57:34  [NesterovSolve] Iter: 1690 overflow: 0.132 HPWL: 5348893280
14:57:34  [ERROR GPL-0307] RePlAce divergence detected. Re-run with a smaller max_phi_cof value.
14:57:34  Error: global_place.tcl, 40 GPL-0307
```

### Expected Behavior

No divergence

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.5.0-1025-gcp
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-17941-g7fb347f37
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
[... 114 lines cut ...]
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.ghacZKCkuQ
```

### To Reproduce

This is taken from the CI run https://jenkins.openroad.tools/job/OpenROAD-flow-scripts-Public/job/PR-2657-head/

Download the attachment below (it's not a video, I've changed the extension to bypass github upload limits), rename it to a .tar.gz archive, and unpack.

To reproduce the placer divergence, run
```
(source vars-bp_be-nangate45-base.sh; openroad scripts/global_place.tcl)
```

https://github.com/user-attachments/assets/cb31391d-3854-48a8-82cb-f6a896217e91

### Relevant log output

```shell

```

maliberty, 2025-01-16:
I don't see a way to download the "video" as it just appears as a player.  Is the original file in Jenkins?  If not you can put it in gdrive and put a link here.

povik, 2025-01-16:
I have a right click -> "save video as" option (Firefox), in any case here's a Jenkins link: https://jenkins.openroad.tools/job/OpenROAD-flow-scripts-Public/job/PR-2657-head/1/artifact/flow/final_report_bp_be_nangate45_base_2025-01-06_13-57.tar.gz

The reason I wanted to upload it into the issue is I am not sure about the persistence of Jenkins artifacts

maliberty, 2025-01-16:
As long as the PR is open the artifacts should persist.

gudeh, 2025-01-16:
I downloaded the artifact from jenkins. I see this placement when the divergence is detected:

[image]

The region on the bottom is interesting. 

Either way, I also noticed that the first non-virtual timing driven iterations is being called in the middle of routability iterations. That's probably the cause of the issue.

gudeh, 2025-01-16:
Never mind my previous message. I didn't noticed I used a modified branch for that run. Here is what we get the moment the divergence is detected. Timing-driven and routability-driven modes happen separately, as they should. I will investigate further.

[image]

gudeh, 2025-01-17:
Hi! Issue #6465 also has a similar divergence on megaboom. We thought of attempting a workaround to save a snapshot prior to the divergence and instead of throwing an error, finish with the saved snapshot.

The divergence is interesting because it usually goes on reducing overflow, although the HPWL increases too much. I saved the snapshot storing the min HPWL if the overflow is below 0.25, and finished with the stored snapshot if a divergence is detected.

For this BP_BE it worked nicely. It finished with the following placement, and successfully finished DRT.

[image]
```
[NesterovSolve] Iter: 1460 overflow: 0.186 HPWL: 4451466576
[NesterovSolve] Iter: 1470 overflow: 0.204 HPWL: 5216452376
[NesterovSolve] Iter: 1480 overflow: 0.183 HPWL: 7694873000
[NesterovSolve] Iter: 1490 overflow: 0.180 HPWL: 9198722870
[NesterovSolve] Iter: 1500 overflow: 0.215 HPWL: 12663756677
[NesterovSolve] Iter: 1510 overflow: 0.232 HPWL: 15009139831
[NesterovSolve] Iter: 1520 overflow: 0.196 HPWL: 4578771851
[NesterovSolve] Iter: 1530 overflow: 0.206 HPWL: 5237016833
[NesterovSolve] Iter: 1540 overflow: 0.226 HPWL: 6824969538
[NesterovSolve] Iter: 1550 overflow: 0.199 HPWL: 8403828118
[NesterovSolve] Iter: 1560 overflow: 0.208 HPWL: 10136766760
[NesterovSolve] Iter: 1570 overflow: 0.202 HPWL: 11161608369
Divergence detected, reverting to snapshot with min hpwl.
Revert to iter: 1453 overflow: 0.201 HPWL: 4099702111
```

gudeh, 2025-01-27:
I tested @povik 's branch with latest master and the flow went successfully for the reported design, so I am closing the issue.

closed 2025-01-27 by gudeh

povik, 2025-01-27:
Thanks Augusto!