The-OpenROAD-Project/OpenROAD #6554: Add support for limiting heatmaps to selection

oharboe, 2025-01-20:
### Description

Estimated/or actual routing heatmap includes all nets:

[image]

However, it would be nice if one could select nets by name(ctrl-f) to look at specific nets. Similarly for placement heatmap.

[image]

### Suggested Solution

Add filtering option(toggle?) for heatmaps to use selected items only.

namanjain2001-code, 2025-05-10:
i am interested to work on this issue.

maliberty, 2025-05-10:
Great!  What help do you need to get started?

namanjain2001-code, 2025-05-11:
Currently, the Ctrl+F 'Find Object' popup is used to search and highlight specific nets or instances. Would it be a good idea to add a checkbox in this popup (e.g., 'Show Heatmap') so that when it's checked, the heatmap around the selected net or instance becomes visible or focused? Would this feature be useful?"

maliberty, 2025-05-12:
I think we should keep selection separate from heatmap display.  There are many ways someone could select objects including tcl commands and layout clicking.   The heatmap setting should have an option to include only selected objects.

namanjain2001-code, 2025-05-13:
Okay, now I got it — thanks for the clarification. I've noticed that there are different settings/configuration windows for each type of heatmap. I think I need to add a checkbox here to control this behavior. Am I right?

namanjain2001-code, 2025-05-18:
I'm able to render the required UI successfully, but I'm facing an issue on the backend side.

My current approach is to retrieve the selected object using:
const gui::Selected& sel = gui::Gui::get()->getInspectorSelection();

Then, I plan to use this selection to update the following function:

void HeatMapDataSource::addToMap(const odb::Rect& region, double value);

Specifically, I intend to filter based on the selected net like this:

if (selected_net != nullptr && map_pt->net != selected_net) {
    continue;
}
Would this be a valid approach? Or is there a better/cleaner way to implement filtering based on the selected object in the GUI?

gadfort, 2025-05-18:
> At the UI level you will need to address each setting box. At the code level the heatmaps do have some sharing so you may be able to store the setting in a more common base class.
> 
> [@gadfort](https://github.com/gadfort) any opinions?

@maliberty I missed this, it's not 100% clear to me what the request is.
1. Does it apply to all heatmaps or only some?
2. For both rudy and routing congestion, the map if built from the all the nets (the change would need to be in GRT to correctly handle this).
3. How would you enable/disable this behavior, I would expect this should be disabled by default as the selection might have nothing to do with the heatmap view and just interfere with the user.

namanjain2001-code, 2025-05-18:
Ok if I am not worng for routing congestion we can filter heatmap on based on it. And for other heatmap it might not work??

gadfort, 2025-05-18:
@namanjain2001-code for the rudy estimate the calculation comes from:
https://github.com/The-OpenROAD-Project/OpenROAD/blob/ce617810e4dc11b3cdb5a267ef3454788f81aaf0/src/grt/src/Rudy.cpp#L117
and for routing congestion:
https://github.com/The-OpenROAD-Project/OpenROAD/blob/ce617810e4dc11b3cdb5a267ef3454788f81aaf0/src/grt/src/heatMap.cpp#L127

It's possible I'm missing something, but this appears to be the source of the data for the heatmaps

namanjain2001-code, 2025-05-19:
"Hi @oharboe, in the two images you shared, could you clarify on what basis the heatmap data was filtered? Was it based on a specific group of related nets (e.g., functionally grouped nets), or was it a random selection of nets—perhaps selected by manually dragging over a region? Just trying to understand the selection/filtering criteria used for generating the filtered heatmap."

oharboe, 2025-05-19:
> "Hi [@oharboe](https://github.com/oharboe), in the two images you shared, could you clarify on what basis the heatmap data was filtered? Was it based on a specific group of related nets (e.g., [...]

I used a regular expression to pick nets that I wanted to study.

maliberty, 2025-07-17:
Welcome and feel to to join in.  There hasn't been an update from @namanjain2001-code in a while so I'm not sure if he is still working on this.

kirobaghdad, 2026-03-11:
@shubhika123 
Hello! I would like to help with this issue, if you like!

shubhika123, 2026-03-14:
Hi @kirobaghdad, thanks for offering. I've already submitted a PR for this: #9756. Feel free to take a look and share any feedback! 😃

shubhika123, 2026-03-14:
Thanks @Divinesoumyadip! That's actually already been done in the PR:- getSelectedInsts() is now a protected helper in 'HeatMapDataSource' so all three data sources use it instead of duplicating. Feel free to check out #9756!