The-OpenROAD-Project/OpenROAD #6555: Area in GUI and `sta::format_area [rsz::design_area] 0` does not match

oharboe, 2025-01-20:
### Describe the bug

I'm unsure what the GUI is telling me...

I tried with `make gui_floorplan` and `make gui_final`

[image]

[image]

### Expected Behavior

I expected area to match `sta::format_area [rsz::design_area] 0`

### Environment

```shell
OpenROAD v2.0-18391-g0cd2a2b8b
```

### To Reproduce

See above

### Relevant log output

```shell

```

maliberty, 2025-01-20:
The area will not be invariant across optimization and CTS which add or resize gates.

oharboe, 2025-01-20:
> The area will not be invariant across optimization and CTS which add or resize gates.

I see.

I'm still puzzled by not being able to find the `sta::format_area [rsz::design_area] 0` in the GUI, shouldn't I be able to find the exact same number to understand what `sta::format_area [rsz::design_area] 0` is telling me?

maliberty, 2025-01-21:
If you use remove_buffers it will vary a lot.  I don't see a specific issue here to address.

oharboe, 2025-01-21:
The issue to address is that I'm trying to have enough information in the GUI that I can understand and verify that I'm getting a reasonable number for `sta::format_area [rsz::design_area] 0`.

So far, I'm not able to figure out what `sta::format_area [rsz::design_area] 0` comprises by looking at the userinterface.

Also, surprisingly, I see `sta::format_area [rsz::design_area] 0` varying a lot through the flow. Since it varies so much, I'm only keener to understand exactly what `sta::format_area [rsz::design_area] 0` is returning.

oharboe, 2025-01-21:
My expectation is to find or easily calculate the report_design_area in the GUI

maliberty, 2025-01-22:
It seems easy to me so please be more specific

oharboe, 2025-01-22:
I don't recall how I was confused, but I've rebuilt with latest and now, yes, the user interface seems clear to me. I can see what report_design_area consists of.

It was surprising to me, because I didn't realize it, but the nature of ASICs, that these cells vary in area significantly across the flow.

[image]