The-OpenROAD-Project/OpenROAD #6556: How to change message limt when encounting "message limit reached, this message will no longer print"?

xxzh12, 2025-01-20:
### Description

How to change message limt when encounting "message limit reached, this message will no longer print"?

maliberty, 2025-01-20:
It is a currently a fixed limit (1000).  What motivates the need to change it?

xxzh12, 2025-01-20:
My `detailed_placement` fails on more than 1000 instances and I am trying to list all of them.
I've spent quite some time searching through the documentation to find a way to modify the max message limit, as I believe there should be an interface for this.

maliberty, 2025-01-20:
Markers are saved to the and can be viewed in the GUI.  You can also use -report_file_name to save them to the file.  I think adding them to the log is not the right place.

xxzh12, 2025-01-22:
I see. Thanks!

closed 2025-01-22 by xxzh12