The-OpenROAD-Project/OpenROAD #6558: Add a Qt splashscreen with logging for `make gui_` to handle long running startup times

oharboe, 2025-01-20:
### Description

https://doc.qt.io/qt-5/qsplashscreen.html

### Suggested Solution

- Add splash screen
- Log output to splash screen
- Make the splash screen responsive so that load can be cancelled(ctrl-c equivalent) without having to terminate OpenROAD via task manager.

maliberty, 2025-01-20:
This is not the usual purpose of a splash screen.  I don't think it a good solution.

oharboe, 2025-01-20:
> This is not the usual purpose of a splash screen. I don't think it a good solution.

Which part?

Showing progress while taking a while to ready the GUI is what is exactly what is described in the manual for splashscreen.

maliberty, 2025-01-20:
I've never used an app whose splash screen stayed up for 20 minutes which could happen with OR.   Trying to transfer the log from one window to another seems like a bother without a benefit.   What is this getting us over just showing the window with the existing log?

We don't support cancel in general so the ctrl-c request is orthogonal and not supported.

oharboe, 2025-01-20:
Yeah, it's not great.

Effort is better spent in storing data that takes so long to generate in the .odb.

closed 2025-01-20 by oharboe