The-OpenROAD-Project/OpenROAD #6568: Instance is not placed ERROR while global_placement

aboke1988, 2025-01-22:
### Describe the bug

I am running coyote_tc design under sky130hd, because i want to run design with pads.
while running the flow i am getting this issue

[ERROR GRT-0010] Instance u_rocc_resp_data_i/u_io\[1\]/u_in is not placed.
Error: global_place.tcl, 40 GRT-0010
Command exited with non-zero status 1

I am attaching the screenshot here .

Please help me in this regards as why is this issue coming and how to negate it.

[image]

### Expected Behavior

The complete run should go without any error.

### Environment

```shell
iitg@iitg-amolb:~/openRoad_dir/OpenROAD-flow-scripts$ source env.sh 
OPENROAD: /home/iitg/openRoad_dir/OpenROAD-flow-scripts/tools/OpenROAD
```

### To Reproduce

[https://drive.google.com/file/d/1Yf24tf_LbpNQ9HX7c1PPvpAEqj8RWFDh/view?usp=sharing](url)

### Relevant log output

```shell

```

aboke1988, 2025-01-30:
So, Which design do you prefer I should which has PADs. I was given the idea that i should take reference of coyot_tc if i want to take place pads in my design. Is there any other working design which has pads and i could take reference to ?

maliberty, 2025-01-30:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/tree/master/flow/designs/ihp-sg13g2/i2c-gpio-expander was recently contributed and has pads.  Most of our test cases with pads are in private PDKs unfortunately.  Coyote_tc was intend to help but it never got finished.