The-OpenROAD-Project/OpenROAD #6589: Improve global router for asap7/ethmac_lvt where it fails with a single DRC error

oharboe, 2025-01-24:
### Description

In https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2694 asap7/ethmac_lvt fails global route with [a single DRC error after 100 iterations](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2694#issuecomment-2611106797).

To reproduce untar and run https://drive.google.com/file/d/1XCfVGa0DHAz5MO_pJwR_VE8qOhZ1oqoO/view?usp=sharing

No luck in whittling down the problem with deltaDebug.

[image]

### Suggested Solution

Improve global router.

maliberty, 2025-01-28:
The IO pin placement is totally different than master which has few pins on the e/w edges.

oharboe, 2025-01-28:
> The IO pin placement is totally different than master which has few pins on the e/w edges.

This is unsurprising to me: a completely different optimal solution can be found with tiny differences in initial conditions.

So what does this mean practically?

- If this is a ppl problem? Pin ppl chooses a pin placements that is never going to work in global routing and it should have done better?
- There's nothing wrong here with ppl, nor grt, just lock down pin placement in ORFS test cases in git to avoid random changes breaking master?
- Wait for initial conditions to change again slightly, then merge https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2694 ?

maliberty, 2025-01-28:
I think it is worth a bit of digging to see if this is just the flap of the butterfly's wings or something more interesting.

eder-matheus, 2025-02-04:
@oharboe @maliberty I believe this is a case where we apply an unnecessarily large reducement in the routing resources. We're reducing it in 50% for every asap7 design, with few exceptions. I think this could be the case of studying different reduction values for this platform and choose a better one.

The attached test case finishes if I reduce the adjustment from 50% to 40%.

oharboe, 2025-02-04:
@eder-matheus Would it be possible to make the error message more actionable and specific?

"grt failed, try increasing  X to Y"?

Even if it takes a while to compute "Y" here, I think it would be worth the extra compute, because all the user can do is to randomly change variables and hope for the best...

eder-matheus, 2025-02-04:
> [@eder-matheus](https://github.com/eder-matheus) Would it be possible to make the error message more actionable and specific?
>
> "grt failed, try increasing X to Y"?
>
> Even if it takes a while to [...]

I created this issue to track this request: https://github.com/The-OpenROAD-Project/OpenROAD/issues/6638.
And I believe this other issue is also related to this topic: https://github.com/The-OpenROAD-Project/OpenROAD/issues/6404.

I will work on it in the next days, hopefully we can have at least a rough measure for new adjustment suggestion.

maliberty, 2025-02-04:
The best we can do is try to guess at the better value.  We can't really guarantee anything as you would have to re-run the flow run global placement forward to see if it works.

oharboe, 2025-02-04:
my guess is completely uniformed... 

All I can do is to scan values and look at the results to try to reason what constitutes a "good" value.