The-OpenROAD-Project/OpenROAD #6598: gpl creates unroutable placement for centered floorpan

stefanottili, 2025-01-27:
### Describe the bug

ICCAD04/RISC2 uses a centered floorplan, something that was very common in that timeframe.

In Aug 2024, OR gpl was able to place/global route/detail route this.

As of today, gpl creates an unroutable placement.

Please see https://github.com/The-OpenROAD-Project/OpenROAD/issues/5557 for pictures and a tar file with the test case.

### Expected Behavior

routable placement

### Environment

```shell
M1 Mac homebrew, OR as of Jan 27 2025
```

### To Reproduce

script in tar file, see https://github.com/The-OpenROAD-Project/OpenROAD/issues/5557

### Relevant log output

```shell

```

maliberty, 2025-01-27:
Support for such is a very low priority.

stefanottili, 2025-02-12:
replaced by https://github.com/The-OpenROAD-Project/OpenROAD/issues/6704
gpl/rudy work just fine with a centered floorplan, it's the detailed_placement that messes up the stdcells.

closed 2025-02-12 by stefanottili