The-OpenROAD-Project/OpenROAD #6615: Catch stale parasitics in resizer

povik, 2025-01-30:
### Description

#6614 was a case of stale parasitics inside repair_timing due to missing invalidation. This can severely degrade the repair process as it's working on the wrong paths.

### Suggested Solution

I suspect the parasitics attached to a net get cleared inside OpenSTA when the connections change. This means we could programmatically check for missing parasitics to catch cases of omitted invalidation.

povik, 2025-01-30:
> I suspect the parasitics attached to a net get cleared inside OpenSTA when the connections change. 

This is probably not the case, there's code to handle changes gracefully instead of removing the parasitic annotation completely: https://github.com/parallaxsw/OpenSTA/blob/9c610f63451bc508f2d150a02ff8b20be74d7f1f/parasitics/ConcreteParasitics.cc#L861

This requires more investigation.

maliberty, 2025-01-30:
The management of parasitics seems overly manual.  Getting it handled by db callback makes more sense.

povik, 2025-06-02:
With #7464 we've made invalidation happen via odb callbacks which removes the need for manual invalidation