The-OpenROAD-Project/OpenROAD #6618: [ERROR PDN-0351] Ring shape falls outside the die bounds.

oharboe, 2025-01-30:
### Describe the bug

untar and run [pdn_sdq_17x64_asap7_mocked_2025-01-30_15-10.tar.gz](https://github.com/user-attachments/files/18603703/pdn_sdq_17x64_asap7_mocked_2025-01-30_15-10.tar.gz)

```
$ ./run-me-sdq_17x64-asap7-mocked.sh 
OpenROAD v2.0-18694-g401a7bb30 
Features included (+) or not (-): +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[ERROR PDN-0351] Ring shape falls outside the die bounds.
Error: BLOCK_grid_strategy.tcl, 23 PDN-0351
openroad> 
```

### Expected Behavior

Used to work until somewhat recently

### Environment

```shell
OpenROAD v2.0-18694-g401a7bb30
```

### To Reproduce

See above

### Relevant log output

```shell

```

oharboe, 2025-01-30:
@jeffng-or FYI fails in https://github.com/The-OpenROAD-Project/megaboom/pull/229

gadfort, 2025-01-30:
@oharboe this is intentional now (https://github.com/The-OpenROAD-Project/OpenROAD/issues/6445)
You need to fix your power grid or explicitly allow the rings to go outside the die (I would recommend fixing the grid instead of allowing it outside the die).

oharboe, 2025-01-30:
But I am using a PDN from ORFS platforms/asap7 folder and this .tcl file is used in a ton of places and this is the first place I have seen it fail. I am not familiar with any design specific knobs that I can tweak.

Could you have a look and give some more specific instructions for non PDN expert...?

gadfort, 2025-01-30:
I think you can just make the width smaller, spacing smaller, or core offset smaller to resolve this.
The issue needs to be fixed in ORFS.

oharboe, 2025-01-30:
> I think you can just make the width smaller, spacing smaller, or core offset smaller to resolve this.
> The issue needs to be fixed in ORFS.

Silly question: which ORFS variables?

gadfort, 2025-01-30:
I think it's in a file in the platform enablement, no variables involved.

oharboe, 2025-01-30:
@maliberty @jeffng-or I guess you are way ahead of me on this :-)

oharboe, 2025-01-30:
Can the error message give more specific advice: increase by X and Y?

maliberty, 2025-01-30:
> Can the error message give more specific actionable advice: increase by X and Y?

@gadfort ?  Of course there are multiple ways to fix it so it would just be one possibiilty.

gadfort, 2025-01-30:
@maliberty pdn could output the amount by which the ring extends outside of the die?

maliberty, 2025-01-30:
Let's have some text to suggest how that value might be used.  Some like
```
PDN rings do not fit inside the die area by <val> um in X and <val2> um in Y.  Either reduce the ring area or increase the core to die spacing to accommodate.  Use -allow_out_of_die if this is intentional.
```

oharboe, 2025-01-30:
I modified bazel-orfs mocked area to scale the die area, but keep the die to core distance unchanged: https://github.com/The-OpenROAD-Project/bazel-orfs/pull/276

oharboe, 2025-01-30:
Closing as duplicate/known problem.

Improved actionable and specific error messages are great usability improvements, suggestion from @maliberty looks good.

closed 2025-01-30 by oharboe