The-OpenROAD-Project/OpenROAD #6619: gpl: gui debug mode does not stop for some designs

gudeh, 2025-01-30:
### Description

It happens only for some designs.
Here is a list of some designs I noticed it is not stopping:
-nangate45/jpeg
-nangate45/gcd
-nangate45/ibex
-nangate45/swerv
-sky130hd/riscvi
-sky130hd/gcd
-sky130hd/jpeg

And here are designs that successfully stopped for gpl debug mode in the gui:
-sky130hd/chameleon
-asap7/microwatt
-asap7/jpeg
-sky130hd/aes

maliberty, 2025-01-30:
Please give an example of the command you are using when it doesn't stop.  I find it hard to understand why it would be design specific.

gudeh, 2025-01-30:
I inserted line 8 to `flow/scripts/global_place_skip_io.tcl` in this image. `global_placement_debug -pause 10 -update 10 -draw_bins`

[image]

Also, for each of the designs I tried I went to `ORFS/flow` directory and did:
`make DESIGN_CONFIG=./designs/sky130hd/chameleon/config.mk global_place_skip_io_issue`
`emacs run-me-chameleon-sky130hd-base.sh`
inserted the `-gui` flag to the openroad command:

[image]

and ran ./run-me-chameleon-sky130hd-base.sh

maliberty, 2025-02-01:
I believe the problem is `[place_density_with_lb_addon]` which runs part of global placement and probably consumes the debug before global_placement happens.