The-OpenROAD-Project/OpenROAD #6641: isSequential returns 0 for all cells in Nangate45

mguthaus, 2025-02-05:
### Describe the bug

If I call isSequential on a DFF_X1 master cell in Nangate45, it returns 0. It returns 1 on DFF cells in other technologies. Is it a problem with the lib parser?

### Expected Behavior

Return 1 for DFF_X1 master cells.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.0.12-76060012-generic
os: Pop!_OS 20.04 LTS
cmake version 3.29.6
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-18676-gc4ec647a4
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
[... 144 lines cut ...]
set inst [$block findInst "ctrl.state.out\\\[0\\\]\$_DFF_P_"]
set inst_name [$inst getName]
set master [$inst getMaster]
set master_name [$master getName]
puts "$inst_name [$master getName] [$master isSequential]"
```

### Relevant log output

```shell
OpenROAD v2.0-18676-gc4ec647a4
Features included (+) or not (-): +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0227] LEF file: platforms/nangate45/lef/NangateOpenCellLibrary.tech.lef, created 22 layers, 27 vias
[INFO ODB-0227] LEF file: platforms/nangate45/lef/NangateOpenCellLibrary.macro.mod.lef, created 135 library cells
[INFO ODB-0127] Reading DEF file: results/nangate45/gcd/base/6_final.def
[INFO ODB-0128] Design: gcd
[INFO ODB-0130]     Created 54 pins.
[INFO ODB-0131]     Created 859 components and 3523 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 1718 connections.
[INFO ODB-0133]     Created 682 nets and 1802 connections.
[INFO ODB-0134] Finished DEF file: results/nangate45/gcd/base/6_final.def
ctrl.state.out\[0\]$_DFF_P_ DFF_X1 0
```

maliberty, 2025-02-05:
I assume you are talking about dbMaster::isSequential.  It is a leftover from the original contribution of OpenDB.  It is not populated as it makes more sense to use sta for this.  We could try to copy over the state but they you are left with questions like is an ICG a sequential?  It has a statetable but its not what people usually think of.  Sta has the full Liberty details.

QuantamHD, 2025-02-05:
@maliberty Should stub isSequential function with an implementation that pull from STA, and just remove the odb layer? Or just remove the function all together?

QuantamHD, 2025-02-05:
Looks like it's used in a few places. Maybe some dead code around levelizing in dbBlock and also in dft.

[image]

maliberty, 2025-02-05:
Looks like dbMTerm::create sets isSequential if you create a clock mterm.  Its a pretty weak method but I suppose it would work on some pdks.  The levelizing stuff is unused afaik.

gadfort, 2025-02-05:
It seems to me that this parameter could be removed and the few places it's used they should be using opensta anyways (the scancells forexample)

maliberty, 2025-02-05:
@fgaray could you clean up the dft usage and then we can remove it?

fgaray, 2025-02-07:
PTAL #6667 for DFT code