The-OpenROAD-Project/OpenROAD #6643: Fanout cells inserted during global placement

kareefardi, 2025-02-05:
### Describe the bug

This is new behavior and there is nothing in the documentation which suggests that this is intentional. I also believe that global placement shouldn't manipulate the design at all. 

### Expected Behavior

Clarification if this is a bug or intentional. 

### Environment

```shell
Git commit: 6f4c5c592f3f5f2e0ee5d4a91b86622deedb57b5
kernel: Linux 6.11.10-1-liquorix-amd64
os: Ubuntu 22.04.5 LTS (Jammy Jellyfish)
cmake version 3.25.1
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-18783-g6f4c5c592
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare
[... 112 lines cut ...]

[reproducible.tar.gz](https://github.com/user-attachments/files/18671471/reproducible.tar.gz)

### Relevant log output

```shell
Cell type report:                       Count       Area
  Buffer                                   40     160.15
  Inverter                                 70     262.75
  Sequential cell                        1596   33947.56
  Multi-Input combinational cell         5038   54407.18
  Total                                  6744   88777.64
..
Cell type report:                       Count       Area
  Buffer                                   40     183.93
  Timing Repair Buffer                   1612   11360.90
  Inverter                                 70     267.76
  Sequential cell                        1596   32349.78
  Multi-Input combinational cell         5038   52237.60
  Total                                  8356   96399.96
```

### Additional Context

Using `report_cell_usage` before and after global placement shows additional `Timing Repair Buffer`

gudeh, 2025-02-05:
Hi @kareefardi, indeed it is not gpl itself that modifies the design. What happens is that gpl makes calls to rsz repair_design if gpl timing_driven option is enabled (which is by default).

On gpl initial iterations we do virtual calls to repair_design meaning we undo all of rsz modifications, and we maintain only some internal weights for gpl calculations.  On later gpl iterations we maintain whatever changes rsz made, meaning we have new cells inserted in the middle of placement execution. Further info available here: https://github.com/The-OpenROAD-Project/OpenROAD/pull/6165, on the PR where we added this feature. Prior to this change all timing-driven iterations by gpl were purely virtual, so there was no changes to the design.

So in summary, gpl modifications to cells could seem unintentional, but they are because we make calls to rsz repair_design, which does modify the design with cell resizings and buffer insertions.

I do not have a deep understanding of all the repair_design modifications, do you think any of its modifications are incorrect somehow?

maliberty, 2025-02-05:
It is intentional to add optimization to gpl as it allows for the placer to account for the added area from optimization.  If you really don't want it the -keep_resize_below_overflow would allow you to avoid it but you may get worse timing results in the total flow.

kareefardi, 2025-02-05:
Thanks for you replies. 

I don't believe the modification done are incorrect but it was surprising behavior to me at least. 

I think a proposed action here is to first add this to the documentation. I was also thinking global placement log should somehow be more verbose about this for example it might print the number of fanout violations it attempted to fix (similar to `repair_design` output)

Finally, would using `-keep_resize_below_overflow` completely match the old behavior or running reszier virtually?

QuantamHD, 2025-02-05:
-keep_resize_below_overflow set to 0 would return to the old behavior, but IMO this generally improves the openroad placement result. I wouldn't disable it by default unless you see degradation in your flow, and is similar to the off the shelf tools you can buy.

gudeh, 2025-02-05:
There is some explanation about timing-driven mode and the `-keep_resize_below_overflow ` [on gpl README](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/gpl/README.md). Let me know if you find it unclear.

Currently gpl reports the change in area due to repair_design changes.

maliberty, 2025-02-05:
Think of it as a fused placement + optimization approach.  Over time I expect more of optimization will be fused (eg repair_timing as well as repair_design).

donn, 2025-02-06:
The documentation claims that the default is 0, which shouldn't be surprising, however the actual default in the source code appears to be 0.3.

https://github.com/The-OpenROAD-Project/OpenROAD/blob/9c3bac8a28f438fa77383fa2035a66c1863a660a/src/gpl/src/replace.cpp#L116

kareefardi, 2025-02-06:
Ok I understand the fused placement + optimization approach and setting `keep_resize_below_overflow` to 0 does prevent resizer from retaining the changes. I do have a couple of comments:
- The documentation states that the default value for `keep_resize_below_overflow` is 0 but given the test case, it seems the default is a non-zero value. Perhaps its this https://github.com/The-OpenROAD-Project/OpenROAD/blob/9c3bac8a28f438fa77383fa2035a66c1863a660a/src/gpl/src/replace.cpp#L116
- I do think it is better if gpl also reports the number cells (fanout, max cap, etc.) added instead of just the area change with each iteration

Side note:
While using search in readthedocs this gpl document came up as well and it appears to be an outdated gpl document. 
https://openroad.readthedocs.io/en/latest/src/test/translator.html

gudeh, 2025-02-06:
I believe the suggestion of reporting the cells added is pertinent. I tried doing that when the integration was implemented but the rsz repair_design API for gpl is a little awkward, I shall have a look at that.

gadfort, 2025-02-07:
I would second the reporting the changes, I think it would provide some useful feedback as well.

gudeh, 2025-02-13:
> Thanks for you replies.
>
> I don't believe the modification done are incorrect but it was surprising behavior to me at least.
>
> I think a proposed action here is to first add this to the documentation. [...]

Hi @kareefardi, could you provide an example of the log message you expect to see in gpl which is similar to repair_design output? 

Would it be the one [with this format](https://github.com/The-OpenROAD-Project/OpenROAD/blob/c8948ab7ca41cf32ce15fa00d8904a5195ff37ba/src/rsz/src/RepairDesign.cc#L2090-L2100)?

gudeh, 2025-02-24:
I added these reports, I hope this is what you were expecting @gadfort @kareefardi, please let me know if there is something missing.

```
[INFO GPL-0100] Timing-driven iteration 4/5, virtual: false.
[INFO GPL-0101] Iter: 1502, overflow: 0.204, keep rsz at: 0.3
Iteration |   Area    | Resized | Buffers | Nets repaired | Remaining
---------------------------------------------------------------------
        0 |     +0.0% |       0 |       0 |             0 |     16543
    final |     +3.3% |     323 |       0 |             0 |         0
---------------------------------------------------------------------
[INFO RSZ-0039] Resized 323 instances.
[INFO GPL-0106] Timing-driven: worst slack -1.04e-10
[INFO GPL-0103] Timing-driven: weighted 1647 nets.
[INFO GPL-0107] Timing-driven: repair_design delta area: 6.017 um^2 (+0.30%)
[INFO GPL-0108] Timing-driven: repair_design, gpl cells created: 0 (+0.00%)
[INFO GPL-0109] Timing-driven: inserted buffers as reported by repair_design: 0
[INFO GPL-0110] Timing-driven: new target density: 0.7253905
```