The-OpenROAD-Project/OpenROAD #6704: detailed_placement creates unroutable placement for centered floorplan

stefanottili, 2025-02-12:
### Describe the bug

revisiting https://github.com/The-OpenROAD-Project/OpenROAD/issues/5557 shows that gpl/rudy work correctly for a centered floorplan.

It's the detailed_placement that messes up the stdcells and creates an unroutable placement.

### Expected Behavior

detailed_placement should be able to handle any DIEAREA ( x1 y1 ) ( x2 y2 ), not just x1 = 0, y1 = 0, x1 > 0 y2 > 0

### Environment

```shell
MacOS M1 homebrew OpenROAD v2.0-18955-g3332c5ee6
```

### To Reproduce

https://github.com/The-OpenROAD-Project/OpenROAD/issues/5557  has a tar file with lef/def/or command file.

### Relevant log output

```shell
[NesterovSolve] Finished with Overflow: 0.099042
Placement Analysis
---------------------------------
total displacement    3613512.4 u
average displacement      110.8 u
max displacement          736.0 u
original HPWL         1091240.1 u
legalized HPWL        4943157.6 u
delta HPWL                  353 %
```

stefanottili, 2025-02-12:
Same netlist using a `first quadrant` floorplan works as expected.
```
Placement Analysis
---------------------------------
total displacement     107598.5 u
average displacement        3.3 u
max displacement           15.1 u
original HPWL         1091253.2 u
legalized HPWL        1194890.2 u
delta HPWL                    9 %
```

stefanottili, 2025-02-13:
detailed_placement worked as of filing https://github.com/The-OpenROAD-Project/OpenROAD/issues/6598 in Aug 15 2024. bisecting is tricky since older versions don’t compile with boost 1.87

maliberty, 2025-02-13:
Don't bother bisecting as there were substantial changes for hybrid rows. This has no priority to fix

stefanottili, 2026-03-24:
It's stale because it has not been addressed, please leave it open.
dpl fails to legalize the ICCAD 2004 testcases, gpl/grt/drt/gui work with a centered floorplan.
No priority, but a bug nevertheless.