The-OpenROAD-Project/OpenROAD #6707: Discrepancies betwen get_nets/get_cells and getNets/getInsts

titan73, 2025-02-13:
### Describe the bug

I made the dumping function below which I used to after cts:
```
set num 1
proc dump {} {
    global num
    set db [ord::get_db_block]
    set file_id [open "dump_nets_sdc_$num" "w"]
    foreach net [get_nets] {
        puts $file_id [get_name $net]
    }
    close $file_id
    set file_id [open "dump_nets_odb_$num" "w"]
    foreach net [$db getNets] {
        puts $file_id [$net getName]
    }
    close $file_id
    set file_id [open "dump_cells_sdc_$num" "w"]
    foreach net [get_cells] {
        puts $file_id [get_name $net]
    }
    close $file_id
    set file_id [open "dump_cells_odb_$num" "w"]
    foreach net [$db getInsts] {
        puts $file_id [$net getName]
    }
    close $file_id
    incr num
}
```

Here are the results for nets:
  - get_nets => names are not escaped for both / and []
  - getNets => nets added by cts have / escaped, [] are not escaped 
And for insts:
  - get_cells => names are not escaped for both / and []
  - getInsts => nets added by cts have / escaped, [] are escaped

Note that appart from nets and insts added by cts, getInsts have [] escaped while getNets not.

### Expected Behavior

No discrepancies

### Environment

```shell
Latests wine
```

### To Reproduce

Whatever design with its tcl script with the dumping code above.

### Relevant log output

```shell

```

maliberty, 2025-02-18:
getInsts returns the name stored in odb.  get_cells converts that name to 'sdc' style in
```
// Translate sta namespace to sdc namespace.
// Remove all escapes.
const char *
SdcNetwork::staToSdc(const char *sta_name) const
```
so it appears they are effectively in different namespaces.  Likewise nets

maliberty, 2025-02-24:
Reopen if there is anything further to resolve but this is just an artifact of sta being sdc compatible.

closed 2025-02-24 by maliberty