The-OpenROAD-Project/OpenROAD #6712: Several DRT-0225: Pin not visited warnings

ozarkstock, 2025-02-13:
### Describe the bug

I am using a PDK with a limited metal stack of only metal 1 and metal 2. When running OpenROAD to place and route a design, the DRT-0225 warnings are produced for several nets. Looking at some other LEF views from other PDK's, I did some experimentation on what layer the tool has the best results routing to pins with. Following another process's example, one attempt was to declare pins on both the metal 1 and metal 2 layer in the same spot. This produced less than ideal results:

[image]

This type of route was not common, however something about metal 2 pins produces the message: "[INFO DRT-1000] Pin not in any guide. Attempting to patch guides to cover (at least part of) the pin." Isolating the pins to M1 had the best results, but despite that the pins are on grid the warnings persist. The design optimizes placement fairly quickly for the GCD circuit (12-15 optimization attempts) with occasional DRC violations. With another PDK that only has three metal layers, I could recreate the errors by limiting the routing layers to M1 and M2. With access to all three, no warnings were produced. I think this could be a bug with only having two metal layers accessible to the tool. Some insight as to why this particular warning is asserted and if there are ways to mitigate them would also be appreciated.

### Expected Behavior

No strange routes with no warnings. 

### Environment

```shell
OpenROAD v2.0-13893-gf0cc0ee1a 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.

[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.8.0-52-generic
os: Ubuntu 22.04.5 LTS (Jammy Jellyfish)
cmake version 3.25.1
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-13893-gf0cc0ee1a
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
[... 86 lines cut ...]
-- STA git sha: b71b48ad16590ea7fa144790a2f7ab2ab3d24640
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /usr/local
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- SSTA: 0
-- Found SWIG: /usr/bin/swig4.0 (found suitable version "4.0.2", minimum required is "3.0")  
-- STA executable: /home/stock/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/sta
-- Configuring incomplete, errors occurred!
```

### To Reproduce

[detail_route_gcd_sky130hd_base_2025-02-13_14-40.tar.gz](https://github.com/user-attachments/files/18789970/detail_route_gcd_sky130hd_base_2025-02-13_14-40.tar.gz)
The design took a long time to route and had a lot of DRC errors, but it was the only way I could reproduce a similar set of warnings with a different PDK. The only change to this setup should be that the routing layers are being limited to li1 through met1.

### Relevant log output

```shell

```

maliberty, 2025-02-14:
Two layer processes are from a different area where channel routing was common.  Its possible you might get OR to work but it is quite uncertain.   I wouldn't add the extra m2 pin shape if it isn't strictly needed.  The tool can add vias as required.

maliberty, 2025-02-14:
Given the unusual PDK this will be a lower priority.