The-OpenROAD-Project/OpenROAD #6715: Build fails due to file conflict with gtest package

vani2620, 2025-02-14:
### Describe the bug

I'm trying to install this on EndeavourOS via AUR using the pacman wrapper yay. The building of OR-tools, abc, and OpenSTA appear to work perfectly fine, but when it gets around to building OpenROAD itself, the build fails with the following error: 

```error: failed to commit transaction (conflicting files)
openroad-git: /usr/include/gmock/gmock-actions.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gmock/gmock-cardinalities.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gmock/gmock-function-mocker.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gmock/gmock-matchers.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gmock/gmock-more-actions.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gmock/gmock-more-matchers.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gmock/gmock-nice-strict.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gmock/gmock-spec-builders.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gmock/gmock.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gmock/internal/custom/README.md exists in filesystem (owned by gtest)
openroad-git: /usr/include/gmock/internal/custom/gmock-generated-actions.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gmock/internal/custom/gmock-matchers.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gmock/internal/custom/gmock-port.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gmock/internal/gmock-internal-utils.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gmock/internal/gmock-port.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gmock/internal/gmock-pp.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/gtest-assertion-result.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/gtest-death-test.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/gtest-matchers.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/gtest-message.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/gtest-param-test.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/gtest-printers.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/gtest-spi.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/gtest-test-part.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/gtest-typed-test.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/gtest.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/gtest_pred_impl.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/gtest_prod.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/internal/custom/README.md exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/internal/custom/gtest-port.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/internal/custom/gtest-printers.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/internal/custom/gtest.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/internal/gtest-death-test-internal.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/internal/gtest-filepath.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/internal/gtest-internal.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/internal/gtest-param-util.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/internal/gtest-port-arch.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/internal/gtest-port.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/internal/gtest-string.h exists in filesystem (owned by gtest)
openroad-git: /usr/include/gtest/internal/gtest-type-util.h exists in filesystem (owned by gtest)
openroad-git: /usr/lib/cmake/GTest/GTestConfig.cmake exists in filesystem (owned by gtest)
openroad-git: /usr/lib/cmake/GTest/GTestConfigVersion.cmake exists in filesystem (owned by gtest)
openroad-git: /usr/lib/cmake/GTest/GTestTargets-none.cmake exists in filesystem (owned by gtest)
openroad-git: /usr/lib/cmake/GTest/GTestTargets.cmake exists in filesystem (owned by gtest)
openroad-git: /usr/lib/pkgconfig/gmock.pc exists in filesystem (owned by gtest)
openroad-git: /usr/lib/pkgconfig/gmock_main.pc exists in filesystem (owned by gtest)
openroad-git: /usr/lib/pkgconfig/gtest.pc exists in filesystem (owned by gtest)
openroad-git: /usr/lib/pkgconfig/gtest_main.pc exists in filesystem (owned by gtest)
Errors occurred, no packages were upgraded.
 -> error installing: [/home/vanitas/.cache/yay/openroad-git/openroad-git-r27008.6645a5370f-1-x86_64.pkg.tar.zst] - exit status 1```

I'm almost certain this is being caused by the inclusion of GoogleTest in the CMakeLists.txt in the main tree, but I don't have enough experience with C/C++ to know how to sidestep this.

### Expected Behavior

I expected the package to install properly.

### Environment

```shell
Git commit: cb7a69d788d96776b417afc2e6bf7f2e77d1a96e
kernel: Linux 6.13.2-arch1-1
os: EndeavourOS 
cmake version 3.31.5
CMake Deprecation Warning at third-party/abc/CMakeLists.txt:1 (cmake_minimum_required):
[... 205 lines cut ...]
This warning is for project developers.  Use -Wno-dev to suppress it.

h
-- Found Boost: /usr/lib/cmake/Boost-1.87.0/BoostConfig.cmake (found version "1.87.0") found components: serialization system thread
-- Found Boost: /usr/lib/cmake/Boost-1.87.0/BoostConfig.cmake (found version "1.87.0")
-- Found Eigen3: /usr/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0")
-- TCL readline disabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done (6.2s)
-- Generating done (0.6s)
-- Build files have been written to: /tmp/tmp.bxzgI4JehX
```

### To Reproduce

1. Use EndeavourOS.
2. Run `yay -S openroad-git`.
3. Press enter at every confirmation message.

### Relevant log output

```shell

```

vani2620, 2025-02-14:
I should add, I can't uninstall gtest as it appears to be a system-wide dependency.

maliberty, 2025-02-14:
@QuantamHD any ideas?

QuantamHD, 2025-02-14:
This seems like an Arch specific issue. You can turn off tests in cmake with -DENABLE_TESTS=OFF

https://wiki.archlinux.org/title/Pacman#%22Failed_to_commit_transaction_(conflicting_files)%22_error

vani2620, 2025-02-15:
Small update: I've tried editing the command in the PKGBUILD file with the specified command and no dice on that front--I still get the same error message. I'll try and see if simply syncing dependencies and then editing the CMakeLists.txt file helps any.

vani2620, 2025-02-16:
So running the build manually via the provided script worked fine...

closed 2025-08-22 by vvbandeira