The-OpenROAD-Project/OpenROAD #6724: openroad fails to start: ERROR: Inconsistency between flag object and registration for flag 'v', likely due to duplicate flags or an ODR violation.

yurivict, 2025-02-17:
### Describe the bug

```
$ openroad 
ERROR: Inconsistency between flag object and registration for flag 'v', likely due to duplicate flags or an ODR violation. Relevant files: wrkdirs/usr/ports/math/or-tools/work/or-tools-9.8/ortools/base/vlog_is_on.cc and /wrkdirs/usr/ports/devel/abseil/work/abseil-cpp-20240722.0/absl/log/flags.cc
```

Revision: 888a3207a
abseil-20240722.0
or-tools-9.8
FreeBSD 14.2

### Expected Behavior

n/a

### Environment

```shell
n/a
```

### To Reproduce

build and run

### Relevant log output

```shell

```

maliberty, 2025-02-17:
We don't build or test on freebsd. It looks like you have incompatible dependencies.  Fwiw we usually build with or-tools 9.11.4210

closed 2025-02-17 by maliberty

yurivict, 2025-02-17:
Where does "4210" come from?
https://github.com/google/or-tools/tags only has 9.11

vvbandeira, 2025-02-17:
> Where does "4210" come from? https://github.com/google/or-tools/tags only has 9.11

It is a patch version. This is just being very specific about the version. You can find the number if you expand each release's "Assets" section (see below). Any 9.11 should be feature equivalent and work for our purposes.

[image]