The-OpenROAD-Project/OpenROAD #6774: gpl: support regions for routability mode.

gudeh, 2025-02-26:
### Description

In the GPL code, NesterovBase consists of a class representing each region. We have the nbVec_ vector, which holds an object for each region. The first object in this vector (nbVec_[0]) contains all instances that do not belong to any specific region. For example, if a Verilog design is described without any defined regions, as is usually the case, GPL will hold all instances in nbVec_[0].

Currently, in routeBase.cpp, whenever we reference a NesterovBase object, we only do so for the object without regions (nbVec_[0]), while all other objects are ignored. For example:
https://github.com/The-OpenROAD-Project/OpenROAD/blob/6c60401f98c6ec75633b3aa892456818e55fdeb2/src/gpl/src/routeBase.cpp#L763

I have not yet tested routability mode on a design with regions, but there is no way it would function correctly. However, I believe it will not necessarily throw an error.

### Suggested Solution

- Test what currently happens on a design with regions (e.g., upf_aes in OpenROAD/test/).
- Most importantly, replace every instance of nbVec_[0] in routeBase.cpp with a loop iterating over nbVec_ objects, and move forward from there.

maliberty, 2025-03-26:
Note the grt or rudy do not operate one a single region but on all regions at once.  You need to apply the results by region.

gudeh, 2025-08-28:
Me @eder-matheus and @LucasYuki had a meeting and thought of having @LucasYuki work on this task.

gudeh, 2025-09-04:
> What do you mean by "Fillers (in purple) seem to be removed from two regions only."?
> 
> It would be nice to draw the region boundaries in the images though you can infer them by the end.

Notice how purple instances are removed on the beginning of the animation, but only on the left regions (top and bottom). The region on the right does not seem to have fillers removed.

This is the state at the first time we reach 0.30 overflow:
[image]

And this 0.30 overflow again but after the first revert, inflations applied (given by instances painted in white), and fillers removed:

[image]

gudeh, 2025-09-22:
The last issue mentioned with the fillers removal should be solved with PR https://github.com/The-OpenROAD-Project/OpenROAD/pull/8296