The-OpenROAD-Project/OpenROAD #6775: gia_test error: boost 1.87 removed depreciated asio::ip::address::from_string

stefanottili, 2025-02-26:
### Describe the bug

compile fails due to boost 1.87 removing the depreciated asio::ip::address::from_string
replacing address::from_string with make_address solves the issue.
```
OpenROAD/src/utl/src/prometheus/metrics_server.cpp:73
boost::asio::ip::address::from_string("127.0.0.1"), port_);
=> boost::asio::ip::make_address("127.0.0.1", port_);
```

### Expected Behavior

compile using boost 1.87

### Environment

```shell
macOS M1 homebrew, boost 1.87
```

### To Reproduce

./build_openroad.sh --local --latest

### Relevant log output

```shell
[ 61%] Built target gia_test
OpenROAD-flow-scripts/tools/OpenROAD/src/utl/src/prometheus/metrics_server.cpp:73:37: error: no member named 'from_string' in 'boost::asio::ip::address'
   73 |           boost::asio::ip::address::from_string("127.0.0.1"), port_);
      |           ~~~~~~~~~~~~~~~~~~~~~~~~~~^
1 error generated.
```

maliberty, 2025-02-26:
We updated to 1.86 not 1.87 which has other bugs (see https://github.com/boostorg/asio/issues/442).  Even if fixed 1.87 is unlikely to work.

It looks like a simple fix though.  Do you want to make a PR?

stefanottili, 2025-03-05:
Fixed by https://github.com/The-OpenROAD-Project/OpenROAD/pull/6795