The-OpenROAD-Project/OpenROAD #6817: Signal 11 received RESYNTH_TIMING_RECOVER=1

oharboe, 2025-03-06:
### Describe the bug

untar and run [floorplan_gcd_asap7_base_2025-03-06_07-32.tar.gz](https://github.com/user-attachments/files/19102704/floorplan_gcd_asap7_base_2025-03-06_07-32.tar.gz)

```
$ ./run-me-gcd-asap7-base.sh  
[deleted]
Stack trace:
 0# 0x00005EF87FE8C883 in openroad
 1# 0x00007FD32DE45330 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbObject::getObjectType() const in openroad
 3# sta::dbNetwork::staToDb(sta::Pin const*, odb::dbITerm*&, odb::dbBTerm*&, odb::dbModITerm*&, odb::dbModBTerm*&) const in openroad
 4# sta::dbNetwork::instance(sta::Pin const*) const in openroad
 5# sta::Network::isTopLevelPort(sta::Pin const*) const in openroad
 6# rsz::RepairDesign::repairDesign(double, double, double, double, bool, int&, int&, int&, int&, int&) in openroad
 7# rsz::RepairDesign::repairDesign(double, double, double, double, bool) in openroad
 8# rsz::Resizer::repairDesign(double, double, double, double, bool, bool) in openroad
 9# 0x00005EF880069263 in openroad
10# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
11# 0x00007FD332DC66BC in /lib/x86_64-linux-gnu/libtcl8.6.so
12# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
13# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
14# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
15# 0x00005EF87FE8BFCF in openroad
16# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
17# main in openroad
18# 0x00007FD32DE2A1CA in /lib/x86_64-linux-gnu/libc.so.6
19# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
20# _start in openroad
./run-me-gcd-asap7-base.sh: line 7: 2637787 Segmentation fault      (core dumped) openroad -no_init ${SCRIPTS_DIR}/floorplan.tcl
```

### Expected Behavior

No crash

### Environment

```shell
v2.0-18763-g357ef5e4a
```

### To Reproduce

See above

### Relevant log output

```shell

```

maliberty, 2025-03-06:
We should fix the crash but I don't think those flow options are that useful.  It was preliminary work on remapping and has never reached maturity.   Hopefully with @QuantamHD 's work it will become useful.

oharboe, 2025-03-07:
> We should fix the crash but I don't think those flow options are that useful.  It was preliminary work on remapping and has never reached maturity.   Hopefully with @QuantamHD 's work it will become useful.

I am not using the flow, I was just tinkering with merged.lib and saw that this code path used it.