The-OpenROAD-Project/OpenROAD #6836: grt/drt: [ERROR DRT-0073] No access point for ces_0_0/io_ins_left[12].

gudeh, 2025-03-12:
### Describe the bug

Error happens with branch [gpl-binSize-float](https://github.com/The-OpenROAD-Project/OpenROAD/pull/6530) on asap7/mock-array on [secure-CI](https://jenkins.openroad.tools/blue/organizations/jenkins/OpenROAD-flow-scripts-Private/detail/secure-gpl-binSize-float/10/). The branch results in a small change in values along the flow starting from gpl, comparing the logs versus master I did not see any big difference in metrics.

Logs show a bunch of these warnings:
```
[2025-03-11T10:23:07.780Z] [WARNING DRT-0419] No routing tracks pass through the center of Term ces_2_3/io_outs_right[46]
[2025-03-11T10:23:07.780Z] [WARNING DRT-0418] Term ces_2_3/io_outs_right[47] has no pins on routing grid
[2025-03-11T10:23:07.780Z] [WARNING DRT-0418] Term ces_2_3/io_outs_right[48] has no pins on routing grid
[2025-03-11T10:23:07.780Z] [WARNING DRT-0418] Term ces_2_3/io_outs_right[51] has no pins on routing grid
[2025-03-11T10:23:07.780Z] [WARNING DRT-0419] No routing tracks pass through the center of Term ces_2_3/io_outs_right[59]
[2025-03-11T10:23:07.780Z] [WARNING DRT-0418] Term ces_2_3/io_outs_right[5] has no pins on routing grid
[2025-03-11T10:23:07.780Z] [WARNING DRT-0419] No routing tracks pass through the center of Term ces_2_4/io_ins_left[12]
[2025-03-11T10:23:07.780Z] [WARNING DRT-0419] No routing tracks pass through the center of Term ces_2_4/io_ins_left[14]
[2025-03-11T10:23:07.780Z] [WARNING DRT-0418] Term ces_2_4/io_ins_left[16] has no pins on routing grid
[2025-03-11T10:23:07.780Z] [WARNING DRT-0419] No routing tracks pass through the center of Term ces_2_4/io_ins_left[18]
[2025-03-11T10:23:07.780Z] [WARNING DRT-0419] No routing tracks pass through the center of Term ces_2_4/io_ins_left[22]
[2025-03-11T10:23:07.780Z] [WARNING DRT-0418] Term ces_2_4/io_ins_left[24] has no pins on routing grid
[2025-03-11T10:23:07.780Z] [WARNING DRT-0419] No routing tracks pass through the center of Term ces_2_4/io_ins_left[27]
[2025-03-11T10:23:07.780Z] [WARNING DRT-0418] Term ces_2_4/io_ins_left[28] has no pins on routing grid
[2025-03-11T10:23:07.780Z] [WARNING DRT-0419] No routing tracks pass through the center of Term ces_2_4/io_ins_left[2]
[2025-03-11T10:23:07.780Z] [WARNING DRT-0419] No routing tracks pass through the center of Term ces_2_4/io_ins_left[31]
[2025-03-11T10:23:07.780Z] [WARNING DRT-0418] Term ces_2_4/io_ins_left[32] has no pins on routing grid

```

and eventually this error:
```
[2025-03-11T10:23:08.355Z] [INFO DRT-0033] M7 shape region query size = 858.
[2025-03-11T10:23:08.355Z] [INFO DRT-0033] V7 shape region query size = 0.
[2025-03-11T10:23:08.355Z] [INFO DRT-0033] M8 shape region query size = 0.
[2025-03-11T10:23:08.355Z] [INFO DRT-0033] V8 shape region query size = 0.
[2025-03-11T10:23:08.355Z] [INFO DRT-0033] M9 shape region query size = 0.
[2025-03-11T10:23:08.355Z] [INFO DRT-0033] V9 shape region query size = 0.
[2025-03-11T10:23:08.355Z] [INFO DRT-0033] Pad shape region query size = 0.
[2025-03-11T10:23:08.355Z] [INFO DRT-0165] Start pin access.
[2025-03-11T10:23:08.922Z] [ERROR DRT-0073] No access point for ces_1_0/io_ins_left[12].
[2025-03-11T10:23:08.922Z] [ERROR DRT-0073] No access point for ces_0_0/io_ins_left[12].
[2025-03-11T10:23:08.922Z] Error: global_route.tcl, 114 DRT-0073
[2025-03-11T10:23:08.922Z] Command exited with non-zero status 1
[2025-03-11T10:23:08.922Z] Elapsed time: 0:05.85[h:]min:sec. CPU time: user 6.97 sys 0.48 (127%). Peak memory: 466288KB.
```
Some images from make gui_cts:
The Iterm:
[image]
The net:
[image]
The macro:
[image]

In master:
[image]

### Expected Behavior

GRT and DRT.

### Environment

```shell
[INFO] Using 16 threads.
-- OpenROAD version: v2.0-19611-g07756ab870
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /home/gudeh/6ORFS/tools/install/OpenROAD
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Performing Test C_COMPILER_SUPPORTS__-std=c++17
[... 511 lines cut ...]
[100%] Linking CXX static library _rmp_py.a
[100%] Built target rmp_py
[100%] Linking CXX static library gpl.a
[100%] Built target gpl
Consolidate compiler generated dependencies of target openroad
[100%] Building CXX object src/CMakeFiles/openroad.dir/Design.cc.o
[100%] Building CXX object src/CMakeFiles/openroad.dir/Timing.cc.o
[100%] Building CXX object src/CMakeFiles/openroad.dir/Tech.cc.o
[100%] Building CXX object src/CMakeFiles/openroad.dir/OpenRoad.cc.o
[100%] Building CXX object src/CMakeFiles/openroad.dir/Main.cc.o
[100%] Linking CXX executable trTest
[100%] Built target trTest
[100%] Linking CXX executable openroad
[100%] Built target openroad
```

### To Reproduce

Run `make DESIGN_CONFIG=./designs/asap7/mock-array/config.mk` with branch `gpl-binSize-float`

### Relevant log output

```shell

```

eder-matheus, 2025-03-12:
@maliberty Even if we skip the nets connected by abutment, we have other macro pins in M2 not aligned with routing tracks. This is probably caused by the PPL update to use the tracks of all track patterns in ASAP7.

I notice that in @gudeh run, the pin placement of mock-array_Element has some changes (caused by the different global placement), and MPL is not snapping the pins to the routing grid correctly. Below is a comparison between using master and using this branch.

master:
[image]

gpl-binSize-float:
[image]

Note that these pins are not connected by abutment, so DRT would have trouble with them. I guess this is a good test case for @joaomai regarding the snapping of macros in MPL.

gudeh, 2025-03-18:
I see this on a recent merge with master. I am not sure if it is related.

```
14:22:32  asap7 mock-cpu (base)

14:22:32    Last log file 3_2_place_iop.log

14:22:32    Found 1 errors in the logs.

14:22:32        [ERROR PPL-0104] Pins  clk clk_uncore rdata[0] rdata[10] rdata[11] rdata[12] rdata[13] rdata[14] rdata[15] rdata[16] rdata[17] rdata[18] rdata[19] rdata[1] rdata[20] rdata[21] rdata[22] rdata[23] rdata[24] rdata[25] rdata[26] rdata[27] rdata[28] rdata[29] rdata[2] rdata[30] rdata[31] rdata[3] rdata[4] rdata[5] rdata[6] rdata[7] rdata[8] rdata[9] rempty rinc rrst_n wdata[0] wdata[10] wdata[11] wdata[12] wdata[13] wdata[14] wdata[15] wdata[16] wdata[17] wdata[18] wdata[19] wdata[1] wdata[20] wdata[21] wdata[22] wdata[23] wdata[24] wdata[25] wdata[26] wdata[27] wdata[28] wdata[29] wdata[2] wdata[30] wdata[31] wdata[3] wdata[4] wdata[5] wdata[6] wdata[7] wdata[8] wdata[9] wfull winc wrst_n are assigned to multiple groups.
```

From this run: https://jenkins.openroad.tools/job/OpenROAD-flow-scripts-Public/job/PR-2947-merge/2/pipeline-console/?selected-node=2288 in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2947

eder-matheus, 2025-03-18:
> I see this on a recent merge with master. I am not sure if it is related.

@gudeh This is a different design, and was caused by recent changes in PPL. I'm updating its scripts to use the pin groups properly.

gudeh, 2025-03-18:
> > I see this on a recent merge with master. I am not sure if it is related.
> 
> [@gudeh](https://github.com/gudeh) This is a different design, and was caused by recent changes in PPL. I'm updating its scripts to use the pin groups properly.

I realized this is happening on master too.

closed 2025-05-01 by AcKoucher