The-OpenROAD-Project/OpenROAD #6854: Make error message more acctionable [ERROR PDN-0351] Ring shape falls outside the die bounds.

oharboe, 2025-03-15:
### Description

Untar and run [pdn-error-message.tar.gz](https://github.com/user-attachments/files/19260387/pdn-error-message.tar.gz)

```
OpenROAD v2.0-19579-g831008f71
[deleted]
[ERROR PDN-0351] Ring shape falls outside the die bounds.
Error: BLOCKS_grid_strategy.tcl, 21 PDN-0351
```

### Suggested Solution

This error tells me something about what is wrong, but I don't know why macro placement placed the macro where it did and what I should do about it.

maliberty, 2025-03-15:
@gadfort

gadfort, 2025-03-15:
@oharboe macro placement has nothing to do with this.  The issue here appears to be the ring you are adding requires a core margin of atleast: 1.688um

oharboe, 2025-03-15:
> @oharboe macro placement has nothing to do with this.  The issue here appears to be the ring you are adding requires a core margin of atleast: 1.688um

I see. Can the error message give more explicit actionable advice or is this as good as it gets?

gadfort, 2025-03-15:
@maliberty previously suggested: 
```
PDN rings do not fit inside the die area by <val> um in X and <val2> um in Y.  Either reduce the ring area or increase the core to die spacing to accommodate.  Use -allow_out_of_die if this is intentional.
```
https://github.com/The-OpenROAD-Project/OpenROAD/issues/6618#issuecomment-2625367933

closed 2025-03-16 by maliberty

oharboe, 2025-03-16:
@gadfort Thanks!

PR #6855 (2025-03-16): pdn: add error about ring area
Closes #6854