The-OpenROAD-Project/OpenROAD #6869: Legacy build for CentOS7

gogito, 2025-03-18:
### Description

I'm looking to build OpenRoad for CentOS 7 but it  seems like the current repo no longer support it. It would be great if we provide instruction on how to build for older distro. For example, clone git back to a specific commit.

maliberty, 2025-03-18:
There is no record of a specific commit that used to work.  You'll have to hunt the history yourself to find something that works.  I would guess 6-12 months ago.  You are better off running docker and being able to stay at head.

closed 2025-03-18 by maliberty

gogito, 2025-03-18:
> There is no record of a specific commit that used to work. You'll have to hunt the history yourself to find something that works. I would guess 6-12 months ago. You are better off running docker and being able to stay at head.

I actually hunted and tried myself and got a commit around  March 2023 that still support CentOS7. However the scripts no longer works fully with outdated links, ...

Is there any chance we can support CentOS 7? The reason I cannot use docker is because this install is going to be for multi user and docker just doesn't  play well with that from what I know.

maliberty, 2025-03-18:
No, we have dropped centos7.  It is too outdated