The-OpenROAD-Project/OpenROAD #6874: [ERROR DRT-0073] No access point for lb_32x128_inst/R0_addr[0].

oharboe, 2025-03-19:
### Describe the bug

Standalone reproduction, untar and run [global_route_lb_32x128_top_asap7_base_2025-03-19_09-26.tar.gz](https://github.com/user-attachments/files/19336259/global_route_lb_32x128_top_asap7_base_2025-03-19_09-26.tar.gz)

Discovered in https://github.com/The-OpenROAD-Project/bazel-orfs/pull/302

Run:

```
bazel build //:lb_32x128_top_grt
```

### Expected Behavior

No error

### Environment

```shell
v2.0-19739-g12d2d36d0
```

### To Reproduce

See above

### Relevant log output

```shell

```

eder-matheus, 2025-03-19:
@oharboe I will look into the details of this case, but FYI, the support for `PLACE_PINS_ARGS` **still existis**. My recent PR only removes the `-exclude` from these args, but other arguments like min distance still have support, so you can use this env var with no problems.

oharboe, 2025-03-19:
@eder-matheus See breadcrumb instructions above to run bazel-orfs. @jeffng-or can show you how to do this locally and run make-issue ultimately. Some spelunking involved...

I believe what you need is the io_placement issue for the macro?

Untar and run this: 

[io_placement_lb_32x128_asap7_base_2025-03-19_17-51.tar.gz](https://github.com/user-attachments/files/19347692/io_placement_lb_32x128_asap7_base_2025-03-19_17-51.tar.gz)

To reproduce

```
bazel run lb_32x128_place_deps /tmp/xxx
/tmp/xxx/make do-place
/tmp/xxx/make io_placement_issue
mkdir foo
cd foo
tar --strip-components=1 -xzvf /tmp/xxx/io_placement_lb_32x128_asap7_base_2025-03-19_17-51.tar.gz 
$ ./run-me-lb_32x128-asap7-base.sh 
OpenROAD v2.0-19779-gd4bb9505a 
Features included (+) or not (-): +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
place_pins -hor_layers M4 -ver_layers M5 -min_distance 1 -min_distance_in_tracks
Found 0 macro blocks.
[INFO PPL-0067] Restrict pins [ R0_addr[0] R0_addr[1] R0_addr[2] R0_addr[3] R0_addr[4] ... ] to region 0.00u-20.71u at the LEFT edge.
[INFO PPL-0001] Number of available slots 1218
[INFO PPL-0002] Number of I/O             270
[INFO PPL-0003] Number of I/O w/sink      258
[INFO PPL-0004] Number of I/O w/o sink    12
[INFO PPL-0005] Slots per section         200
[INFO PPL-0008] Successfully assigned pins to sections.
[INFO PPL-0012] I/O nets HPWL: 1600.89 um.
openroad> 
```

eder-matheus, 2025-03-20:
@oharboe Something is really of here. If I run the artifacts from the IO placement run that generates the macro, I get a reasonable pin placement, respecting the constraints:
[image]

However, when I look at the first artifact you attached, from the global routing run of the top level design, the pins are not present in the macro:
[image]

The yellow flyline corresponds to a net connected to one of the macro pins. This happens to every pin, because they don't have geometries defined.

Looking at the lef file  `results/asap7/lb_32x128/mocked/lb_32x128.lef` (which I believe is the one that describes the macro), the pins don't have geometries:
[image]

This happens for every pin except the the VDD/VSS pins.

eder-matheus, 2025-03-20:
I found some possible explanation for this. During the macro flow, the command `write_abstract_lef` is called twice. The first time is right after CTS. The odd thing is that it seems to load the floorplan files again to write the lef:

```
[...]
Elapsed time: 0:04.12[h:]min:sec. CPU time: user 3.72 sys 0.41 (100%). Peak memory: 573576KB.
**cp bazel-out/k8-fastbuild/bin/results/asap7/lb_32x128/base/4_1_cts.odb bazel-out/k8-fastbuild/bin/results/asap7/lb_32x128/base/4_cts.odb**
OpenROAD HEAD-HASH-NOTFOUND
Features included (+) or not (-): +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ORD-0030] Using 48 thread(s).
mkdir -p bazel-out/k8-fastbuild/bin/objects/asap7/lb_32x128/mocked
Running floorplan.tcl, stage 2_1_floorplan
[INFO ODB-0227] LEF file: external/_main~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/platforms/asap7/lef/asap7_tech_1x_201209.lef, created 24 layers, 9 vias
[INFO ODB-0227] LEF file: external/_main~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/platforms/asap7/lef/asap7sc7p5t_28_R_1x_220121a.lef, created 212 library cells

==========================================================================
**Floorplan check_setup**
--------------------------------------------------------------------------
Warning: There are 140 input ports missing set_input_delay.
number instances in verilog is 129
[INFO IFP-0001] Added 45 rows of 108 site asap7sc7p5t.
[INFO RSZ-0026] Removed 1 buffers.
Default units for flow
 time 1ps
 capacitance 1fF
 resistance 1kohm
 voltage 1v
 current 1mA
 power 1pW
 distance 1um
Elapsed time: 0:01.66[h:]min:sec. CPU time: user 1.53 sys 0.14 (100%). Peak memory: 213500KB.
Running macro_place.tcl, stage 2_2_floorplan_macro
No macros found: Skipping macro_placement
Elapsed time: 0:01.83[h:]min:sec. CPU time: user 1.55 sys 0.13 (91%). Peak memory: 209280KB.
Running tapcell.tcl, stage 2_3_floorplan_tapcell
Elapsed time: 0:01.52[h:]min:sec. CPU time: user 1.33 sys 0.15 (97%). Peak memory: 208896KB.
Running pdn.tcl, stage 2_4_floorplan_pdn
[INFO PDN-0001] Inserting grid: top
Elapsed time: 0:01.44[h:]min:sec. CPU time: user 1.34 sys 0.11 (101%). Peak memory: 211584KB.
cp bazel-out/k8-fastbuild/bin/results/asap7/lb_32x128/mocked/2_4_floorplan_pdn.odb bazel-out/k8-fastbuild/bin/results/asap7/lb_32x128/mocked/2_floorplan.odb
cp bazel-out/k8-fastbuild/bin/results/asap7/lb_32x128/mocked/2_1_floorplan.sdc bazel-out/k8-fastbuild/bin/results/asap7/lb_32x128/mocked/2_floorplan.sdc
mkdir -p bazel-out/k8-fastbuild/bin/logs/asap7/lb_32x128/mocked bazel-out/k8-fastbuild/bin/reports/asap7/lb_32x128/mocked
(env time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' external/_main~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init -threads 48  external/_main~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/scripts/generate_abstract.tcl -metrics bazel-out/k8-fastbuild/bin/logs/asap7/lb_32x128/mocked/generate_abstract.json) 2>&1 | tee /home/oyvind/.cache/bazel/_bazel_oyvind/6e078b44989a9ce1500a63165a20fd2a/sandbox/processwrapper-sandbox/12/execroot/_main/bazel-out/k8-fastbuild/bin/logs/asap7/lb_32x128/mocked/generate_abstract.log
OpenROAD HEAD-HASH-NOTFOUND
Features included (+) or not (-): +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ORD-0030] Using 48 thread(s).
**load_design 2_floorplan.odb 2_floorplan.sdc**
Generating abstract views
write_timing_model bazel-out/k8-fastbuild/bin/results/asap7/lb_32x128/mocked/lb_32x128.lib
**write_abstract_lef -bloat_occupied_layers bazel-out/k8-fastbuild/bin/results/asap7/lb_32x128/mocked/lb_32x128.lef**
Elapsed time: 0:01.44[h:]min:sec. CPU time: user 1.32 sys 0.11 (100%). Peak memory: 210044KB.
[...]
```
In this first time, it writes the DEF to the `mocked` directory.

Then it runs it again in the end of the flow for the macro:
```
[...]
[INFO] No orphan cells
[INFO] Writing out GDS/OAS 'bazel-out/k8-fastbuild/bin/results/asap7/lb_32x128/base/6_1_merged.gds'
Elapsed time: 0:01.13[h:]min:sec. CPU time: user 0.97 sys 0.15 (99%). Peak memory: 398432KB.
cp bazel-out/k8-fastbuild/bin/results/asap7/lb_32x128/base/6_1_merged.gds bazel-out/k8-fastbuild/bin/results/asap7/lb_32x128/base/6_final.gds
bazel-out/k8-fastbuild/bin/logs/asap7/lb_32x128/base
Log                            Elapsed seconds Peak Memory/MB
6_1_fill                                     1            206
6_1_merge                                    1            389
6_report                                     4            276
[... 2 lines cut ...]
(env time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' external/_main~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init -threads 48  external/_main~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/scripts/generate_abstract.tcl -metrics bazel-out/k8-fastbuild/bin/logs/asap7/lb_32x128/base/generate_abstract.json) 2>&1 | tee /home/oyvind/.cache/bazel/_bazel_oyvind/6e078b44989a9ce1500a63165a20fd2a/sandbox/processwrapper-sandbox/17/execroot/_main/bazel-out/k8-fastbuild/bin/logs/asap7/lb_32x128/base/generate_abstract.log
[... cut ...]
[INFO ODB-0227] LEF file: external/_main~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/platforms/asap7/lef/asap7sc7p5t_28_R_1x_220121a.lef, created 212 library cells
[INFO ODB-0227] LEF file: bazel-out/k8-fastbuild/bin/results/asap7/lb_32x128/mocked/lb_32x128.lef, created 5 vias, 1 library cells
[...]
```
This explains why the macro don't have the pin placement. In fact, this macro doesn't have nothing but the PDN. I don't understand bazel-orfs enough to detect why it is using the "wrong" LEF, or why are we writing the LEF from the floorplan files. But I guess it is not a PPL issue.

oharboe, 2025-03-20:
@eder-matheus @maliberty Aha! Previously it was possible to create an abstract from floorplan. This was useful for mock abstracts and fast turnaround time when smoke-testing macro placement. This is no longer possible out of the box with the ORFS flow.

I've reworked bazel-orfs to produce an error message if the user attempts this. I'll open a fresh github issue if there are any further issues.

closed 2025-03-20 by oharboe