The-OpenROAD-Project/OpenROAD #6883: Crash or Segmentation Fault when reading DEF file

rixmaestro, 2025-03-20:
### Describe the bug

When reading a DEF file, the software crashes.

Scenario: technology DATABASE MICRONS is 1000
If you change coordinates of the rectangle such that the dimensions of both axes are not divisible by 10, then the software crashes. If I change the DATABASE MICRONS to 2000, the software can read the DEF file successfully.

A sample stack trace:

OpenROAD v2.0-19779-gd4bb9505a
Features included (+) or not (-): +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
openroad> read_lef sample_tech.lef
read_lef cell.lef
read_def sample_def.def
[INFO ODB-0227] LEF file: sample_tech.lef, created 3 layers
[INFO ODB-0227] LEF file: cell.lef, created 1 library cells
[INFO ODB-0127] Reading DEF file: sample_def.def
[INFO ODB-0128] Design: test_def
Signal 11 received
Stack trace:
 0# 0x0000563C63775E82 in ../tools/install/OpenROAD/bin/openroad
 1# 0x00007F1799945520 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbBox::setLayerMask(unsigned int) in ../tools/install/OpenROAD/bin/openroad
 3# odb::definSNet::rect(char const*, int, int, int, int, char const*, unsigned int) in ../tools/install/OpenROAD/bin/openroad
 4# odb::definReader::specialNetCallback(DefParser::defrCallbackType_e, DefParser::defiNet*, void*) in ../tools/install/OpenROAD/bin/openroad
 5# DefParser::defyyparse(DefParser::defrData*) in ../tools/install/OpenROAD/bin/openroad
 6# DefParser::defrRead(_IO_FILE*, char const*, void*, int) in ../tools/install/OpenROAD/bin/openroad
 7# odb::definReader::createBlock(char const*) in ../tools/install/OpenROAD/bin/openroad
 8# odb::definReader::createChip(std::vector<odb::dbLib*, std::allocator<odb::dbLib*> >&, char const*, odb::dbTech*) in ../tools/install/OpenROAD/bin/openroad
 9# odb::defin::createChip(std::vector<odb::dbLib*, std::allocator<odb::dbLib*> >&, char const*, odb::dbTech*) in ../tools/install/OpenROAD/bin/openroad
10# ord::OpenRoad::readDef(char const*, odb::dbTech*, bool, bool, bool, bool) in ../tools/install/OpenROAD/bin/openroad
11# read_def_cmd(char const*, char const*, bool, bool, bool, bool) in ../tools/install/OpenROAD/bin/openroad
12# 0x0000563C6380236D in ../tools/install/OpenROAD/bin/openroad
13# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
14# 0x00007F179DE97028 in /lib/x86_64-linux-gnu/libtcl8.6.so
15# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
16# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
17# 0x0000563C637752E0 in ../tools/install/OpenROAD/bin/openroad
18# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
19# main in ../tools/install/OpenROAD/bin/openroad
20# 0x00007F179992CD90 in /lib/x86_64-linux-gnu/libc.so.6
21# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
22# _start in ../tools/install/OpenROAD/bin/openroad
Segmentation fault (core dumped)

### Expected Behavior

The expected behavior is for the tool to successfully read the DEF file.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.167.4-microsoft-standard-WSL2
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.22.1
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-19779-gd4bb9505a
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
[... 110 lines cut ...]

### To Reproduce

[read_def_crash.tar.gz](https://github.com/user-attachments/files/19369029/read_def_crash.tar.gz)

read_lef sample_tech.lef
read_lef cell.lef
read_def sample_def.def

If you replace the DATABASE MICRONS to 2000, the software will not crash and it will read the DEF file correctly.

With DATABASE MICRONS 1000, if you change coordinates of the rectangle such that the dimensions of both axes are not divisible by 10, the software will crash:

-  ( 4750 1145 )  ( 4785 1480 )
-  ( 4755 1140 )  ( 4780 1485 )

### Relevant log output

```shell

```

### Additional Context

[build_openroad.log](https://github.com/user-attachments/files/19369329/build_openroad.log)

maliberty, 2025-03-20:
The problem is that the wire width is odd which is not allowed in odb (we should error rather than crash but that is another issue).  Switch to 2000 makes the width double to an even value.  Did you intend to have an odd width?

Thanks for making a nice small test case!

rixmaestro, 2025-03-21:
Thanks for the explanation! I did not know it is not allowed. Currently, I intend to have odd widths (for consistency/accuracy), but it won't be a huge problem if I round things up to make even widths.