The-OpenROAD-Project/OpenROAD #6887: Installation on Ubuntu 24.04

ant1sir, 2025-03-20:
### Describe the bug

Hello!
Congratulations for these fantastic achievements!
I was able to install OpenRoad and ORFS on Ubuntu 24.04 yesterday building upon the source code.
I met a few minor issues.
- When fetching the repo, abc and sta do not come along.
- When executing etc/DependencyInstaller.sh, I get a minor error due to the last lines of the script.

`if [[ ! -z ${saveDepsPrefixes} ]]; then`
`    mkdir -p "$(dirname $saveDepsPrefixes)"`
`    echo "$CMAKE_PACKAGE_ROOT_ARGS" > $saveDepsPrefixes`
`fi`
```
It seems that the variable **saveDepsPrefixes** does not contain a file name.
So the script ends with an error at the end of the installation process.

Enjoy the Spring :-)

Best regards,

### Expected Behavior

Seamless installation or Updated documentation
I know that Ubuntu 24.04 is not officially supported yet.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.8.0-55-generic
os: Ubuntu 24.04.2 LTS (Noble Numbat)
cmake version 3.24.2
-- The CXX compiler identification is GNU 13.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-19804-gcae3d89c2
-- System name: Linux
-- Compiler: GNU 13.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 13.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
[... 115 lines cut ...]
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /home/faradomus/.local/lib/cmake/Boost-1.86.0/BoostConfig.cmake (found suitable version "1.86.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /home/faradomus/.local/lib/cmake/Boost-1.86.0/BoostConfig.cmake (found version "1.86.0") found components: serialization system thread 
-- Found Boost: /home/faradomus/.local/lib/cmake/Boost-1.86.0/BoostConfig.cmake (found version "1.86.0")  
-- Found Eigen3: /home/faradomus/.local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.npeipNA4JZ
```

### To Reproduce

Install OpenRoad on Ubuntu 24.04 from scratch upon the source files following the current documentation.
I know, this is a bit awkward.

### Relevant log output

```shell

```

maliberty, 2025-03-20:
> When fetching the repo, abc and sta do not come along.

That is normal git and the build script will update the submodules (or you can manually do `git submodule update --init --recursive`)

@vvbandeira any comments the other issue?

vvbandeira, 2025-03-20:
> * When fetching the repo, abc and sta do not come along.

The documentation includes the use of `--recursive` for this purpose:
https://openroad.readthedocs.io/en/latest/user/Build.html

> * When executing etc/DependencyInstaller.sh, I get a minor error due to the last lines of the script.

Could you paste the error so we can better understand the issue? The variable mentioned always has either the value given by the user or the default value set at the following lines. So the issue is not with the variable not being set.

https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/etc/DependencyInstaller.sh#L822-L825

ant1sir, 2025-03-21:
@wbandeira. Here you are. The --recursive option did work :-)

# OpenROAD#6887

## Reproducing the artefact

2025-03-21 08:50:00 CET

1) git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git

> Submodule path 'src/sta': checked out '5a46b3ceb2ad40eddcf7f118093a24258240c74c'
> Submodule path 'third-party/abc': checked out '5c9448c085eb8bf1e433a22a954532e44206c6f9'

The documentation says:

OpenROAD git submodules (cloned by the --recursive flag) are located in src/.

Actually, abc is located in third-party. It may not be considered as a submodule :-)

2) cd OpenROAD

3) sudo ./etc/DependencyInstaller.sh -base

4) ./etc/DependencyInstaller.sh -common -local

2025-03-21 09:02:23 (34.5 MB/s) - ‘or-tools_amd64_ubuntu-22.04_cpp_v9.11.4210.tar.gz’ saved [49050271/49050271]

./etc/DependencyInstaller.sh: line 953: <HOME>/Projects/Sandbox/OpenRoad#6887/OpenROAD/etc/openroad_deps_prefixes.txt: Permission denied

vvbandeira, 2025-03-24:
looks like a duplicate of https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/2904

closed 2025-03-24 by maliberty