The-OpenROAD-Project/OpenROAD #6895: CentOS 8 Builds

gogito, 2025-03-23:
### Description

Hi everyone, I tried to revert to an older build of the repo (that still had CentOS 8 support) in order to compile it for CentOS8. I managed to get most of the dependency, except for devtoolset8 and centos-release-scl.

However, when running etc/Build.sh I get this error

[ 76%] Building CXX object src/gui/CMakeFiles/gui.dir/gui_autogen/mocs_compilation.cpp.o
[ 76%] Building CXX object src/grt/src/fastroute/CMakeFiles/FastRoute4.1.dir/src/FastRoute.cpp.o
/home/gogito/OpenROAD/src/dbSta/src/dbReadVerilog.cc:80:12: error: ‘sta::deleteVerilogReader’ has not been declared
 using sta::deleteVerilogReader;
            ^~~~~~~~~~~~~~~~~~~
/home/gogito/OpenROAD/src/dbSta/src/dbReadVerilog.cc: In function ‘void ord::dbReadVerilog(const char*, ord::dbVerilogNetwork*)’:
/home/gogito/OpenROAD/src/dbSta/src/dbReadVerilog.cc:140:8: error: ‘readVerilogFile’ is not a member of ‘sta’
   sta::readVerilogFile(filename, verilog_network);
        ^~~~~~~~~~~~~~~
/home/gogito/OpenROAD/src/dbSta/src/dbReadVerilog.cc:140:8: note: suggested alternative: ‘VerilogDcl’
   sta::readVerilogFile(filename, verilog_network);
        ^~~~~~~~~~~~~~~
        VerilogDcl
/home/gogito/OpenROAD/src/dbSta/src/dbReadVerilog.cc: In function ‘void ord::dbLinkDesign(const char*, ord::dbVerilogNetwork*, odb::dbDatabase*, utl::Logger*, bool)’:
/home/gogito/OpenROAD/src/dbSta/src/dbReadVerilog.cc:213:5: error: ‘deleteVerilogReader’ was not declared in this scope
     deleteVerilogReader();
     ^~~~~~~~~~~~~~~~~~~
/home/gogito/OpenROAD/src/dbSta/src/dbReadVerilog.cc:213:5: note: suggested alternative: ‘deleteDbVerilogNetwork’
     deleteVerilogReader();
     ^~~~~~~~~~~~~~~~~~~
     deleteDbVerilogNetwork
gmake[2]: *** [src/dbSta/src/CMakeFiles/dbSta_lib.dir/build.make:118: src/dbSta/src/CMakeFiles/dbSta_lib.dir/dbReadVerilog.cc.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:4175: src/dbSta/src/CMakeFiles/dbSta_lib.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
[ 76%] Building CXX object src/odb/src/swig/python/CMakeFiles/odb_py.dir/odb_py-pythonInitVar.cc.o
[ 76%] Building CXX object src/grt/src/fastroute/CMakeFiles/FastRoute4.1.dir/src/RSMT.cpp.o
[ 76%] Building CXX object src/grt/src/fastroute/CMakeFiles/FastRoute4.1.dir/src/RipUp.cpp.o
[ 76%] Building CXX object src/grt/src/fastroute/CMakeFiles/FastRoute4.1.dir/src/maze.cpp.o
[ 76%] Building CXX object src/grt/src/fastroute/CMakeFiles/FastRoute4.1.dir/src/maze3D.cpp.o

I'm not a C dev so this kinda stump me after many hours of trial. Can anyone help or have a build of OpenRoad available for CentOS8 built already? It would be great if we can provide these builds due to many system still running  older versions of CentOS (for compatibility issue).

maliberty, 2025-03-23:
@vvbandeira do build on centos8 or just 9?

vvbandeira, 2025-03-24:
CentOS 8 has reached EOL, we only build on RHEL/RockyLinux9.

@gogito, your problem looks like either the sta submodule is out of date or a build-cache issue.

rovinski, 2025-03-25:
> It's unfortunate that we can't builld for Cent8 just because it's EOL (and only just a few months).

CentOS 8 was EOL in 2021: https://www.centos.org/centos-linux-eol/

gogito, 2025-03-25:
> > It's unfortunate that we can't builld for Cent8 just because it's EOL (and only just a few months).
> 
> CentOS 8 was EOL in 2021: https://www.centos.org/centos-linux-eol/

Ah my mistake, I confused it with Centos7

vvbandeira, 2025-08-22:
@gogito, I don't think there's much we can do to support CentOS7.
CentOS8 should be supported as it is compatible with RHEL8/RockyLinux/AlmaLinux.
Looking at the logs, and that you did not find `devtoolset8`, your issue is likely due to the compiler version.
Please try with GCC 9 or greater.