The-OpenROAD-Project/OpenROAD #6991: [ERROR DRT-0073] No access point for ces_0_0/io_ins_left[10].

oharboe, 2025-04-04:
### Describe the bug

[image]

[image]

### Expected Behavior

Should work

### Environment

```shell
asf
```

### To Reproduce

See above

### Relevant log output

```shell

```

maliberty, 2025-04-04:
Are they on the same layer?  The colors differ.

oharboe, 2025-04-04:
> Are they on the same layer? The colors differ.

I added one to a selection set, they are mirrored. I think they work everywhere else, or at least there are only two errors listed.

maliberty, 2025-04-04:
@bnmfw there is no need to do PA on abutted pins as there is nothing to route.   dbNet::isConnectedByAbutment can be used

maliberty, 2025-04-05:
grt calls pin access to get a better sense which gcell it should route to.

bnmfw, 2025-04-18:
I'm currently working on this task. The issue here is a bit deeper than only abutment. I'm currently working on a branch where any pin already routed by abutment will skip PA. However the error still persists.
Error 73 is a bit misleading, I think there is a case for changing it. The PA problem is not only with `ces_1_0/io_ins_left[10]` but actually any instance on its unique inst family, such as `ces_1_7`. The latter inst is not routed by abutment and PA can't find a valid access point for that pin as well (it should). I'm currently trying to understand why that is the case.

maliberty, 2025-04-18:
That constrait does prevent off-grid planar access so the PA problem is real.  It looks like this is an mpl placement issue (@joaomai @AcKoucher )

oharboe, 2025-04-18:
Interesting.

Could the error message be improved?

It was evidently incredibly tough to figure out that this is an mpl issue...

AcKoucher, 2025-04-20:
@oharboe #6738 should ensure that the alignment of the signal pins is correct.

> Could the error message be improved?

@joaomai It would be interesting to add a warning when we failed to align all the pins.
Perhaps something like "Couldn't align pins <pin_1> <pin_2> ... <pin_n> with the  track-grid."

maliberty, 2025-04-21:
If the lef58RightWayOnGridOnlyConstraint_ exists for the layer on which you can't align the pins then I think we should make it an error, otherwise a warning is fine.

I think we should be able to align the pins.  The block is a hierarchical macro and the pins should have been on grid when the block was implemented.

AcKoucher, 2025-04-22:
@joaomai Please, add the warning/error subsequently to #6738.

AcKoucher, 2025-05-01:
Pin alignment problem addressed by #6360.