The-OpenROAD-Project/OpenROAD #7005: [NOT READY YET COLLECTING DATA] `bazel build :openroad -c opt` results in slow binary

oharboe, 2025-04-05:
### Description

Not ready yet...

Saving issue while I'm collecting data

From OpenROAD folder:

```
$ bazel build :openroad -c opt
$ export PATH=$(pwd)/bazel-out/k8-opt/bin:$PATH
```

Untar and run [global_route_ibex_core_sky130hd_base_2025-04-05_15-20.tar.gz](https://github.com/user-attachments/files/19615554/global_route_ibex_core_sky130hd_base_2025-04-05_15-20.tar.gz)

```
perf top --pid=$(pgrep openroad)
```

```
Overhead  Shared Object        Symbol
  13,26%  libc.so.6            [.] _int_malloc
  12,64%  libc.so.6            [.] _int_free
  11,51%  libc.so.6            [.] malloc
   4,60%  libc.so.6            [.] cfree@GLIBC_2.2.5
   3,91%  libc.so.6            [.] malloc_consolidate
   3,35%  openroad             [.] boost::polygon::polygon_formation::ScanLineToPolygonItrs<false, int, boost::polygon::polygon_90_with_hole
   2,69%  openroad             [.] boost::polygon::polygon_formation::ActiveTail<int>::pushCoordinate(int)
   2,67%  libstdc++.so.6.0.33  [.] operator new(unsigned long)
   2,29%  openroad             [.] std::pair<boost::polygon::polygon_formation::ActiveTail<int>*, boost::polygon::polygon_formation::ActiveT
   1,99%  libstdc++.so.6.0.33  [.] std::_Rb_tree_rebalance_for_erase(std::_Rb_tree_node_base*, std::_Rb_tree_node_base&)
   1,98%  libstdc++.so.6.0.33  [.] std::_Rb_tree_insert_and_rebalance(bool, std::_Rb_tree_node_base*, std::_Rb_tree_node_base*, std::_Rb_tre
   1,93%  libc.so.6            [.] unlink_chunk.isra.0
   1,90%  openroad             [.] unsigned int boost::polygon::get_polygons<std::vector<boost::polygon::polygon_90_with_holes_data<int>, st
   1,62%  libc.so.6            [.] __memmove_avx_unaligned_erms
   1,62%  openroad             [.] boost::polygon::enable_if<boost::polygon::gtl_and<boost::polygon::is_mutable_polygon_90_with_holes_type<b
   1,27%  openroad             [.] drt::FlexGCWorker::Impl::initNet_pins_polygonCorners_helper(drt::gcNet*, drt::gcPin*)
   1,26%  openroad             [.] std::map<int, boost::polygon::polygon_formation::ActiveTail<int>*, std::less<int>, std::allocator<std::pa

```

oharboe, 2025-04-06:
Paused stack trace for more context:

```
boost::polygon::polygon_formation::PolyLine<int>::pushPoint(boost::polygon::point_data<int> const&) (@boost::polygon::polygon_formation::PolyLine<int>::pushPoint(boost::polygon::point_data<int> const&):72)
boost::polygon::polygon_formation::ScanLineToPolygonItrs<false, int, boost::polygon::polygon_90_with_holes_concept>::processEdges(std::__1::__wrap_iter<boost::polygon::polygon_formation::PolyLinePolygonWithHolesData<false, int>*>&, std::__1::__wrap_iter<boost::polygon::polygon_formation::PolyLinePolygonWithHolesData<false, int>*>&, int, std::__1::vector<boost::polygon::interval_data<int>, std::__1::allocator<boost::polygon::interval_data<int>>>&, std::__1::vector<boost::polygon::interval_data<int>, std::__1::allocator<boost::polygon::interval_data<int>>>&, unsigned long) (@boost::polygon::polygon_formation::ScanLineToPolygonItrs<false, int, boost::polygon::polygon_90_with_holes_concept>::processEdges(std::__1::__wrap_iter<boost::polygon::polygon_formation::PolyLinePolygonWithHolesData<false, int>*>&, std::__1::__wrap_iter<boost::polygon::polygon_formation::PolyLinePolygonWithHolesData<false, int>*>&, int, std::__1::vector<boost::polygon::interval_data<int>, std::__1::allocator<boost::polygon::interval_data<int>>>&, std::__1::vector<boost::polygon::interval_data<int>, std::__1::allocator<boost::polygon::interval_data<int>>>&, unsigned long):257)
unsigned int boost::polygon::get_polygons<std::__1::vector<boost::polygon::polygon_90_with_holes_data<int>, std::__1::allocator<boost::polygon::polygon_90_with_holes_data<int>>>, std::__1::__wrap_iter<std::__1::pair<int, std::__1::pair<int, int>>*>, boost::polygon::polygon_90_with_holes_concept>(std::__1::vector<boost::polygon::polygon_90_with_holes_data<int>, std::__1::allocator<boost::polygon::polygon_90_with_holes_data<int>>>&, std::__1::__wrap_iter<std::__1::pair<int, std::__1::pair<int, int>>*>, std::__1::__wrap_iter<std::__1::pair<int, std::__1::pair<int, int>>*>, boost::polygon::orientation_2d, bool, boost::polygon::polygon_90_with_holes_concept, unsigned long) (@unsigned int boost::polygon::get_polygons<std::__1::vector<boost::polygon::polygon_90_with_holes_data<int>, std::__1::allocator<boost::polygon::polygon_90_with_holes_data<int>>>, std::__1::__wrap_iter<std::__1::pair<int, std::__1::pair<int, int>>*>, boost::polygon::polygon_90_with_holes_concept>(std::__1::vector<boost::polygon::polygon_90_with_holes_data<int>, std::__1::allocator<boost::polygon::polygon_90_with_holes_data<int>>>&, std::__1::__wrap_iter<std::__1::pair<int, std::__1::pair<int, int>>*>, std::__1::__wrap_iter<std::__1::pair<int, std::__1::pair<int, int>>*>, boost::polygon::orientation_2d, bool, boost::polygon::polygon_90_with_holes_concept, unsigned long):137)
drt::FlexGCWorker::Impl::initNet_pins_polygonCorners_helper(drt::gcNet*, drt::gcPin*) (@drt::FlexGCWorker::Impl::initNet_pins_polygonCorners_helper(drt::gcNet*, drt::gcPin*):415)
drt::FlexGCWorker::Impl::initNet(drt::gcNet*) (@drt::FlexGCWorker::Impl::initNet(drt::gcNet*):44)
drt::FlexGCWorker::Impl::initPA1() (@drt::FlexGCWorker::Impl::initPA1():18)
drt::FlexPA::genPatternsGC(std::__1::set<drt::frBlockObject*, std::__1::less<drt::frBlockObject*>, std::__1::allocator<drt::frBlockObject*>> const&, std::__1::vector<std::__1::pair<drt::frConnFig*, drt::frBlockObject*>, std::__1::allocator<std::__1::pair<drt::frConnFig*, drt::frBlockObject*>>> const&, drt::FlexPA::PatternType, std::__1::set<drt::frBlockObject*, std::__1::less<drt::frBlockObject*>, std::__1::allocator<drt::frBlockObject*>>*) (@drt::FlexPA::genPatternsGC(std::__1::set<drt::frBlockObject*, std::__1::less<drt::frBlockObject*>, std::__1::allocator<drt::frBlockObject*>> const&, std::__1::vector<std::__1::pair<drt::frConnFig*, drt::frBlockObject*>, std::__1::allocator<std::__1::pair<drt::frConnFig*, drt::frBlockObject*>>> const&, drt::FlexPA::PatternType, std::__1::set<drt::frBlockObject*, std::__1::less<drt::frBlockObject*>, std::__1::allocator<drt::frBlockObject*>>*):149)
drt::FlexPA::getEdgeCost(drt::FlexDPNode*, drt::FlexDPNode*, std::__1::vector<drt::frInst*, std::__1::allocator<drt::frInst*>> const&) (@drt::FlexPA::getEdgeCost(drt::FlexDPNode*, drt::FlexDPNode*, std::__1::vector<drt::frInst*, std::__1::allocator<drt::frInst*>> const&):94)
drt::FlexPA::genInstRowPatternPerform(std::__1::vector<std::__1::vector<std::__1::unique_ptr<drt::FlexDPNode, std::__1::default_delete<drt::FlexDPNode>>, std::__1::allocator<std::__1::unique_ptr<drt::FlexDPNode, std::__1::default_delete<drt::FlexDPNode>>>>, std::__1::allocator<std::__1::vector<std::__1::unique_ptr<drt::FlexDPNode, std::__1::default_delete<drt::FlexDPNode>>, std::__1::allocator<std::__1::unique_ptr<drt::FlexDPNode, std::__1::default_delete<drt::FlexDPNode>>>>>>&, std::__1::vector<drt::frInst*, std::__1::allocator<drt::frInst*>> const&) (@drt::FlexPA::genInstRowPatternPerform(std::__1::vector<std::__1::vector<std::__1::unique_ptr<drt::FlexDPNode, std::__1::default_delete<drt::FlexDPNode>>, std::__1::allocator<std::__1::unique_ptr<drt::FlexDPNode, std::__1::default_delete<drt::FlexDPNode>>>>, std::__1::allocator<std::__1::vector<std::__1::unique_ptr<drt::FlexDPNode, std::__1::default_delete<drt::FlexDPNode>>, std::__1::allocator<std::__1::unique_ptr<drt::FlexDPNode, std::__1::default_delete<drt::FlexDPNode>>>>>>&, std::__1::vector<drt::frInst*, std::__1::allocator<drt::frInst*>> const&):93)
[... cut ...]
drt::TritonRoute::pinAccess(std::__1::vector<odb::dbInst*, std::__1::allocator<odb::dbInst*>> const&) (@drt::TritonRoute::pinAccess(std::__1::vector<odb::dbInst*, std::__1::allocator<odb::dbInst*>> const&):97)
pin_access_cmd(char const*, char const*, char const*, int, int) (@pin_access_cmd(char const*, char const*, char const*, int, int):70)
_wrap_pin_access_cmd(void*, Tcl_Interp*, int, Tcl_Obj* const*) (@_wrap_pin_access_cmd(void*, Tcl_Interp*, int, Tcl_Obj* const*):152)
Tcl_EvalObjv (@Tcl_EvalObjv:42)
TclEvalEx (@TclEvalEx:593)
Tcl_Eval (@Tcl_Eval:15)
```

oharboe, 2025-04-06:
A pef report after a full run:

```
Overhead  Command   Shared Object         Symbol
   9,38%  openroad  libc.so.6             [.] malloc                                                                                                         ◆
   8,82%  openroad  libc.so.6             [.] _int_free                                                                                                      ▒
   6,85%  openroad  openroad              [.] boost::polygon::polygon_formation::ScanLineToPolygonItrs<false, int, boost::polygon::polygon_90_with_holes_conc▒
   6,02%  openroad  libc.so.6             [.] cfree@GLIBC_2.2.5                                                                                              ▒
   5,31%  openroad  libc.so.6             [.] _int_malloc                                                                                                    ▒
   2,51%  openroad  openroad              [.] void std::__1::__tree_balance_after_insert[abi:ne190101]<std::__1::__tree_node_base<void*>*>(std::__1::__tree_n▒
   2,24%  openroad  openroad              [.] void std::__1::__tree_remove[abi:ne190101]<std::__1::__tree_node_base<void*>*>(std::__1::__tree_node_base<void*▒
   1,95%  openroad  openroad              [.] void boost::polygon::boolean_op::BooleanOp<int, int>::processInterval<std::__1::vector<std::__1::pair<boost::po▒
   1,85%  openroad  openroad              [.] operator new(unsigned long)                                                                                    ▒
   1,83%  openroad  libc.so.6             [.] __memmove_avx_unaligned_erms                                                                                   ▒
   1,79%  openroad  openroad              [.] boost::polygon::polygon_formation::PolyLine<int>::pushPoint(boost::polygon::point_data<int> const&)            ▒
   1,79%  openroad  openroad              [.] boost::polygon::polygon_formation::ScanLineToPolygonItrs<false, int, boost::polygon::polygon_90_with_holes_conc▒
   1,73%  openroad  openroad              [.] unsigned int boost::polygon::get_polygons<std::__1::vector<boost::polygon::polygon_90_with_holes_data<int>, std▒
   1,58%  openroad  openroad              [.] boost::polygon::enable_if<boost::polygon::gtl_and<boost::polygon::is_mutable_polygon_90_with_holes_type<boost::▒
   1,47%  openroad  openroad              [.] std::__1::vector<int, std::__1::allocator<int> >::insert(std::__1::__wrap_iter<int const*>, int&&)             ▒
   1,41%  openroad  libc.so.6             [.] malloc_consolidate                                                                                             ▒
   1,31%  openroad  openroad              [.] drt::FlexGCWorker::Impl::initNet_pins_polygonCorners_helper(drt::gcNet*, drt::gcPin*)                          ▒
   1,05%  openroad  openroad              [.] drt::FlexGCWorker::Impl::initNet_pins_polygonEdges_helper_outer(drt::gcNet*, drt::gcPin*, drt::gcPolygon*, int,▒
   0,94%  openroad  openroad              [.] std::__1::pair<boost::polygon::polygon_formation::ActiveTail<int>*, boost::polygon::polygon_formation::ActiveTa▒
   0,88%  openroad  openroad              [.] std::__1::__tree_node_base<void*>*& std::__1::__tree<std::__1::__value_type<int, boost::polygon::polygon_format▒
   0,80%  openroad  openroad              [.] void std::__1::__introsort<std::__1::_ClassicAlgPolicy, std::__1::__less<void, void>&, std::__1::pair<int, std:▒
   0,79%  openroad  libc.so.6             [.] unlink_chunk.isra.0                                                                                            ▒
   0,78%  openroad  openroad              [.] boost::geometry::index::detail::rtree::visitors::spatial_query<boost::geometry::index::rtree<std::__1::pair<odb▒
   0,76%  openroad  openroad              [.] void boost::polygon::boolean_op::default_arg_workaround<int>::applyBooleanOr<int>(std::__1::vector<std::__1::pa▒
   0,76%  openroad  openroad              [.] std::__1::vector<boost::polygon::rectangle_data<int>, std::__1::allocator<boost::polygon::rectangle_data<int> >▒
   0,71%  openroad  openroad              [.] boost::polygon::polygon_formation::ActiveTail<int>* boost::polygon::polygon_formation::ActiveTail<int>::joinCha▒
```

oharboe, 2025-04-06:
Interesting. This appears to be a problem in route as well as grt:

```
3516631 ./external/_main~_repo_rules~openroad/openroad -exit -no_init -threads 16 -no_splash ./scripts/detail_route.tcl -metrics bazel-out/k8-fastbuild/bin/designs/asap7/ethmac_lvt/logs/asap7/ethmac/base/5_2_route.json
$ perf top --pid=3516631
```

```
Overhead  Shared Object     Symbol
  11,06%  libc.so.6         [.] _int_free
  10,86%  libc.so.6         [.] malloc
   8,46%  libc.so.6         [.] _int_malloc
   7,22%  openroad          [.] boost::polygon::polygon_formation::ScanLineToPolygonItrs<false, int, boost::polygon::polygon_90_with_holes_concept>::processEd
   6,12%  openroad          [.] drt::FlexGCWorker::Impl::initDesign_skipObj(drt::frBlockObject*)
   4,45%  libc.so.6         [.] cfree@GLIBC_2.2.5
   4,35%  openroad          [.] boost::geometry::index::detail::rtree::visitors::spatial_query<boost::geometry::index::rtree<std::__1::pair<odb::Rect, drt::fr
   2,97%  openroad          [.] std::__1::back_insert_iterator<std::__1::vector<std::__1::pair<odb::Rect, drt::frBlockObject*>, std::__1::allocator<std::__1::
   2,25%  libc.so.6         [.] __memmove_avx_unaligned_erms
   1,93%  openroad          [.] boost::polygon::polygon_formation::ScanLineToPolygonItrs<false, int, boost::polygon::polygon_90_with_holes_concept>::clearOutp
   1,91%  libc.so.6         [.] malloc_consolidate
   1,84%  openroad          [.] unsigned int boost::polygon::get_polygons<std::__1::vector<boost::polygon::polygon_90_with_holes_data<int>, std::__1::allocato
   1,60%  openroad          [.] operator new(unsigned long)
   1,57%  openroad          [.] boost::polygon::polygon_formation::PolyLine<int>::pushPoint(boost::polygon::point_data<int> const&)
   1,52%  openroad          [.] void std::__1::__tree_balance_after_insert[abi:ne190101]<std::__1::__tree_node_base<void*>*>(std::__1::__tree_node_base<void*>
   1,50%  openroad          [.] void std::__1::__tree_remove[abi:ne190101]<std::__1::__tree_node_base<void*>*>(std::__1::__tree_node_base<void*>*, std::__1::_
   1,50%  openroad          [.] boost::polygon::enable_if<boost::polygon::gtl_and<boost::polygon::is_mutable_polygon_90_with_holes_type<boost::polygon::polygo
   1,36%  openroad          [.] void boost::polygon::boolean_op::BooleanOp<int, int>::processInterval<std::__1::vector<std::__1::pair<boost::polygon::interval
   1,24%  openroad          [.] std::__1::vector<int, std::__1::allocator<int> >::insert(std::__1::__wrap_iter<int const*>, int&&)
   1,16%  openroad          [.] drt::FlexGCWorker::Impl::initNet_pins_polygonCorners_helper(drt::gcNet*, drt::gcPin*)  
   0,95%  openroad          [.] std::__1::__tree_node_base<void*>*& std::__1::__tree<std::__1::__value_type<int, boost::polygon::polygon_formation::ActiveTail
   0,89%  openroad          [.] drt::frInstBlockage::typeId() const
   0,81%  libc.so.6         [.] unlink_chunk.isra.0
   0,81%  openroad          [.] std::__1::pair<boost::polygon::polygon_formation::ActiveTail<int>*, boost::polygon::polygon_formation::ActiveTail<int>*> boost
   0,69%  openroad          [.] drt::FlexGCWorker::Impl::initNet_pins_polygonEdges_helper_outer(drt::gcNet*, drt::gcPin*
```

oharboe, 2025-04-06:
Running with gperftools. I haven't used this before, so I don't quite know what I'm looking at, but I saw the other day that it can be really quite useful and it is easy to get the data. The picture is a bit more complicated than I first imagined, because a lot of the allocations appear to be from within boost. I'm unsure if those allocations are unavoidable or if they can be avoided if boost is used differently.

Some cribs below on what I did to profile:

```
sudo apt-get install google-perftools
CPUPROFILE=prof.out $(command -v openroad) -exit -no_init ${SCRIPTS_DIR}/global_route.tcl
google-pprof -web $(command -v openroad) prof.out
```

```
$ git diff
diff --git a/BUILD.bazel b/BUILD.bazel
index 7ef2c1d12..17ef67b58 100644
--- a/BUILD.bazel
+++ b/BUILD.bazel
@@ -57,7 +57,7 @@ cc_binary(
     copts = OPENROAD_COPTS,
     features = ["-use_header_modules"],
     linkopts = select({
-        ":platform_cli": [],
+        ":platform_cli": ["-lprofiler"],
     }),
     visibility = ["//visibility:public"],
     deps = OPENROAD_BINARY_DEPS + [
```

[image]

QuantamHD, 2025-04-06:
I'll bet money this the return by value stuff we were arguing about

maliberty, 2025-04-06:
Since we will move to bazel eventually it isn't important to me to figure out why the current cmake approach is slower.

closed 2025-04-06 by maliberty