The-OpenROAD-Project/OpenROAD #7006: Make it possible to add bazel built OpenROAD to path

oharboe, 2025-04-05:
### Description

There's a problem with adding the folder of the openroad built by Bazel to the path, which is not an uncommon use-case.

```
$ bazel build :openroad -c opt
$ export PATH=$(pwd)/bazel-out/k8-opt/bin:$PATH
$ openroad
ERROR: external/rules_cc~/cc/runfiles/runfiles.cc(123): cannot find runfiles (argv0="openroad")
```

If you launch it passing in the entire path, $(command -v openroad), it works.

```
$ command -v openroad
.../tools/OpenROAD/bazel-out/k8-opt/bin/openroad
$ .../tools/OpenROAD/bazel-out/k8-opt/bin/openroad
OpenROAD HDL-HEAD bazel-build
Features included (+) or not (-): -GPU -GUI -Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
$
```

hzeller, 2025-04-11:
This hints at another problem: we probably can't simply copy the binary somewhere ('make install' like) as it will now have trouble to deal with the runfiles. We have to figure out exactly what these runfiles comprise and make it possible to have an install script that copies the binary into another place and it work there.

@QuantamHD do you know what currently is in there ? Is it just the tclsh system files ?

hzeller, 2025-04-11:
```
find bazel-bin/openroad.runfiles
```
yep, looks like mostly tcl stuff

oharboe, 2025-04-11:
> ```
> find bazel-bin/openroad.runfiles
> ```
> yep, looks like mostly tcl stuff

The big blockers to using bazel built openroad outside of bazel is to add python and qt support, AFAIK.

hzeller, 2025-04-11:
I think Python might be straightforward, and I think @QuantamHD has some ideas for Qt.

commit 10526296 (2025-04-12): bazel: make OpenROAD in path work, review feedback

fixes #7006
closed 2025-04-12 by maliberty