The-OpenROAD-Project/OpenROAD #7019: Convert `[INFO PPL-0070] Pin xxxx placed at (677.53um, 0.04um).` to debug info

oharboe, 2025-04-07:
### Description

Make this a debug message and let the user inspect pin positions in the GUI, or move it 

```
[INFO PPL-0070] Pin io_foo placed at (677.53um, 0.04um).
```

oharboe, 2025-04-07:
> How about we print it only if the computed location differs from the user given one?  That seems worth knowing.

I didnt know that could happen. The position is specified by float, so what defines different? No such thing as two equal floats....

oharboe, 2025-04-07:
> We snap it to the manufacturing grid.  Internally the float in micron is converted to an integer in DBU.  Also -force_to_die_boundary will move the pin if needed.

So if something is moved will the log say why?

maliberty, 2025-04-07:
Not presently but it could.

closed 2025-04-09 by maliberty