The-OpenROAD-Project/OpenROAD #7041: Placement ignores PDN stripes

steux, 2025-04-08:
### Describe the bug

I observe that OpenRoad is totally unable to route a very simple circuit when PDN stripes using MET2 and MET3 are being used. Indeed, this is due to placement totally ignoring these stripes, placing cells under the stripes, making routing almost impossible. Possibly not an issue with platforms with many metal layers (like sky130), but I'm using XH035 and 3 metal layers only. Is it possible to use fences for placement ? I don't see the option, but fences are supposed to be supported by detailed placement algorithm.
[image]

### Expected Behavior

Standard cells not placed under PDN stripes to ease the routing (or even make it possible).

### Environment

```shell
Git commit: 79d44314901d4ba36bb43d26bc14614aac1d3b11
kernel: Linux 5.15.167.4-microsoft-standard-WSL2
os: Ubuntu 22.04.5 LTS (Jammy Jellyfish)
cmake version 3.22.1
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-20349-g79d443149
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare - Success
[... 86 lines cut ...]
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /usr/local
-- Found FLEX: /usr/bin/flex (found version "2.6.4")
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL readline library: /usr/lib/x86_64-linux-gnu/libtclreadline.so
-- TCL readline header: /usr/include/x86_64-linux-gnu/tclreadline.h
-- CUDD library: not found
-- SSTA: 0
-- Found SWIG: /usr/bin/swig4.0 (found suitable version "4.0.2", minimum required is "3.0")
-- STA executable: /home/steux/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/sta
-- Configuring incomplete, errors occurred!
```

### To Reproduce

I will make an example on Sky130 to showcase the issue. At the moment, as I'm using a platform under NDA, I can't submit this as a tarball.

### Relevant log output

```shell

```

maliberty, 2025-04-08:
@osamahammad21 marking any site that overlaps a fixed M2 object as unusable should fix this.  Its the second time its come up so we should address it when time allows.

QuantamHD, 2025-04-08:
I think we should support this in the detailed placer by selectively enabling sites for cells with no overlapping shapes would be a better long term solution.

This comes up in advanced nodes as well.

maliberty, 2025-04-08:
What is the difference between what you are suggesting and what I'm suggesting?  It sounds the same to me.

It is unusual to have strips in m2 in a more advanced node.  Let's discuss where this comes up offline.

steux, 2025-04-08:
Thank you very much for your quick answer. I'll follow the instructions on the other discussion.

steux, 2025-04-09:
I've tried inserting blockages in post_pdn.tcl, and it's basically working, but now there is an issue with antenna cell insertion. It doesn't pass the check. Do you know how to bypass it ? I get :

```
[INFO GRT-0015] Inserted 2354 diodes.
[WARNING DPL-0004] Placed in rows check failed (29).
 ANTENNA_11243
 ANTENNA_11245
 ANTENNA_11247
 ANTENNA_11249
 ANTENNA_11251
 ANTENNA_11253
 ANTENNA_11332
 ANTENNA_11334
 ANTENNA_11336
 ANTENNA_11338
 ANTENNA_11340
 ANTENNA_11658
 ANTENNA_11660
 ANTENNA_11694
 ANTENNA_11696
 ANTENNA_11698
 ANTENNA_11700
 ANTENNA_11702
 ANTENNA_11704
 ANTENNA_11711
 ANTENNA_11712
 ANTENNA_11713
 ANTENNA_11714
 ANTENNA_11715
 ANTENNA_12070
 ANTENNA_12072
 ANTENNA_12074
 ANTENNA_12076
 ANTENNA_12078
[ERROR DPL-0033] detailed placement checks failed.
Error: global_route.tcl, 114 DPL-0033
Command exited with non-zero status 1

```
[image]

maliberty, 2025-04-09:
How many diodes are getting inserted?  I wonder if it is too large to be able to place them.

steux, 2025-04-09:
2354 diodes. But no, it's not the issue. I've tested with place density 0.1. There is much room for diode insertion. As soon as I remove the stripes, it's OK. I guess it's trying to place antenna diodes in the blockages. Is there a way to bypass the warning/error and go on ? It's not really an issue if these diodes are not placed in rows (i.e. behind blockages). They will be connected through VDD and GND horizontal stripes anyway...

maliberty, 2025-04-09:
If you have created a blockage the placer should obey it regardless of cell type.  Can you give an close up of one failure?  

Its hard to debug this on an unknown PDK without access.  You might want professional support under NDA from https://precisioninno.com/

maliberty, 2025-04-14:
Depending on the size of the violation and the diode area you may need multiple diodes.

@eder-matheus @luis201420 does ant repair not respect placement blockages?

eder-matheus, 2025-04-14:
> [@eder-matheus](https://github.com/eder-matheus) [@luis201420](https://github.com/luis201420) does ant repair not respect placement blockages?

I think it is not considering placement blockages. We get only the fixed instances geometries and check if a diode would overlap with them, and if it overlaps, we allow DPL to legalize the diode. The support would require adding the placement blockages geometries to the structure used to check the overlaps. @luis201420 Could you look at it? I believe it would be a straightforward fix.

luis201420, 2025-04-16:
I implemented a function to add the geometries for placement blockages. This prevents diodes from being placed in blockages. 
If there are any diodes with illegal positions, DPL will legalize the positions of those diodes, but will ignore soft blockages.

@steux What type of blockage did you define in that region?

steux, 2025-04-18:
I guess they are hard, since I haven't specified anything.
Here is the post_pdn.tcl file I've used: 
```
puts "SandCastle: Creating PDN stripes blockages"
proc block_stripes {} {
  set tech [ord::get_db_tech]
  set units [$tech getDbUnitsPerMicron]
  set block [ord::get_db_block]
  set ymin [expr round(441 * $units)]
  set ymax [expr round(3645 * $units)]
  set xmin [expr round((691 - 10) * $units)]
  set xmax [expr round((691 + 10) * $units)]
  odb::dbBlockage_create $block $xmin $ymin $xmax $ymax
  set xmin [expr round((941 - 10) * $units)]
  set xmax [expr round((941 + 10) * $units)]
  odb::dbBlockage_create $block $xmin $ymin $xmax $ymax
  set xmin [expr round((1191 - 10) * $units)]
  set xmax [expr round((1191 + 10) * $units)]
  odb::dbBlockage_create $block $xmin $ymin $xmax $ymax
  set xmin [expr round((1441 - 10) * $units)]
  set xmax [expr round((1441 + 10) * $units)]
  odb::dbBlockage_create $block $xmin $ymin $xmax $ymax
  set xmin [expr round((1691 - 10) * $units)]
  set xmax [expr round((1691 + 10) * $units)]
  odb::dbBlockage_create $block $xmin $ymin $xmax $ymax
  set xmin [expr round((1941 - 10) * $units)]
  set xmax [expr round((1941 + 10) * $units)]
  odb::dbBlockage_create $block $xmin $ymin $xmax $ymax
  set xmin [expr round((2191 - 10) * $units)]
  set xmax [expr round((2191 + 10) * $units)]
  odb::dbBlockage_create $block $xmin $ymin $xmax $ymax
  set xmin [expr round((2441 - 10) * $units)]
  set xmax [expr round((2441 + 10) * $units)]
  odb::dbBlockage_create $block $xmin $ymin $xmax $ymax
  set xmin [expr round((2691 - 10) * $units)]
  set xmax [expr round((2691 + 10) * $units)]
  odb::dbBlockage_create $block $xmin $ymin $xmax $ymax
  set xmin [expr round((2941 - 10) * $units)]
  set xmax [expr round((2941 + 10) * $units)]
  odb::dbBlockage_create $block $xmin $ymin $xmax $ymax
  set xmin [expr round((3191 - 10) * $units)]
  set xmax [expr round((3191 + 10) * $units)]
  odb::dbBlockage_create $block $xmin $ymin $xmax $ymax
  set xmin [expr round((3441 - 10) * $units)]
  set xmax [expr round((3441 + 10) * $units)]
  odb::dbBlockage_create $block $xmin $ymin $xmax $ymax
}
block_stripes
```

eder-matheus, 2025-04-22:
@steux You're right, these blockages should not be soft when created using this command.

@luis201420 Please create the PR that will solve the diode insertion error regarding blockages.

luis201420, 2025-04-25:
The PR #7200 has been merged to fix the diode insertion error regarding placement blockages. 
@steux, please update your branch and let us know if the issue persists.

closed 2025-06-12 by osamahammad21