The-OpenROAD-Project/OpenROAD #7109: bazel: A test fails as it attempts to write to writeonly directories

hzeller, 2025-04-12:
### Describe the bug

This test currently fails `bazel test -c opt //test:ibex_sky130hs-tcl`

Last line of the failure is

```
...
# checkpoint
set dpl_db [make_result_file ${design}_${platform}_dpl.db]
write_db $dpl_db
set verilog_file [make_result_file ${design}_${platform}.v]
write_verilog $verilog_file
Error: flow.tcl, 195 cannot write file /long/bazel/prefix/sandbox/linux-sandbox/6/execroot/_main/bazel-out/k8-opt/bin/test/ibex_sky130hs-tcl.runfiles/_main/test/results/ibex_sky130hs-tcl.v.
```

.. so it looks like the test is attempting to write to the sandboxed directory (which bazel is strict about not to allow) instead of its tmpdir.

### Expected Behavior

writing to its assigned tmpdir relative to `TEST_TMPDIR`.

### Environment

```shell
bazel
```

### To Reproduce

```
bazel test -c opt //test:ibex_sky130hs-tcl
```

This used to work, but for the last few days it fails, so something that should write to the tmpdir started failing.

### Relevant log output

```shell

```

### Additional Context

too much stuff to fill out in this bug report :)

hzeller, 2025-04-12:
@oharboe this sounds like something you might know

maliberty, 2025-04-12:
Virtually every test writes to its local results dir so I'm curious why this one is flagged.

oharboe, 2025-04-12:
> Virtually every test writes to its local results dir so I'm curious why this one is flagged.

my fix fixed the helper fn, i searched for info script, so I think I got it all.

PR #7110 (2025-04-13): bazel: use TEST_TEMPDIR for results folder
fixes #7109

To examine a failing test, re-run it with:

    bazel test --test_output=all --sandbox_debug //path/to:test
commit e43bed0c (2025-04-13): bazel: use TEST_TEMPDIR for results folder

fixes #7109

To examine a failing test, re-run it with:

bazel test --test_output=all --sandbox_debug //path/to:test
closed 2025-04-13 by maliberty