The-OpenROAD-Project/OpenROAD #7111: bazel - OpenROAD splash screen does not print out build configuration

oharboe, 2025-04-12:
### Description

```
bazel build -c dbg :openroad
bazel-bin/openroad
```

=> does not print out the Debug header. 

```
OpenROAD HDL-HEAD bazel-build
Features included (+) or not (-): -GPU -GUI -Python
```

### Suggested Solution

Print out bazel compile config

gadfort, 2025-04-12:
it also doesn't print the proper version tag either

oharboe, 2025-04-12:
> it also doesn't print the proper version tag either

It should not print the git tag, that would not be idiomatic for content hashes.

Version + possibly content hash would be idiomatic.

gadfort, 2025-04-12:
that is what the `-no_splash` cmd argument is for, that supresses this header. We still need a way to know the version of the software and I don't see a reason to move away from what is currently there (which gives us a number we can compare against).

By the way, to get the printing to work, you just need to make sure `BUILD_TYPE` is defined as something other than "release" (not sure how to define those for bazel).

QuantamHD, 2025-04-13:
This is supported in bazel too. https://github.com/bazel-contrib/bazel-lib/blob/main/docs/stamping.md

oharboe, 2025-04-13:
> This is supported in bazel too. https://github.com/bazel-contrib/bazel-lib/blob/main/docs/stamping.md

Good for a release build, yes, for dev builds no, ref. caching.

> Typically this is off by default to make sure caching works 

👍

gadfort, 2025-04-15:
Moving the version issue to #7140

closed 2025-04-15 by maliberty