The-OpenROAD-Project/OpenROAD #7118: tap: insertion pattern may leave "tap deserts" with higher values

donn, 2025-04-12:
### Describe the bug

With a higher tap distance value for sky130, the tap insertion pattern causes some parts of the design (specifically in this design, the beginning of odd-numbered rows) to lack tap cells…

[image]

…which results in DRC issues.

[image]

### Expected Behavior

When the maximum tap distance is 15, the distance between tap cells provided to `tap` should logically be something like 28-30. `tap` should be able to handle those values…

### Environment

```shell
Git commit: fe78ace331be1851bc79081022cb2c59cb734f5f
kernel: Darwin 24.3.0
os: macOS 15.3.1
cmake version 3.30.5
CMake Warning (dev) at src/CMakeLists.txt:138 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning at src/CMakeLists.txt:230 (message):
  spdlog: SPDLOG_FMT_EXTERNAL=ON

CMake Warning (dev) at src/dpl/CMakeLists.txt:6 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at src/cts/CMakeLists.txt:4 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at src/drt/CMakeLists.txt:17 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at src/utl/CMakeLists.txt:6 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at src/dst/CMakeLists.txt:12 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning:
  Manually-specified variables were not used by the project:

    VERBOSE

-- The CXX compiler identification is Clang 17.0.6
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/9i42ps9fng6akgv3fbv2c21fnsij7ay1-clang-wrapper-17.0.6/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-20588-gfe78ace33
-- System name: Darwin
-- Compiler: Clang 17.0.6
-- Build type: RELEASE
-- Install prefix: /var/empty/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found OpenSTA: /nix/store/lqgzvrrhkn8vzrwwxj21m0rff9p4v1as-opensta/lib/libOpenSTA.a
-- Using OPENSTA_INCLUDE_DIR=/usr/include/sta
-- Using ABC_INCLUDE_DIR=/usr/include/abc
-- TCL library: /nix/store/pq5zjj1m0ql2as08yagn0ngnl00ndqki-tcl-8.6.15/lib/libtcl.dylib
-- TCL header: /nix/store/pq5zjj1m0ql2as08yagn0ngnl00ndqki-tcl-8.6.15/include/tcl.h
-- TCL readline library: /nix/store/safnrcyg7z47hl27qy0wri1v893rssf6-tclreadline-2.4.0/lib/libtclreadline.dylib
-- TCL readline header: /nix/store/safnrcyg7z47hl27qy0wri1v893rssf6-tclreadline-2.4.0/include
-- Found SWIG: /nix/store/rwpr3adsq8wrvq4mv4w1ndgy6qc4jv59-swig-4.2.1/bin/swig (found suitable version "4.2.1", minimum required is "4.0")
-- Using SWIG >= 4.2.1 -flatstaticmethod flag for python
-- Found Boost: /nix/store/w1kx6l12gsvj127gsn8b4a7hiqnirqng-boost-1.86.0-dev/lib/cmake/Boost-1.86.0/BoostConfig.cmake (found version "1.86.0")
-- boost: 1.86.0
-- Found Python3: /nix/store/7c494qcmh62av43zsxr3wvzh8hcpy1vl-python3-3.12.7/include/python3.12 (found version "3.12.7") found components: Development Development.Module Development.Embed
-- Found ZLIB: /nix/store/10r87kpkn4zrj0d1qhq8nfr8g2gf0imp-zlib-1.3.1/lib/libz.dylib (found version "1.3.1")
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- spdlog: 1.14.1
-- Found BISON: /nix/store/gynxhf1z1h7610ddq5fa5dpk5jnqs1xl-bison-3.8.2/bin/bison (found version "3.8.2")
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Found Protobuf: /nix/store/q04cfyxpr3ijb038bjwg6h10djlzrvzq-protobuf-28.3/lib/libprotobuf.dylib (found version "5.28.3")
-- Found re2: /nix/store/4jxydg60agwghwz9l11m3bdf7wq03818-re2-2024-07-02-dev/lib/cmake/re2/re2Config.cmake (found version "11.0.0")
-- Found GLPK: /nix/store/1libkw361rhsg15x41dspsdzn76hgyrn-glpk-5.0/lib/libglpk.dylib
-- Found OpenMP_CXX: -fopenmp=libomp (found version "5.1")
-- Found OpenMP: TRUE (found version "5.1")
-- Found OR-Tools: /nix/store/1rrgbv84blrlarbpyg12s34d29j66c4v-or-tools-9.11/lib/cmake/ortools (version: 9.11.9999)
-- TCL library: /nix/store/pq5zjj1m0ql2as08yagn0ngnl00ndqki-tcl-8.6.15/lib/libtcl.dylib
-- TCL header: /nix/store/pq5zjj1m0ql2as08yagn0ngnl00ndqki-tcl-8.6.15/include/tcl.h
-- The C compiler identification is Clang 17.0.6
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/9i42ps9fng6akgv3fbv2c21fnsij7ay1-clang-wrapper-17.0.6/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found OpenMP_C: -fopenmp=libomp (found version "5.1")
-- Found OpenMP: TRUE (found version "5.1")
-- Found OpenMP: TRUE (found version "5.1")
-- GUI is enabled
-- Found Boost: /nix/store/w1kx6l12gsvj127gsn8b4a7hiqnirqng-boost-1.86.0-dev/lib/cmake/Boost-1.86.0/BoostConfig.cmake (found version "1.86.0") found components: serialization
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /nix/store/w1kx6l12gsvj127gsn8b4a7hiqnirqng-boost-1.86.0-dev/lib/cmake/Boost-1.86.0/BoostConfig.cmake (found suitable version "1.86.0", minimum required is "1.78")
-- TCL library: /nix/store/pq5zjj1m0ql2as08yagn0ngnl00ndqki-tcl-8.6.15/lib/libtcl.dylib
-- TCL header: /nix/store/pq5zjj1m0ql2as08yagn0ngnl00ndqki-tcl-8.6.15/include/tcl.h
-- Found Boost: /nix/store/w1kx6l12gsvj127gsn8b4a7hiqnirqng-boost-1.86.0-dev/lib/cmake/Boost-1.86.0/BoostConfig.cmake (found version "1.86.0") found components: serialization system thread
-- Found Eigen3: /nix/store/d18dxj2z2vz78993djs6cgsxfp50l7bq-eigen-3.4.0/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0")
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done (7.4s)
-- Generating done (0.8s)
-- Build files have been written to: /tmp/nix-shell.x9mrcg/tmp.jlVmyJjTM6
```

### To Reproduce

Untar and execute run.sh.

[reproducible.tgz](https://github.com/user-attachments/files/19721271/reproducible.tgz)

### Relevant log output

```shell

```

### Additional Context

I added my cmake flags to ./etc/Env.sh

maliberty, 2025-04-12:
Do the endcaps contain taps?

gadfort, 2025-04-13:
@maliberty @donn is using a decap for the end cap cells so I assume there isn't one in there.

maliberty, 2025-04-13:
Why not use a tapcell as the endcap in that case and render this moot?

eder-matheus, 2025-04-22:
@donn I've run tapcell with the `-distance 15` and I got this placement:
[image]

Is this what you expect? When using distance of 15, the distance between the tapcells in a specific row is around 28-30, as you mentioned.

When using 28, the distance in a single row is the double of it.

eder-matheus, 2025-04-28:
@donn I believe this figure might help to understand what the `-distance` option is:
[image]

donn, 2025-04-28:
Somehow, I missed this diagram. Was 100% sure I took a look at the Readme.

Welp. Thank you.

closed 2025-04-28 by donn