The-OpenROAD-Project/OpenROAD #7140: bazel - needs to show git describe in splash

gadfort, 2025-04-15:
### Description

The CMake generates a useful version for use to use, which the bazel build should do as well. This allows for easy debug and if the log files are a concern it can already be disabled with `-no_splash`

oharboe, 2025-04-21:
I disagree with this feature request: this breaks bazel hashes and would cause a re-run of all tests when it is not necessary, for instance, changing a git comment or modifying a comment in the OpenROAD source code.

maliberty, 2025-04-21:
Changing a comment in the source code does not change the commit id unless you commit the change.

I think this is a useful feature.  Knowing what version of the tool generated some result is often helpful.

oharboe, 2025-04-21:
> Changing a comment in the source code does not change the commit id unless you commit the change.

Yes: I want to avoid rebuilds/retests in this case, such as a no-op merge commit.

> I think this is a useful feature.  Knowing what version of the tool generated some result is often helpful.

Yes, but it needs to be done in a different way.

I am thinking a build.txt file that is only read in runtime so that bazel does not consider it a dependency.

This has to be a solved problem.

@hzeller Any experience on how to implement this without the downsides?

QuantamHD, 2025-04-21:
You do it with the build tagging system --stamo I mentioned earlier. You just run --nostamp in everything except a special --config=release mode.

maliberty, 2025-04-21:
Given that we don't do formal releases I'm not sure how helpful that is.  If the CI has stamping on then developers will see no cache benefit.  If the CI has it off then likely nobody will ever turn it on.

hzeller, 2025-04-21:
Yes there are ways and I can take care of it when I have a minute (but next two weeks: no minutes left)

maliberty, 2026-03-01:
@hzeller this is blocking @gadfort for Bazel adoption.  Do you have an ETA to look at this?

hzeller, 2026-03-01:
In general, using the workspace command to extract info and then feed that into some sort of defines is the way to do it.

Here are the relevant parts in Verible doing it

https://github.com/chipsalliance/verible/blob/master/.bazelrc#L10
https://github.com/chipsalliance/verible/blob/master/bazel/build-version.sh

... and then using it
https://github.com/chipsalliance/verible/blob/master/verible/common/util/BUILD#L82-L100