The-OpenROAD-Project/OpenROAD #7142: `bazel -c opt :openroad` is 10x slower than `./build_openroad --local`

oharboe, 2025-04-15:
Measuring...

```
$ cd test
$ time ../bazel-bin/openroad -no_splash -no_init -exit jpeg_sky130hd.tcl
[running]
real	74m18,849s
user	73m59,972s
sys	0m17,779s
```

```
[INFO DRT-0179] Init gr pin query.
[while waiting here, I have the perf top below]
```

```
Overhead  Shared Object                                  Symbol
   4,90%  openroad                                       [.] boost::polygon::polygon_formation::ScanLineToPolygonItrs<false, int, bo
   4,89%  openroad                                       [.] drt::FlexWavefront::pop()
   3,89%  [kernel]                                       [k] 0xffffffffa46330a3
   3,81%  openroad                                       [.] drt::FlexGridGraph::expand(drt::FlexWavefrontGrid&, drt::frDirEnum cons
   3,58%  openroad                                       [.] drt::FlexGridGraph::isExpandable(drt::FlexWavefrontGrid const&, drt::fr
   3,38%  openroad                                       [.] drt::FlexGridGraph::getCosts(int, int, int, drt::frDirEnum, drt::frLaye
   3,35%  openroad                                       [.] tcmalloc::tcmalloc_internal::TCMallocPolicy<tcmalloc::tcmalloc_internal
   3,13%  openroad                                       [.] std::__1::__bit_iterator<std::__1::__bitset<1ul, 6ul>, false, 0> std::_
   2,79%  openroad                                       [.] tcmalloc::tcmalloc_internal::FreeWithHooksOrPerThread(void*, unsigned l
   2,49%  openroad                                       [.] tcmalloc::tcmalloc_internal::TCMallocPolicy<tcmalloc::tcmalloc_internal
   2,49%  openroad                                       [.] drt::FlexGridGraph::getEstCost(drt::FlexMazeIdx const&, drt::FlexMazeId
   2,42%  openroad                                       [.] drt::FlexDRWorker::modMinSpacingCostViaHelper(odb::Rect const&, int, dr
   2,40%  openroad                                       [.] drt::FlexGridGraph::hasEdge(int, int, int, drt::frDirEnum) const
   2,35%  openroad                                       [.] operator delete[](void*, unsigned long)
```

```
$ cd test
$ time ../../install/OpenROAD/bin/openroad -no_splash -no_init -exit jpeg_sky130hd.tcl
#set_pdnsim_net_voltage -net $vdd_net_name -voltage $vdd_voltage
#analyze_power_grid -net $vdd_net_name

real	8m29,438s
user	181m55,613s
sys	1m24,379s
```

```
[INFO DRT-0179] Init gr pin query.
[while waiting here, I have the perf top below]
```

```
Overhead  Shared Object                                  Symbol
   6,78%  openroad                                       [.] drt::FlexGridGraph::isExpandable(drt::FlexWavefrontGrid const&, drt::fr
   6,46%  libc.so.6                                      [.] _int_free
   5,85%  openroad                                       [.] drt::FlexGridGraph::search(std::vector<drt::FlexMazeIdx, std::allocator
   5,15%  libc.so.6                                      [.] malloc
   4,34%  openroad                                       [.] drt::FlexGridGraph::expand(drt::FlexWavefrontGrid&, drt::frDirEnum cons
   4,17%  libc.so.6                                      [.] _int_malloc
   3,79%  openroad                                       [.] drt::FlexGridGraph::getCosts(int, int, int, drt::frDirEnum, drt::frLaye
   3,49%  libc.so.6                                      [.] cfree@GLIBC_2.2.5
   2,78%  openroad                                       [.] boost::polygon::polygon_formation::ScanLineToPolygonItrs<false, int, bo
   2,64%  libstdc++.so.6.0.33                            [.] std::_Rb_tree_rebalance_for_erase(std::_Rb_tree_node_base*, std::_Rb_tr
   2,22%  openroad                                       [.] drt::FlexGridGraph::getEstCost(drt::FlexMazeIdx const&, drt::FlexMazeId
   2,09%  openroad                                       [.] drt::FlexDRWorker::modMinSpacingCostViaHelper(odb::Rect const&, int, dr
   2,00%  openroad                                       [.] void std::__push_heap<__gnu_cxx::__normal_iterator<drt::FlexWavefrontGr
   1,77%  libstdc++.so.6.0.33                            [.] std::_Rb_tree_insert_and_rebalance(bool, std::_Rb_tree_node_base*, std:
   1,73%  libstdc++.so.6.0.33                            [.] operator new(unsigned long)
   1,68%  openroad                                       [.] drt::frLayer::getMinSpacingValue(int, int, int, bool)
   1,65%  openroad                                       [.] drt::FlexGridGraph::getNextPathCost(drt::FlexWavefrontGrid const&, drt:
```

```
$ git log -1
commit 359623a96826d2137a82ffcd90b6d94627f2466a (HEAD, origin/master, origin/HEAD)
Merge: 5319993441 858947a5f0
Author: Matt Liberty <mliberty@precisioninno.com>
Date:   Tue Apr 15 03:16:31 2025 +0000

    Merge pull request #7139 from gadfort/bazel-build-type
    
    bazel: expose compilation mode to build_type to allow to proper splash
$ bazel build -c opt //:openroad
$ cd test
$ time bazel test -c opt jpeg_sky130hd-tcl
[tbd]
```

_Originally posted by @oharboe in https://github.com/The-OpenROAD-Project/OpenROAD/issues/7122#issuecomment-2802926183_

hzeller, 2025-04-15:
Looks like the bazel binary is not using threads (real time and user time are the same), whlie the cmake binary use a much shorter real time than user time which indicates use of threads.

Is there maybe something in the boost libraries or other places that does not properly trigger choosing multiple threads ?

oharboe, 2025-04-15:
By default, openroad built by cmake uses 1 thread.

```
$ OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad 
thread_openroad> thread_count
1
openroad> 
$ OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad  -threads 0
openroad> thread_count
48
openroad> 
$ OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad  -threads 5
openroad> thread_count
5
openroad> 
```

oharboe, 2025-04-15:
Same here...

```
$ OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
openroad> cpu_count
48
openroad> 
$ OpenROAD-flow-scripts/tools/OpenROAD/bazel-bin/openroad
% cpu_count
48
```

oharboe, 2025-04-15:
@hzeller threads and DRT did change recently, but I think it is a red herring 1a894769e29365382fa374266315f6a3b64fbf9b, since I see the following output for bazel built OpenROAD:

```
set_thread_count [cpu_count]
Max threads: 16
# Temporarily disable sta's threading due to random failures
sta::set_thread_count 1
```

hzeller, 2025-04-15:
it might know about the threads internally, but maybe not actually using them.
If there is a difference in user time vs. real time this indicates that there are  threads involved, as per second real-time, there are multiple seconds core time used.

This is why I am thinking something internally is not running multi-threaded where it should. So I suspect some commandline define somewhere that is different.

oharboe, 2025-04-15:
> it might know about the threads internally, but maybe not actually using them. If there is a difference in user time vs. real time this indicates that there are threads involved, as per second [...]

Actually, I think it is oposite from my observations: the cmake built openroad is using multithreading when it was told not to on the command line.

oharboe, 2025-04-15:
```
$ time ../bazel-bin/openroad -threads 0 -no_init -exit jpeg_sky130hd.tcl
OpenROAD HDL-HEAD bazel-build
Features included (+) or not (-): -GPU -GUI -Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ORD-0030] Using 48 thread(s).
[deleted]
running...
```

hzeller, 2025-04-15:
@maliberty is there something you can see where the cmake build and the bazel build result in different behaviour w.r.t. using threads ?

maliberty, 2025-04-15:
flow.tcl has `set_thread_count [cpu_count]`

oharboe, 2025-04-15:
> flow.tcl has `set_thread_count [cpu_count]`

Ah. Yet: that doesn't make it all the way into DRT in the bazel OpenROAD built case.

oharboe, 2025-04-18:
@maliberty This is pretty easy to reproduce and someone who knows what is going on in the code could get to the bottom of it quickly?

maliberty, 2025-04-19:
There is no problem setting the thread count that I see.  If I disable `suppress_message ORD 30` in helpers.tcl I see:
```
set_thread_count [cpu_count]
[INFO ORD-0030] Using 64 thread(s).
```

It seems to be systemically slower in the bazel build.  I would guess the compiler itself or perhaps boost.

maliberty, 2025-04-19:
More concerning is that you don't get the same result.  They are close enough not to fail but not equal.

maliberty, 2025-04-19:
Thread count is set correctly but it appears omp does not work in the bazel build!

oharboe, 2025-04-19:
> Thread count is set correctly but it appears omp does not work in the bazel build!

Nice find! Yes, this matches user vs real time observation from @hzeller

oharboe, 2025-04-19:
> More concerning is that you don't get the same result.  They are close enough not to fail but not equal.

Is there a test that is not running in bazel that catches this?

maliberty, 2025-04-19:
No, we don't test that the compiler works...  this has nothing to do with OR itself afaik.

hzeller, 2025-04-19:
@QuantamHD mentioned that he suspected that the libomp might not be properly configured.

maliberty, 2025-04-19:
> > More concerning is that you don't get the same result.  They are close enough not to fail but not equal.
> 
> Is there a test that is not running in bazel that catches this?

Sorry I misread and thought you were asking about omp.  jpeg_sky130hd.tcl shows this if you diff the tool output.

oharboe, 2025-04-19:
> > > More concerning is that you don't get the same result.  They are close enough not to fail but not equal.
> > 
> > Is there a test that is not running in bazel that catches this?
> 
> Sorry I misread and thought you were asking about omp.  jpeg_sky130hd.tcl shows this if you diff the tool output.

I guess it is hard to conclude much: if you disable OMP in ./build_openroad.sh in the same way as in bazel(if that is even possible), it is possible that you would see the same difference.

Perhaps best to wait for a bazel omp fix before expending more energy...

maliberty, 2025-04-19:
Using one thread would be sort of equivalent but it is so slow that I'd rather wait.

maliberty, 2025-04-24:
I think the problem is the lack of `-fopenmp` in the compile.  I tried adding to the .bazelrc 
```
build --cxxopt "-fopenmp"          --host_cxxopt "-fopenmp"
```
but now I get:
```
In file included from src/sta/dcalc/DelayCalc.cc:34:
In file included from src/sta/dcalc/PrimaDelayCalc.hh:29:
In file included from external/eigen~/Eigen/SparseCore:11:
external/eigen~/Eigen/Core:70:10: fatal error: 'omp.h' file not found
   70 | #include <omp.h>
      |          ^~~~~~~
```
@hzeller or @QuantamHD thoughts?

maliberty, 2025-04-24:
If I turn off the llvm toolchain and use my local compiler with the above -fopenmp then I can compile and the runtime is actually better than cmake.   We just need to find a way to get it through with the llvm toolchain.

maliberty, 2025-04-24:
Comparing actual compiler invocations I also find OR has
```
-O3
-Wp,-D_GLIBCXX_ASSERTIONS
-fPIE
-fopenmp
-fwrapv
```
while bazel has
```
-O2
-fdata-sections
-fexceptions
-ffunction-sections
-fno-cxx-modules
-fno-omit-frame-pointer
-frandom-seed=bazel-out/k8-opt/bin/_objs/openroad/OpenRoad.o
-fstack-protector
-g0
-no-canonical-prefixes
```
(plus various warnings on/off discrepancies and some -D that seem less important)

maliberty, 2025-04-24:
I'm guessing the runtime difference may result from `-Wp,-D_GLIBCXX_ASSERTIONS` which is helpful but makes things slower.

hzeller, 2025-04-25:
Runtime difference could also be, because we [use tcmalloc](https://github.com/The-OpenROAD-Project/OpenROAD/pull/7057) in the bazel build which is a faster memory allocator.

Since openmp is a crucial flag, I'd put `-fopenmp` in the [`OPENROAD_COPTS`](https://github.com/The-OpenROAD-Project/OpenROAD/blob/12bfc8eb4570fcde9b6f87aee3ccbbfd913dd1b3/BUILD.bazel#L82) set, but @QuantamHD should confirm if that then actually propagates to all  compilation units we depend on or if we should indeed have it in the global `.bazelrc` that Matt tried (the `.bazelrc` is used for local compilation, but once you use OpenROAD in some other workspace, only settings in BUILD files are used; `.bazelrc` files from imported projects are not read).

You can observe the flags passed to the compiler with running with `-s`
```
bazel build -c opt :openroad -s
```

I don't really know the details what bazel needs under the hood, but Ethan can help figuring out how to make sure that it ends up properly at the compiler invocation.

QuantamHD, 2025-04-25:
We probably should add it to Eigen as well, but if you just add -fopenmp to the copts of openroad does that compile?