The-OpenROAD-Project/OpenROAD #7154: MPL should support incremental placement in designs with FIXED macros

mikesinouye, 2025-04-16:
### Description

In https://github.com/The-OpenROAD-Project/OpenROAD/commit/259a4a6f8e084e19edb7d2bc46b31aa27f2536d3, it seems mpl was changed such that it cannot place UNPLACED macros if there are already existing FIXED macros in the FP area. If a user attempts to do mpl in this case, this logic will return an error: https://github.com/The-OpenROAD-Project/OpenROAD/blob/359623a96826d2137a82ffcd90b6d94627f2466a/src/mpl/src/clusterEngine.cpp#L157-L166

`[ERROR MPL-0050] Found fixed instance a/b/c/d/inst inside the floorplan area.`

My understanding is that the original mpl algorithm could handle this scenario. Is that understanding correct, and if so, is there a workaround to support this in the new mpl? I notice there are checks to ignore fixed macros, which makes me think this is possible to support it, e.g. https://github.com/The-OpenROAD-Project/OpenROAD/blob/359623a96826d2137a82ffcd90b6d94627f2466a/src/mpl/src/clusterEngine.cpp#L787-L793

The use case is: The RTL is modified but the reference FP DEF with fixed macros is not. Changes to add more macros in RTL and will require incremental placement, but I would like any original placement to be respected. If this is not currently possible, it would be great to support this use case.

AcKoucher, 2025-04-16:
@mikesinouye We never actually had the full support for partial macro placement i.e., having some macros placed and run mpl for others. Although these `ignore` checks exist in the clustering algorithm, the mechanism to handle fixed macros inside the annealer was never actually implemented.

The changes in [`259a4a6`](https://github.com/The-OpenROAD-Project/OpenROAD/commit/259a4a6f8e084e19edb7d2bc46b31aa27f2536d3) were to cover cases in which there are special fixed cells inside the core, but outside the area specified for the macro placement - these should not be touched by mpl.

I have some ideas to handle this scenario.

maliberty, 2025-04-16:
> We never actually had the full support for partial macro placement 

I think he was referring to TritonMacroPlacer (old mpl)