The-OpenROAD-Project/OpenROAD #7192: Feature request: `-connect_to_pads` option for `add_pdn_stripe`

mole99, 2025-04-22:
### Description

I'm not sure whether my request makes sense, so I'll describe the use case:

For my latest chip [Greyhound](https://github.com/mole99/greyhound-ihp/tree/main) I used a PDN ring that connects to the pads using the `-connect_to_pads` option. But since the pads themselves form an IO ring, I don't think that the PDN ring is really necessary (might depend on the PDN drop). Therefore it would be great if the PDN stripes could directly connect to the IO ring of the pads.

[image]

Since the entire IO ring is on TopMetal1, it means only the vertical PDN stripes (TopMetal2) could easily connect to the ring just by extending. How could the horinzontal stripes (TopMetal1) connect to the ring? Would it be possible that they change the layer before they connect to the outer part of the IO ring?

Would love to hear your thoughts on this.

gadfort, 2025-04-23:
For the vertical straps it would be fairly easy to implement something, but the horizontal would require some effort since layer switching is not supported. It would only really apply to libraries were access to power comes from the top. It would not be a quick fix and we would need a testcase the needs it to be able to test it out properly.

mole99, 2025-04-24:
Maybe there's a better method that we don't see yet. This seems to be a common situation. How would this be done with proprietary tooling? Do you always have a PDN ring there as well?

For example, I have this padring in gf180mcu (please ignore the PDN ring, it's not properly hooked up):

[image]

Here the power/ground io cells only have pins on Metal2. If they had pins on two consecutive metal layers, then it would be possible to just connect them with the PDN stripes, but this way we need a layer change for one of the directions.

[image]

I'm assuming these io cells weren't made just for the open PDK, so there must be a proper way to do this?

rovinski, 2025-04-26:
@mole99 for what it's worth, it is very common to have a PDN ring because usually the pin widths/layers do not match up with the preferred widths and layers of the top layers of the routing mesh. This results in less optimal usage of the top layers and also straps which do not align to the routing grid, which creates reduced routing resources.

Also, another aspect is that this works best if the pins of supply cells on opposite sides are aligned. Otherwise, if there is any offset, there will need to be separate straps created for each side to ensure power integrity.

So, I would advise being carful with this approach because it could prove to be much more trouble than it's worth just to save a bit of core area.

mole99, 2025-04-27:
Thanks for your input @rovinski, I can see why a PDN ring is usually preferred.

Regarding saving a bit of core area (which was my motivation): Right now the PDN ring is created outside the core area. Would it perhaps be possible to add an option to create it inside the corea area, i.e. to have standard cell rows under the PDN ring? I think this should be possible with the metal stackup in this case.

Any thoughts on this?

rovinski, 2025-04-27:
I think that generally it is possible by manually creating and connecting stripes, but for right now I don't believe there is a convenient option to do this. Rings are usually created around the boundary of the core area.

gadfort, 2025-04-27:
@mole99 the rings can be over the core, just need to use a negative offset for that, or specify the offset from the pads.

Syndace, 2025-09-05:
I would like to second this feature request. We are currently working on a very small design with just a few test structures connected to I/O pads:

[image]

Obviously the padcells and bondpads take up the most amount of space, but that's something we can't avoid as of now due to our measuring setup. However, the power ring also takes up a fair amount of space and is overkill for that handful of cells and super short stdcell stripes. It would be nice if we could avoid the additional power ring altogether and connect the stdcell power stripes directly to the power rails built-in to the I/O pads