The-OpenROAD-Project/OpenROAD #7226: Bazel: encode_tcl() not working yet.

hzeller, 2025-04-25:
### Describe the bug

When running a `tcl_encode()` target, this fails. e.g.

```
bazel build //:openroad_tcl --sandbox_debug
```

On a system that has strict control over the items used (NixOS), it fails with

```
env: 'python3': No such file or directory
```

Invoking the `py_binary()` target directly seems to work

```bash
bazel run //etc:file_to_string
usage: file_to_string.py [-h] --inputs INPUTS [INPUTS ...] --output OUTPUT --namespace NAMESPACE --varname VARNAMES
file_to_string.py: error: the following arguments are required: --inputs/-i, --output/-o, --namespace/-n, --varname/-v
```

This maybe indicates that instead of running the `py_binary()`, the `tcl_encode()` scripts attempts to execute a Python script directly without some bazel runtime which should not be visible to Bazel (or even could be expected to be on the system).

I suspect the action in `bazel/tcl_encode_or.bzl` needs to be done differently to explicitly use the `py_binary()`, though I don't know what. Maybe there needs to be a Python toolchain to use as the scripting environment used within bazel.

bottomline: Build still broken and I can't compile or work on OpenROAD.

(This is one of the reasons why I strongly dislike using Python during the build as it is a wobbly base to get some rock-solid build-experience).

### Expected Behavior

It is possible to compile OpenROAD without assumptions of system python available.

### Environment

```shell
NixOS.
```

### To Reproduce

```bash
nix-shell --pure -p bazel_7

bazel run //etc:file_to_string  # <- works fine
bazel build  //:openroad_tcl  # <- fails
```

### Relevant log output

```shell

```

QuantamHD, 2025-04-25:
I think they need to install a hermetic python toolchain in order for it to be found by your nixos system

QuantamHD, 2025-04-25:
https://www.anthonyvardaro.com/blog/hermetic-python-toolchain-with-bazel

hzeller, 2025-04-25:
The interesting things is that `bazel run` is working, so it already is using some Python that is available, probably through rules_pyton ?

The "hermetic" toolchains are all broken, as they are in fact _not_ hermetic (which maybe they could be if they would be fully statically linked or compiled from scratch in the context of the project). Hermetic should always mean that it is compiled from first principles just bootstrapping of a c++ compiler, compiling the Python interepreter witht that. What these "hermetic" toolchains do instead is downloading some random binary from the Internet, hoping it works on the host system.

(Also, same problem with the "hermetic" llvm toolchain, it creates compatibility issues)

Historically the Python interpreter had issues as it was linking random shared libraries https://github.com/bazelbuild/rules_python/issues/1211 - maybe that is fixed.

In the caes of this bug, I suspect that the use of tool/executable in the action context needs to be different to actually execute the `py_binary()` not attempt to interpret the file as a script.

oharboe, 2025-04-26:
`bazel run` has full access to the environment of the system, so no wonder it can give a different result...

hzeller, 2025-04-26:
> `bazel run` has full access to the environment of the system, so no wonder it can give a different result...

Well, there is no Python visible or in the `PATH` on my system (yes, I really don't like Python).

But with `bazel run`, it uses the Python interpreter that is baked into `rules_python` (in the runfiles of that binary, there is the Python interpreter linked in `bazel-bin/etc//file_to_string.runfiles/rules_python~~python~python_3_13_x86_64-unknown-linux-gnu/bin/python3` )

So the `encode_tcl()` needs to do make sure to use the `rules_python`-provided python interpreter.

oharboe, 2025-04-26:
@hzeller Shot in the dark(almost, we had a similar problem in bazel-orfs on Ubunut).

I don't have Nix:

```
$ git diff bazel
diff --git a/bazel/tcl_encode_or.bzl b/bazel/tcl_encode_or.bzl
index 781efc24f..4658b2f07 100644
--- a/bazel/tcl_encode_or.bzl
+++ b/bazel/tcl_encode_or.bzl
@@ -22,7 +22,7 @@ def _tcl_encode_or_impl(ctx):
         outputs = [output_file],
         inputs = ctx.files.srcs,
         arguments = [args],
-        #tools = [ctx.executable._tclsh, ctx.file._encode_script],
+        tools = depset(ctx.files._encode_script),
         executable = ctx.executable._encode_script,
     )
     return [DefaultInfo(files = depset([output_file]))]
```

oharboe, 2025-04-26:
I installed Nix. You have no idea how much work that is :-)

```
$ nix-shell --pure -p bazel_7

[nix-shell:~/OpenROAD-flow-scripts/tools/OpenROAD]$ bazel build  //:openroad_tcl
Extracting Bazel installation...
Starting local Bazel server and connecting to it...
INFO: Analyzed target //:openroad_tcl (144 packages loaded, 6003 targets configured).
INFO: Found 1 target...
Target //:openroad_tcl up-to-date:
  bazel-bin/openroad_tcl.cc
INFO: Elapsed time: 5.639s, Critical Path: 0.10s
INFO: 2 processes: 2 internal.
INFO: Build completed successfully, 2 total actions
```

oharboe, 2025-04-26:
```
$ git log -1
commit bb2a3e831a6cf49b64ca09cfa5bdca617fdb5bcd (HEAD, origin/master, origin/HEAD)
Merge: a8ef34ce2 8d984a2cf
Author: Cho Moon <141182150+precisionmoon@users.noreply.github.com>
Date:   Fri Apr 25 20:37:27 2025 -0700

    Merge pull request #7227 from andyfox-rushc/master
    
    Disconnect dbModBTerms during destroy to allow restore from journalling
```

hzeller, 2025-04-26:
The fix didn't work for me yet. I suspect you somehow need to get the `rules_python` provided python interpeter and pass it the script (I guess similar how the original thing was using the tcl interpeter to invoke the tcl script).

This is how  my minimal shell.nix looks like

$ cat shell.nix
```nix
{ pkgs ? import <nixpkgs> {} }:
pkgs.clangStdenv.mkDerivation {
  name = "or-bazel-env";

  GIT_SSL_CAINFO="${pkgs.cacert}/etc/ssl/certs/ca-bundle.crt";

  buildInputs = with pkgs; [
    bazel_7
    git
  ];
}
```

Edit MODULE.bazel to not use the llvm toolchain:

```
sed -i 's/^register_toolchains/#register_toolchains/' MODULE.bazel
```

Then

```
nix-shell --pure shell.nix
```

The `--pure` makes sure that not Python or other things leak into your environment.

... and in that environment:

```bash
bazel clean --expunge  # make sure we start from scratch
bazel build //:openroad_tcl
```

This is the error I get

```
ERROR: /home/hzeller/src/OpenROAD/BUILD.bazel:392:11: Action openroad_tcl.cc failed: (Exit 127): file_to_string failed: error executing Action command (from target //:openroad_tcl) bazel-out/k8-opt-exec-ST-d57f47055a04/bin/etc/file_to_string --inputs src/sta/tcl/Util.tcl src/Metrics.tcl src/OpenRoad.tcl --output bazel-out/k8-opt/bin/openroad_tcl.cc --varname ord_tcl_inits ... (remaining 2 arguments skipped)
```

Yes, first steps in Nix seem hard. But once you use NixOS, you never want to go back. I am using Linux for over 30 years now, and this is the best way to describe and run an operating system. Hermetic, well defined and reproducible all the way down. But it also allows to easily detect if some contexts are not well defined, like here, we we accidentally execute a Python script that has an assumption built-in to be using some System Python. So it really helps making sure things really work as they are meant to.

hzeller, 2025-04-26:
In the tcl version of this script, the tcl interpreter is used as the executable and the script as its argument.

https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/bazel/tcl_encode_sta.bzl#L21-L27

I suspect something similar would need to be done for the Python version, where the executable is the `rules_python` -provided interpreter.

oharboe, 2025-04-27:
@hzeller I tried your recipe, but couldn't reproduce the problem:

```
oyvind@small-cigar:~/OpenROAD-flow-scripts/tools/OpenROAD$ cat shell.nix 
{ pkgs ? import <nixpkgs> {} }:
pkgs.clangStdenv.mkDerivation {
  name = "or-bazel-env";

  GIT_SSL_CAINFO="${pkgs.cacert}/etc/ssl/certs/ca-bundle.crt";

  buildInputs = with pkgs; [
    bazel_7
    git
  ];
}

oyvind@small-cigar:~/OpenROAD-flow-scripts/tools/OpenROAD$ nix-shell --pure shell.nix

[nix-shell:~/OpenROAD-flow-scripts/tools/OpenROAD]$ bazel clean --expunge
INFO: Starting clean (this may take a while). Consider using --async if the clean takes more than several minutes.

[nix-shell:~/OpenROAD-flow-scripts/tools/OpenROAD]$ bazel build //:openroad_tcl
Starting local Bazel server and connecting to it...
INFO: Analyzed target //:openroad_tcl (141 packages loaded, 4099 targets configured).
INFO: Found 1 target...
Target //:openroad_tcl up-to-date:
  bazel-bin/openroad_tcl.cc
INFO: Elapsed time: 5.472s, Critical Path: 0.36s
INFO: 6 processes: 5 internal, 1 processwrapper-sandbox.
INFO: Build completed successfully, 6 total actions

[nix-shell:~/OpenROAD-flow-scripts/tools/OpenROAD]$ exit
exit
oyvind@small-cigar:~/OpenROAD-flow-scripts/tools/OpenROAD$ git diff
diff --git a/MODULE.bazel b/MODULE.bazel
index ea20d1185..86f9521e9 100644
--- a/MODULE.bazel
+++ b/MODULE.bazel
@@ -76,7 +76,7 @@ use_repo(llvm, "llvm_toolchain")
 # use_repo(llvm, "llvm_toolchain_llvm") # if you depend on specific tools in scripts
 
 # FYI: Comment out on NixOS where you'd like to use the local clang toolchain.
-register_toolchains("@llvm_toolchain//:all")
+#register_toolchains("@llvm_toolchain//:all")
 
 bazel_dep(
     name = "rules_python",
diff --git a/MODULE.bazel.lock b/MODULE.bazel.lock
index 6af53f0ab..07b202487 100644
--- a/MODULE.bazel.lock
+++ b/MODULE.bazel.lock
@@ -13571,7 +13571,7 @@
     },
     "@@rules_rust~//rust:extensions.bzl%rust": {
       "general": {
-        "bzlTransitiveDigest": "FnIJoZKUde7q+pZRBU+tmOS5ZKOBGq15Pgnh3+YD29Q=",
+        "bzlTransitiveDigest": "F8KXjOsPcF/0BaZgYTYcE+fL5TVGJvuAstwigwoONdw=",
         "usagesDigest": "p3vlEY5e6iHCg9d0OSqN8SPxiXdJFxI1wg1MKecojvs=",
         "recordedFileInputs": {},
         "recordedDirentsInputs": {},
oyvind@small-cigar:~/OpenROAD-flow-scripts/tools/OpenROAD$ git log -1
commit bb2a3e831a6cf49b64ca09cfa5bdca617fdb5bcd (HEAD, origin/master, origin/HEAD)
Merge: a8ef34ce2 8d984a2cf
Author: Cho Moon <141182150+precisionmoon@users.noreply.github.com>
Date:   Fri Apr 25 20:37:27 2025 -0700

    Merge pull request #7227 from andyfox-rushc/master
    
    Disconnect dbModBTerms during destroy to allow restore from journalling
```

oharboe, 2025-04-27:
@hzeller I'm a big fan of nix and we tried to switch to it in our organization, but if you have raft of dependencies that are installed in ubuntu you need, it isn't easy to get it all set up in Nix.

hzeller, 2025-04-27:
Are you using NixOS or the Nix package manager.

If you use the nix package manager on a regular system, your regular system still leaks through as you have Python and other things installed. So the first hashbang-line in the script `/usr/bin/env python3` will find a Python even though it should not be visible.

If you invoke the Python script with the interpreter provided by the `rules_python`, the first line doesn't matter anymore.

oharboe, 2025-04-27:
I installed Nix in Ubuntu and used --pure

hovind, 2025-04-28:
This might not be the idiomatic way to solve the problem, but is
```diff
diff --git a/bazel/tcl_encode_or.bzl b/bazel/tcl_encode_or.bzl
index 781efc24f..e03c006ee 100644
--- a/bazel/tcl_encode_or.bzl
+++ b/bazel/tcl_encode_or.bzl
@@ -13,6 +13,7 @@ def _tcl_encode_or_impl(ctx):
     output_file = ctx.actions.declare_file(outfile_name)
 
     args = ctx.actions.args()
+    args.add(ctx.executable._encode_script)
     args.add_all("--inputs", ctx.files.srcs)
     args.add("--output", output_file)
     args.add("--varname", ctx.attr.char_array_name)
@@ -22,8 +23,8 @@ def _tcl_encode_or_impl(ctx):
         outputs = [output_file],
         inputs = ctx.files.srcs,
         arguments = [args],
-        #tools = [ctx.executable._tclsh, ctx.file._encode_script],
-        executable = ctx.executable._encode_script,
+        tools = [ctx.executable._encode_script],
+        executable = ctx.toolchains["@rules_python//python:toolchain_type"].py3_runtime.interpreter,
     )
     return [DefaultInfo(files = depset([output_file]))]
 
@@ -53,4 +54,5 @@ tcl_encode = rule(
             cfg = "exec",
         ),
     },
+    toolchains = ["@rules_python//python:toolchain_type"],
 )
```
an option?

hzeller, 2025-04-28:
Nice!
Yes, this is working. And it actually looks properly idiomatic as you actually use the interpreter provided by `rules_python`.

Please send a PR.

commit b66d2093 (2025-04-30): bazel: fix python toolchain usage

fixes #7226
closed 2025-04-30 by maliberty