The-OpenROAD-Project/OpenROAD #7249: `bazel query --output=label_kind 'deps(//:openroad)'` failed: preloading transitive closure failed: no such package '@@[unknown repo 'openssl' requested from @@boost.asio~]//'

oharboe, 2025-05-01:
### Description

```
$ bazel query --output=label_kind 'deps(//:openroad)'
ERROR: Evaluation of query "deps(//:openroad)" failed: preloading transitive closure failed: no such package '@@[unknown repo 'openssl' requested from @@boost.asio~]//': The repository '@@[unknown repo 'openssl' requested from @@boost.asio~]' could not be resolved: No repository visible as '@openssl' from repository '@@boost.asio~'
Loading: 3 packages loaded
    Fetching repository @@bazel_tools~cc_configure_extension~local_config_cc; starting
    Fetching repository @@boringssl~; starting
```

maliberty, 2025-05-01:
I'm not clear this is an OR issue.

oharboe, 2025-05-01:
> I'm not clear this is an OR issue.

It happens when I try it in OR. a WORKSPACE (deprecated) vs bazlmod issue?

oharboe, 2025-05-01:
@hzeller Any insights?

hzeller, 2025-05-01:
This looks like a missing (optional, not used) dependency in `boost.asio`.

Filed https://github.com/bazelbuild/bazel-central-registry/issues/4489

maliberty, 2025-05-01:
Thanks.  Not an OR issue so closing.

closed 2025-05-01 by maliberty

oharboe, 2025-05-02:
@hzeller Thanks!