The-OpenROAD-Project/OpenROAD #7289: bazel: use http_archive() and patch file for OpenSTA instead of using git submodule

oharboe, 2025-05-06:
### Description

Use http_archive() with a patch for OpenSTA instead of using it as a git submodule.

This is more idiomatic for dependencies in bazel and makes it easy to patch OpenSTA while either waiting for an upstream change or patching it permantently such as for finding .tcl just like OpenROAD had to be modified for Bazel:

https://github.com/The-OpenROAD-Project/OpenROAD/blob/8408d9e67daba2a288371476ddcd7a0e7f81236d/src/Main.cc#L240-L251

oharboe, 2025-05-06:
@hzeller @QuantamHD @maliberty Thoughts?

hzeller, 2025-05-06:
+1 yes, sounds good.

oharboe, 2025-05-06:
@hzeller Any chance you could switch OpenSTA from git submodule to http_archive() or have a look to outline what work needs to be done?

This is either quickly done or there's some work needed to separate OpenROAD/OpenSTA BUILD.bazel. I'm not familiar with BUILD.bazel for OpenROAD, it is being heavily modified and I'm still learning bazel.

I can create a followon patch PR to port the tcl path finding problem.

hzeller, 2025-05-06:
I'd start by putting the opensta BUILD rules in a separate build file (maybe initially in `bazel/` ), and also store the patch in the bazel/ directory.

`http_archive()` accepts a patcch and a `build_file` so that is how you can assemble things.
Inspiration e.g. in https://github.com/hdl/bazel_rules_hdl/tree/main/dependency_support (it uses `maybe()` but in your case you'd use `http_archive()` directly)

lromor, 2025-05-06:
Hi all,

do you think it would be a good idea to put this in a [bzlmod extensions](https://bazel.build/external/extension)?

oharboe, 2025-05-06:
Made a quick attempt https://github.com/The-OpenROAD-Project/OpenROAD/pull/7291

QuantamHD, 2025-05-06:
I would suggest we not do this. Opensta and open road are a lot more linked than your average dep. When you http archive stuff you lose the ability to quickly edit the code in opensta, and gdb debug config gets more complicated.

Benefits do not outweigh the costs