The-OpenROAD-Project/OpenROAD #7349: `bazelisk build -c dbg :openroad --keep_going` fails to link

oharboe, 2025-05-12:
### Description

```
$ bazelisk build -c dbg :openroad --keep_going
INFO: Analyzed target //:openroad (0 packages loaded, 0 targets configured).
ERROR: /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/BUILD.bazel:115:10: Linking openroad failed: (Exit 1): cc_wrapper.sh failed: error executing CppLink command (from target //:openroad) external/toolchains_llvm~~llvm~llvm_toolchain/bin/cc_wrapper.sh @bazel-out/k8-dbg/bin/openroad-2.params

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
ld.lld: error: undefined symbol: std::__glibcxx_assert_fail(char const*, int, char const*, char const*)
>>> referenced by atomic_base.h:498 (/usr/lib/gcc/x86_64-linux-gnu/15/../../../../include/c++/15/bits/atomic_base.h:498)
>>>               bazel-out/k8-dbg/bin/external/edu_berkeley_abc/_objs/abc-lib/utilPth.pic.o:(std::atomic<bool>::load(std::memory_order) const)
>>> referenced by atomic_base.h:499 (/usr/lib/gcc/x86_64-linux-gnu/15/../../../../include/c++/15/bits/atomic_base.h:499)
>>>               bazel-out/k8-dbg/bin/external/edu_berkeley_abc/_objs/abc-lib/utilPth.pic.o:(std::atomic<bool>::load(std::memory_order) const)
>>> referenced by atomic_base.h:473 (/usr/lib/gcc/x86_64-linux-gnu/15/../../../../include/c++/15/bits/atomic_base.h:473)
>>>               bazel-out/k8-dbg/bin/external/edu_berkeley_abc/_objs/abc-lib/utilPth.pic.o:(std::atomic<bool>::store(bool, std::memory_order))
>>> referenced 2 more times
clang: error: linker command failed with exit code 1 (use -v to see invocation)
INFO: Found 1 target...
Target //:openroad failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 6.029s, Critical Path: 5.76s
INFO: 2 processes: 2 internal.
ERROR: Build did NOT complete successfully
```

oharboe, 2025-05-12:
@hzeller @QuantamHD Ideas?

QuantamHD, 2025-05-12:
[image]

It looks like this file is trying to distinguish between c++ and c. I added -x c++ to all the abc files in bazel rules maybe it needs some kind of link flag since these are c files.

QuantamHD, 2025-05-12:
Perhaps it wants a -latomic here? https://github.com/hdl/bazel_rules_hdl/blob/main/dependency_support/edu_berkeley_abc/bundled.BUILD.bazel#L1220

oharboe, 2025-05-13:
> Perhaps it wants a -latomic in the link opts? https://github.com/hdl/bazel_rules_hdl/blob/main/dependency_support/edu_berkeley_abc/bundled.BUILD.bazel#L1220

Mucked around with that, couldn't get it to work, same linker error.

```
diff --git a/WORKSPACE b/WORKSPACE
index 30963dbf73..f58940f2bd 100644
--- a/WORKSPACE
+++ b/WORKSPACE
@@ -12,15 +12,20 @@ rules_hdl_git_hash = "4bfc8987e521f2002e7b898ba94d3df4c6204913"
 
 rules_hdl_git_sha256 = "227ac0288299f2b0f31a188113cef9f733258398fd616215275bddab1e43d019"
 
-http_archive(
+# http_archive(
+#     name = "rules_hdl",
+#     sha256 = rules_hdl_git_sha256,
+#     strip_prefix = "bazel_rules_hdl-%s" % rules_hdl_git_hash,
+#     urls = [
+#         "https://github.com/hdl/bazel_rules_hdl/archive/%s.tar.gz" % rules_hdl_git_hash,
+#     ],
+# )
+local_repository(
     name = "rules_hdl",
-    sha256 = rules_hdl_git_sha256,
-    strip_prefix = "bazel_rules_hdl-%s" % rules_hdl_git_hash,
-    urls = [
-        "https://github.com/hdl/bazel_rules_hdl/archive/%s.tar.gz" % rules_hdl_git_hash,
-    ],
+    path = "../../../bazel_rules_hdl",
 )
```

```
$ git diff
diff --git a/dependency_support/edu_berkeley_abc/bundled.BUILD.bazel b/dependency_support/edu_berkeley_abc/bundled.BUILD.bazel
index a804cfa..9ab9586 100644
--- a/dependency_support/edu_berkeley_abc/bundled.BUILD.bazel
+++ b/dependency_support/edu_berkeley_abc/bundled.BUILD.bazel
@@ -1219,6 +1219,7 @@ cc_library(
         "-Wno-constant-logical-operand",
         "-x c++",
         "-std=c++17",
+        "-Wl,-latomic",
     ],
     defines = [
         "LIN64",
@@ -1232,7 +1233,8 @@ cc_library(
         "ABC_NAMESPACE=abc",
     ],
     includes = ["src/"],
-    linkopts = ["-ldl", "-lpthread"],
+    linkopts = ["-ldl", "-lpthread",         "-latomic",
+],
     linkstatic = True,
     textual_hdrs = glob(
         [
```

oharboe, 2025-05-13:
```
bazelisk build --cxxopt=-stdlib=libstdc++ --linkopt=-lstdc++ -c dbg :openroad --keep_going
```

From copilot:

The error indicates that the linker (ld.lld) is unable to resolve the symbol std::__glibcxx_assert_fail, which is part of the GNU C++ Standard Library implementation (libstdc++). This issue typically arises due to a mismatch between the compiler and the standard library being used, or because the correct standard library is not linked.

If you are using Clang as the compiler, ensure that it is configured to use the correct C++ standard library (libstdc++ or libc++).

oharboe, 2025-05-13:
Trying:

```
bazelisk build --cxxopt=-stdlib=libc++ --linkopt=-lc++ -c dbg :openroad
```

...

oharboe, 2025-05-13:
Pinata programming: this now produces a debug build... https://github.com/The-OpenROAD-Project/OpenROAD/pull/7358

oharboe, 2025-08-03:
@QuantamHD @maliberty FYI, I get the same problem now with `bazelisk build :openroad`; use-case bisection.

This works:

```
bazelisk build --cxxopt=-stdlib=libstdc++ --linkopt=-lstdc++ :openroad
```

oharboe, 2025-10-23:
@hzeller Any new insights on this one?

maliberty, 2025-10-23:
Why `--cxxopt=-stdlib=libstdc++ --linkopt=-lstdc++` ?

oharboe, 2025-10-23:
> Why `--cxxopt=-stdlib=libstdc++ --linkopt=-lstdc++` ?

A workaround that's good 'nuf to use for local testing.

https://github.com/The-OpenROAD-Project/OpenROAD/issues/7349

oharboe, 2025-10-23:
> `bazelisk build -c dbg //:openroad` builds without error on [eebcc18](https://github.com/The-OpenROAD-Project/OpenROAD/commit/eebcc18972ca47174e7af6904dd552efa3572cf6) for me.

Yes, some hermeticity is missing, it would seem. It will work on some platforms and others not.

oharboe, 2026-01-14:
Workaround:

    bazelisk build --cxxopt=-stdlib=libstdc++ --linkopt=-lstdc++ -c dbg :openroad

```
$ gdb --args bazel-out/k8-dbg/bin/openroad
GNU gdb (Ubuntu 16.3-1ubuntu2) 16.3
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from bazel-out/k8-dbg/bin/openroad...
(gdb) break main
Breakpoint 1 at 0x361a6d6: main. (12 locations)
(gdb) r
Starting program: /home/oyvind/.cache/bazel/_bazel_oyvind/896cc02f64446168f604c13ad7b60f8b/execroot/_main/bazel-out/k8-dbg/bin/openroad 

This GDB supports auto-downloading debuginfo from the following URLs:
  <https://debuginfod.ubuntu.com>
Enable debuginfod for this session? (y or [n]) y
Debuginfod has been enabled.
To make this setting permanent, add 'set debuginfod enabled on' to .gdbinit.
Downloading separate debug info for system-supplied DSO at 0x7ffff7fbc000
[Thread debugging using libthread_db enabled]                                                                                                                  
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Breakpoint 1.1, main (argc=1, argv=0x7fffffffdb58) at src/Main.cc:208
208	  std::array locales = {"en_US.UTF-8", "C.UTF-8", "C"};
(gdb) list
203	
204	int main(int argc, char* argv[])
205	{
206	  // This avoids problems with locale setting dependent
207	  // C functions like strtod (e.g. 0.5 vs 0,5).
208	  std::array locales = {"en_US.UTF-8", "C.UTF-8", "C"};
209	  for (auto locale : locales) {
210	    if (std::setlocale(LC_ALL, locale) != nullptr) {
211	      setenv("LC_ALL", locale, /* override */ 1);
212	      break;
(gdb) 
```