The-OpenROAD-Project/OpenROAD #7375: Segfault in `rcx::extDistWidthRCTable::~extDistWidthRCTable()`

gadfort, 2025-05-14:
### Describe the bug

Segfault in parasitic extraction.

### Expected Behavior

No segfault

### Environment

```shell
Check https://github.com/The-OpenROAD-Project/OpenROAD/compare/5932228c68f463ad701f2deacdb38c5c0ee120a8...f9cfe3ce45b77a2bb7ef9b34c8909c9770da9b5c for changes
```

### To Reproduce

https://drive.google.com/file/d/1-5pdFo-5iyK3HcJEnPg_vBrlZO6ju6X_/view?usp=sharing

```
tar xvf sc_issue_ethmac_job0_asap7_asap7sc7p5t_rvt_write.views0_20250514-011923.tar.gz
cd sc_issue_ethmac_job0_asap7_asap7sc7p5t_rvt_write.views0_20250514-011923
./run.sh
```

### Relevant log output

```shell
| ERROR    | job0_asap7_asap7s... | write.views          | 0 | Signal 11 received
| ERROR    | job0_asap7_asap7s... | write.views          | 0 | Stack trace:
| ERROR    | job0_asap7_asap7s... | write.views          | 0 |  0# 0x000055F29DBAE802 in /sc_tools/bin/openroad
| ERROR    | job0_asap7_asap7s... | write.views          | 0 |  1# 0x00007F9090A1D520 in /lib/x86_64-linux-gnu/libc.so.6
| ERROR    | job0_asap7_asap7s... | write.views          | 0 |  2# rcx::extDistWidthRCTable::~extDistWidthRCTable() in /sc_tools/bin/openroad
| ERROR    | job0_asap7_asap7s... | write.views          | 0 |  3# rcx::extMetRCTable::~extMetRCTable() in /sc_tools/bin/openroad
| ERROR    | job0_asap7_asap7s... | write.views          | 0 |  4# rcx::extRCModel::~extRCModel() in /sc_tools/bin/openroad
| ERROR    | job0_asap7_asap7s... | write.views          | 0 |  5# rcx::extMain::makeBlockRCsegs(char const*, unsigned int, unsigned int, double, bool, double, int, char const*) in /sc_tools/bin/openroad
| ERROR    | job0_asap7_asap7s... | write.views          | 0 |  6# 0x000055F29E3366D6 in /sc_tools/bin/openroad
| ERROR    | job0_asap7_asap7s... | write.views          | 0 |  7# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
| ERROR    | job0_asap7_asap7s... | write.views          | 0 |  8# 0x00007F9094FCB028 in /lib/x86_64-linux-gnu/libtcl8.6.so
| ERROR    | job0_asap7_asap7s... | write.views          | 0 |  9# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
| ERROR    | job0_asap7_asap7s... | write.views          | 0 | 10# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
| ERROR    | job0_asap7_asap7s... | write.views          | 0 | 11# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /sc_tools/bin/openroad
| ERROR    | job0_asap7_asap7s... | write.views          | 0 | 12# 0x000055F29DBAE003 in /sc_tools/bin/openroad
| ERROR    | job0_asap7_asap7s... | write.views          | 0 | 13# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
| ERROR    | job0_asap7_asap7s... | write.views          | 0 | 14# main in /sc_tools/bin/openroad
| ERROR    | job0_asap7_asap7s... | write.views          | 0 | 15# 0x00007F9090A04D90 in /lib/x86_64-linux-gnu/libc.so.6
| ERROR    | job0_asap7_asap7s... | write.views          | 0 | 16# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
| ERROR    | job0_asap7_asap7s... | write.views          | 0 | 17# _start in /sc_tools/bin/openroad
| WARNING  | job0_asap7_asap7s... | write.views          | 0 | Command failed with code -11. See log file /__w/siliconcompiler/siliconcompiler/build/ethmac/job0_asap7_asap7sc7p5t_rvt/write.views/0/write.views.log
```

maliberty, 2025-05-14:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/7372 should fix this (we had similar)

closed 2025-05-14 by maliberty

gadfort, 2025-05-14:
Thanks, I have it rerunning with the fix and it's passing again