The-OpenROAD-Project/OpenROAD #7388: In clock tree view add choice for macro / std cell display

maliberty, 2025-05-15:
### Description

In the clock tree view add an option to show only macros, only std cells, or both (current)

LucasYuki, 2025-05-21:
I've talked to @arthurjolo about this, and from what I understand, the problem seems to be that some branches of the tree are too narrow to see, usually the macro branches, but clockgates can also cause the same problem.

I suggest implementing a feature to view a subtree, so that the user can right-click on a node in the tree and select to view just its subtree, perhaps it opens a new CTS Viewer tab that can be closed.

maliberty, 2025-05-21:
A show/ hide subtree option would be more general and would work

PR #7484 (2025-06-09): gui: CTS Viewer show/hide subtree
Closes #7388.

Adds the feature to hide/show subtrees in CTS Viewer by right-clicking on a buffer.