The-OpenROAD-Project/OpenROAD #7397: segfault in `write_sdc` after `eliminate_dead_logic`

gadfort, 2025-05-18:
### Describe the bug

Calling `write_sdc` after `eliminate_dead_logic` with `group_path`'s causes a segfault (when a segfault does not occur, the written SDC contains non-ascii character for the names of the cells:
```
           [get_cells {}]\
           [get_cells {g�}]\
```

### Expected Behavior

No segfault and correct SDC file

### Environment

```shell
OpenROAD v2.0-21623-gce617810e4 
Features included (+) or not (-): +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
```

### To Reproduce

https://drive.google.com/file/d/1P6sdZN1h0YsYWkGJPGwXSYLfzTu-gUwr/view?usp=sharing

```
tar xvf test.tar.gz
openroad -exit run.tcl
```

### Relevant log output

```shell
[INFO ODB-0227] LEF file: asap7_tech.lef, created 24 layers, 9 vias
[INFO ODB-0227] LEF file: asap7sc7p5t_28_R.lef, created 212 library cells
Removed 2545 unused instances and 2545 unused nets.
Signal 11 received
Stack trace:
 0# 0x0000559A67C3BF63 in openroad
 1# 0x000072C675045330 in /lib/x86_64-linux-gnu/libc.so.6
 2# 0x000072C67519B8DC in /lib/x86_64-linux-gnu/libc.so.6
 3# odb::dbInst::getName[abi:cxx11]() in openroad
 4# sta::dbNetwork::name(sta::Instance const*) const in openroad
 5# sta::SdcNetwork::name(sta::Instance const*) const in openroad
 6# sta::Network::pathNameCmp(sta::Instance const*, sta::Instance const*) const in openroad
 7# sta::Network::pathNameLess(sta::Instance const*, sta::Instance const*) const in openroad
 8# void std::__insertion_sort<__gnu_cxx::__normal_iterator<sta::Instance const**, std::vector<sta::Instance const*, std::allocator<sta::Instance const*> > >, __gnu_cxx::__ops::_Iter_comp_iter<sta::InstancePathNameLess> >(__gnu_cxx::__normal_iterator<sta::Instance const**, std::vector<sta::Instance const*, std::allocator<sta::Instance const*> > >, __gnu_cxx::__normal_iterator<sta::Instance const**, std::vector<sta::Instance const*, std::allocator<sta::Instance const*> > >, __gnu_cxx::__ops::_Iter_comp_iter<sta::InstancePathNameLess>) in openroad
 9# void std::__merge_sort_with_buffer<__gnu_cxx::__normal_iterator<sta::Instance const**, std::vector<sta::Instance const*, std::allocator<sta::Instance const*> > >, sta::Instance const**, __gnu_cxx::__ops::_Iter_comp_iter<sta::InstancePathNameLess> >(__gnu_cxx::__normal_iterator<sta::Instance const**, std::vector<sta::Instance const*, std::allocator<sta::Instance const*> > >, __gnu_cxx::__normal_iterator<sta::Instance const**, std::vector<sta::Instance const*, std::allocator<sta::Instance const*> > >, sta::Instance const**, __gnu_cxx::__ops::_Iter_comp_iter<sta::InstancePathNameLess>) in openroad
10# sta::sortByPathName(sta::InstanceSet*, sta::Network const*) in openroad
11# sta::WriteSdc::writeExceptionFromTo(sta::ExceptionFromTo*, char const*, bool) const in openroad
12# sta::WriteSdc::writeException(sta::ExceptionPath*) const in openroad
13# sta::WriteSdc::writeExceptions() const in openroad
14# sta::WriteSdc::writeTiming() const in openroad
15# sta::writeSdc(sta::Instance*, char const*, char const*, bool, bool, int, bool, bool, sta::Sdc*) in openroad
16# sta::Sta::writeSdc(char const*, bool, bool, int, bool, bool) in openroad
17# 0x0000559A67E60E61 in openroad
18# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
19# 0x000072C67A0876BC in /lib/x86_64-linux-gnu/libtcl8.6.so
20# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
21# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
22# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
23# 0x0000559A67C3B6CF in openroad
24# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
25# main in openroad
26# 0x000072C67502A1CA in /lib/x86_64-linux-gnu/libc.so.6
27# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
28# _start in openroad
```

povik, 2025-05-29:
Minimized and reported to OpenSTA in parallaxsw/OpenSTA#248

povik, 2025-05-31:
STA upstream have fixed the issue

gadfort, 2025-05-31:
@maliberty can you bump the opensta module?

closed 2025-06-02 by povik