The-OpenROAD-Project/OpenROAD #7434: WARNING DRT-0088: Access point generation for .. did not meet: At least 3 sparse access points

stefanottili, 2025-05-26:
### Describe the bug

1. New warning
Rerunning ICCAD 2004 RISC2 on May 26 2025:
OR issues DRT-0088 warnings that previous run's didn't show.
WARNING DRT-0088: Access point generation for .. did not meet: At least 3 sparse access points

2. What does this warning mean ?

3. Runtime
OR places/routes 32k stdcell in ~25min without violations using 7 detailed route iterations.
Is 25min for 32k stdcells on 8 threads a reasonable runtime ?

### Expected Behavior

Compute pin access without warnings or make the warning more meaningful.

### Environment

```shell
M1 MacOS Sequoia homebrew, OpenROAD v2.0-21737-g5e7c8c4f45, May 26 2025
```

### To Reproduce

get lef/def or command from https://github.com/The-OpenROAD-Project/OpenROAD/issues/5548

ICCAD 2004, IBM 4+2 routing layer, artisan stdcells.

### Relevant log output

```shell
261 [INFO DRT-0165] Start pin access.
 262 [WARNING DRT-0088] Access point generation for lx1/ejtag/datain/U1/C did not meet :
 263         At least 3 sparse access points
 264 [WARNING DRT-0088] Access point generation for lx1/ejtag/datain/U1/A did not meet :
 265         At least 3 sparse access points
...
 725 [WARNING DRT-0088] Access point generation for lx1/CE0/macd/macu0/add_913/U145/AN did not meet :
 726         At least 3 sparse access points
 727 [WARNING DRT-0088] Access point generation for lx1/tap/JtagInst_JR_reg_0_/CK did not meet : 
 728         At least 3 sparse access points
 729 [INFO DRT-0078]   Complete 1598 pins.
 730 [INFO DRT-0079]   Complete 100 unique inst patterns.
 731 [INFO DRT-0079]   Complete 200 unique inst patterns.
 732 [INFO DRT-0081]   Complete 248 unique inst patterns.
 733 [INFO DRT-0082]   Complete 10000 groups.
 734 [INFO DRT-0084]   Complete 14210 groups.
 735 #scanned instances     = 32615
 736 #unique  instances     = 248
 737 #stdCellGenAp          = 9951
 738 #stdCellValidPlanarAp  = 0
 739 #stdCellValidViaAp     = 6083
 740 #stdCellPinNoAp        = 0
 741 #stdCellPinCnt         = 127785
 742 #instTermValidViaApCnt = 0
 743 #macroGenAp            = 0
 744 #macroValidPlanarAp    = 0
 745 #macroValidViaAp       = 0
 746 #macroNoAp             = 0
 747 [INFO DRT-0166] Complete pin access.
 748 [INFO DRT-0267] cpu time = 00:09:14, elapsed time = 00:01:11, memory = 895.41 (MB), peak = 926.03 (MB)
```

### Screenshots

After 4 iterations there are only couple of hotspots, all issues are in M1/M2.

[image]

closed 2025-05-27 by stefanottili

bnmfw, 2025-05-27:
Oh, sorry, tunneled on the warning question. Actually no, I'm not sure. I will run some designs of similar sizes, compare run time and get back to you

stefanottili, 2025-05-28:
well, it seems that iccad04/risc2 with 34k nets takes as long to route as ispd19_test6 with 180k nets.

1) FARADAY_ICCAD04Bench/RISC2

 15 [INFO ODB-0127] Reading DEF file: def/risc2.first_quadrant.noram.def.gz
 16 [INFO ODB-0128] Design: frisc
 17 [INFO ODB-0130]     Created 629 pins.
 18 [INFO ODB-0131]     Created 32615 components and 195545 component-terminals.
 19 [INFO ODB-0132]     Created 2 special nets and 65230 connections.
 20 [INFO ODB-0133]     Created 34034 nets and 128024 connections.

792 [INFO DRT-0267] cpu time = 02:44:59, elapsed time = 00:26:58, memory = 3706.42 (MB), peak = 3877.53 (MB)

2) https://www.ispd.cc/contests/19/#finalfiles

  7 [INFO ODB-0128] Design: ispd19_test6
  8 [INFO ODB-0094]                 Created 100000 Insts
  9 [INFO ODB-0097]                 Created 100000 Nets
 10 [INFO ODB-0130]     Created 1211 pins.
 11 [INFO ODB-0131]     Created 179881 components and 1152181 component-terminals.
 12 [INFO ODB-0132]     Created 2 special nets and 359762 connections.
 13 [INFO ODB-0133]     Created 179863 nets and 792078 connections.

536 [INFO DRT-0267] cpu time = 03:11:12, elapsed time = 00:27:33, memory = 6516.66 (MB), peak = 7104.47 (MB)

stefanottili, 2025-05-28:
ispd19_test6 needs 4 iterations, iccad04_risc2 needs 6

ispd19_test6 uses $$$ tool generated placement + guides
```
[INFO DRT-0194] Start detail routing.
[INFO DRT-0195] Start 0th optimization iteration.
[INFO DRT-0199]   Number of violations = 57045.
[INFO DRT-0267] cpu time = 01:28:01, elapsed time = 00:12:12, memory = 6680.25 (MB), peak = 6910.50 (MB)
[INFO DRT-0195] Start 1st optimization iteration.
[INFO DRT-0199]   Number of violations = 3580.
[INFO DRT-0267] cpu time = 01:16:52, elapsed time = 00:11:16, memory = 6993.77 (MB), peak = 7058.72 (MB)
[INFO DRT-0195] Start 2nd optimization iteration.
[INFO DRT-0199]   Number of violations = 1230.
[INFO DRT-0267] cpu time = 00:20:02, elapsed time = 00:03:02, memory = 6913.09 (MB), peak = 7104.47 (MB)
[INFO DRT-0195] Start 3rd optimization iteration.
[INFO DRT-0199]   Number of violations = 10.
[INFO DRT-0267] cpu time = 00:04:09, elapsed time = 00:00:41, memory = 6632.53 (MB), peak = 7104.47 (MB)
[INFO DRT-0195] Start 4th stubborn tiles iteration.
[INFO DRT-0199]   Number of violations = 0.
[INFO DRT-0267] cpu time = 00:02:06, elapsed time = 00:00:20, memory = 6515.72 (MB), peak = 7104.47 (MB)
[INFO DRT-0198] Complete detail routing.
[INFO DRT-0267] cpu time = 03:11:12, elapsed time = 00:27:33, memory = 6516.66 (MB), peak = 7104.47 (MB)
[INFO DRT-0180] Post processing.
```
iccad04_risc2 uses OR placement + groute
```
[INFO DRT-0194] Start detail routing.
[INFO DRT-0195] Start 0th optimization iteration.
[INFO DRT-0199]   Number of violations = 32208.
[INFO DRT-0267] cpu time = 00:52:56, elapsed time = 00:08:17, memory = 3008.58 (MB), peak = 3182.59 (MB)
[INFO DRT-0195] Start 1st optimization iteration.
[INFO DRT-0199]   Number of violations = 12772.
[INFO DRT-0267] cpu time = 00:44:28, elapsed time = 00:06:57, memory = 3392.05 (MB), peak = 3564.91 (MB)
[INFO DRT-0195] Start 2nd optimization iteration.
[INFO DRT-0199]   Number of violations = 11524.
[INFO DRT-0267] cpu time = 00:37:46, elapsed time = 00:05:43, memory = 3610.20 (MB), peak = 3693.50 (MB)
[INFO DRT-0195] Start 3rd optimization iteration.
[INFO DRT-0199]   Number of violations = 1397.
[INFO DRT-0267] cpu time = 00:24:38, elapsed time = 00:04:17, memory = 3723.19 (MB), peak = 3877.53 (MB)
[INFO DRT-0195] Start 4th optimization iteration.
[INFO DRT-0199]   Number of violations = 222.
[INFO DRT-0267] cpu time = 00:04:14, elapsed time = 00:01:13, memory = 3771.05 (MB), peak = 3877.53 (MB)
[INFO DRT-0195] Start 5th optimization iteration.
[INFO DRT-0199]   Number of violations = 22.
[INFO DRT-0267] cpu time = 00:00:50, elapsed time = 00:00:25, memory = 3725.88 (MB), peak = 3877.53 (MB)
[INFO DRT-0195] Start 6th optimization iteration.
[INFO DRT-0199]   Number of violations = 0.
[INFO DRT-0267] cpu time = 00:00:05, elapsed time = 00:00:03, memory = 3706.42 (MB), peak = 3877.53 (MB)
[WARNING DRT-0290] Warning: no DRC report specified, skipped writing DRC report
[INFO DRT-0198] Complete detail routing.
[INFO DRT-0267] cpu time = 02:44:59, elapsed time = 00:26:58, memory = 3706.42 (MB), peak = 3877.53 (MB)
[INFO DRT-0180] Post processing.
```

maliberty, 2025-05-28:
Elapsed time is nearly the same

stefanottili, 2025-05-29:
That’s my point, same detailed route time, 5x difference in design size.
iccad04/risc2 with 32k components takes as long to route as ispd19_test6 with 180k.
Why ?

From the info in the logs I cannot tell why the route times are similar, despite a 5x difference in components. Neither have a dense placement.
Obviously different stdcell libs and a different metal stack.

maliberty, 2025-05-29:
Put both runs in vtune or the like and see where the time goes.  It could be more complex DRC rules that take more time to check.

stefanottili, 2025-05-29:
> to reproduce
> get lef/def or command from https://github.com/The-OpenROAD-Project/OpenROAD/issues/5548

I had to shift the floorplan to 0,0 because dpl can’t handle a centered floorplan.

stefanottili, 2025-05-29:
The openroad script in the issues places and global/detail routes the def.
The first picture with rams is the global route map of ispd19_test6, the 2nd pic is risc2.
Both takes 25min to route.

maliberty, 2025-05-30:
I don't get the same results as you in terms or runtime or iterations.  Mixing the placer into the flow just adds noise along with not knowing what commit you are using.  Please file a proper issue with all the relevant information to make this reproducible if you want to pursue it further.